pyvisa/pyvisa-py #12: Problem connecting to TCPIP instrument

lmcculley, 2015-01-05:
Hi there! I'm trying to connect to my instrument via TCP/IP and I'm running into an issue. I can connect with telnet, but cannot connect using PyVISA with PyVISA-py. Here's what I'm getting. Any help debugging this will be greatly appreciated.

```
pi@raspberrypi ~ $ telnet
telnet> open 10.0.0.113 5025
Trying 10.0.0.113...
Connected to 10.0.0.113.
Escape character is '^]'.
*IDN?
B&K Precision ,891,4030MP1002, 1.24/ 1.04/ 1.03B
^]

telnet> quit
Connection closed.
pi@raspberrypi ~ $ sudo python
Python 2.7.8 (default, Oct 30 2014, 18:30:15)
[GCC 4.9.1] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import visa; rm = visa.ResourceManager("@py");
>>> inst = rm.open_resource("TCPIP::10.0.0.113::5025::INSTR")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/PyVISA-1.6.3.dev0-py2.7.egg/pyvisa/highlevel.py", line 1638, in open_resource
    res.open(access_mode, open_timeout)
  File "/usr/local/lib/python2.7/dist-packages/PyVISA-1.6.3.dev0-py2.7.egg/pyvisa/resources/resource.py", line 165, in open
    self.session, status = self._resource_manager.open_bare_resource(self._resource_name, access_mode, open_timeout)
  File "/usr/local/lib/python2.7/dist-packages/PyVISA-1.6.3.dev0-py2.7.egg/pyvisa/highlevel.py", line 1602, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.1.dev0-py2.7.egg/pyvisa-py/highlevel.py", line 180, in open
    sess = cls(session, resource_name, parsed)
  File "/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.1.dev0-py2.7.egg/pyvisa-py/sessions.py", line 183, in __init__
    self.after_parsing()
  File "/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.1.dev0-py2.7.egg/pyvisa-py/tcpip.py", line 45, in after_parsing
    self.interface = vxi11.CoreClient(self.parsed['host_address'])
  File "/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.1.dev0-py2.7.egg/pyvisa-py/protocols/vxi11.py", line 199, in __init__
    super(CoreClient, self).__init__(host, DEVICE_CORE_PROG, DEVICE_CORE_VERS)
  File "/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.1.dev0-py2.7.egg/pyvisa-py/protocols/rpc.py", line 569, in __init__
    pmap = TCPPortMapperClient(host)
  File "/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.1.dev0-py2.7.egg/pyvisa-py/protocols/rpc.py", line 547, in __init__
    RawTCPClient.__init__(self, host, PMAP_PROG, PMAP_VERS, PMAP_PORT)
  File "/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.1.dev0-py2.7.egg/pyvisa-py/protocols/rpc.py", line 321, in __init__
    self.connect()
  File "/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.1.dev0-py2.7.egg/pyvisa-py/protocols/rpc.py", line 325, in connect
    self.sock.connect((self.host, self.port))
  File "/usr/lib/python2.7/socket.py", line 224, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 111] Connection refused
>>> from pyvisa import util
>>> util.get_debug_info()
Machine Details:
   Platform ID:    Linux-3.12.33+-armv6l-with-debian-7.6
   Processor:

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        2.7.8
   Compiler:       GCC 4.9.1
   Bits:           32bit
   Build:          Oct 30 2014 18:30:15 (#default)
   Unicode:        UCS4

PyVISA Version: 1.6.3.dev0

Backends:
   ni:
      Version: 1.6.3.dev0 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.1.dev0
      ASRL INSTR: Available via PySerial (N/A)
      TCPIP INSTR: Available
      USB INSTR: Available via PyUSB (1.0.0rc1). Backend: libusb1

>>>
```

hgrecco, 2015-01-06:
A few hours ago we merged some important commits (and I just bumped the version to make it clear). Can you confirm that this happens with the latest version?

lmcculley, 2015-01-06:
Yes, I can confirm this happens with the latest version.

```
pi@raspberrypi ~/pyvisa-py $ sudo python
Python 2.7.8 (default, Oct 30 2014, 18:30:15)
[GCC 4.9.1] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import visa; rm = visa.ResourceManager("@py");
>>> inst = rm.open_resource("TCPIP::10.0.0.113::5025::INSTR")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/PyVISA-1.6.3.dev0-py2.7.egg/pyvisa/highlevel.py", line 1638, in open_resource
    res.open(access_mode, open_timeout)
  File "/usr/local/lib/python2.7/dist-packages/PyVISA-1.6.3.dev0-py2.7.egg/pyvisa/resources/resource.py", line 165, in open
    self.session, status = self._resource_manager.open_bare_resource(self._resource_name, access_mode, open_timeout)
  File "/usr/local/lib/python2.7/dist-packages/PyVISA-1.6.3.dev0-py2.7.egg/pyvisa/highlevel.py", line 1602, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "pyvisa-py/highlevel.py", line 180, in open
    sess = cls(session, resource_name, parsed)
  File "pyvisa-py/sessions.py", line 183, in __init__
    self.after_parsing()
  File "pyvisa-py/tcpip.py", line 45, in after_parsing
    self.interface = vxi11.CoreClient(self.parsed['host_address'])
  File "pyvisa-py/protocols/vxi11.py", line 199, in __init__
    super(CoreClient, self).__init__(host, DEVICE_CORE_PROG, DEVICE_CORE_VERS)
  File "pyvisa-py/protocols/rpc.py", line 569, in __init__
    pmap = TCPPortMapperClient(host)
  File "pyvisa-py/protocols/rpc.py", line 547, in __init__
    RawTCPClient.__init__(self, host, PMAP_PROG, PMAP_VERS, PMAP_PORT)
  File "pyvisa-py/protocols/rpc.py", line 321, in __init__
    self.connect()
  File "pyvisa-py/protocols/rpc.py", line 325, in connect
    self.sock.connect((self.host, self.port))
  File "/usr/lib/python2.7/socket.py", line 224, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 111] Connection refused
>>> from pyvisa import util; util.get_debug_info()
Machine Details:
   Platform ID:    Linux-3.12.33+-armv6l-with-debian-7.6
   Processor:

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        2.7.8
   Compiler:       GCC 4.9.1
   Bits:           32bit
   Build:          Oct 30 2014 18:30:15 (#default)
   Unicode:        UCS4

PyVISA Version: 1.6.3.dev0

Backends:
   ni:
      Version: 1.6.3.dev0 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.1.dev1
      ASRL INSTR: Available via PySerial (N/A)
      TCPIP INSTR: Available
      USB INSTR: Available via PyUSB (1.0.0rc1). Backend: libusb1

>>> quit()
pi@raspberrypi ~/pyvisa-py $ telnet 10.0.0.113 5025
Trying 10.0.0.113...
Connected to 10.0.0.113.
Escape character is '^]'.
*IDN?
B&K Precision ,891,4030MP1002, 1.24/ 1.04/ 1.03B
^]

telnet> quit
Connection closed.
```

hgrecco, 2015-01-06:
Does it work with the ni backend? 

Are you sure that this is an INSTR and not a SOCKET device?. 

The SOCKET resource class is not implemented yet in pyvisa-py but should be easy.

lmcculley, 2015-01-06:
I believe it is a SOCKET resource class. That would explain why. :-)

thliebig, 2015-01-06:
I just tried an "INSTR" device and it worked fine. But most devices I need are SOCKET as well. Would be nice to have ;)

great work anyways... BIG thanks

lmcculley, 2015-01-06:
I, too, would like to have SOCKET support. :-)

lmcculley, 2015-01-06:
I can confirm that it works as a SOCKET device with NI backend.

hgrecco, 2015-01-07:
@lmcculley If this is another instrument, please move it to another issue as it seems to have  a different source.

lmcculley, 2015-01-07:
No problem

thliebig, 2015-01-14:
@hgrecco what would be necessary to do to make a SOCKET device work? What is the difference between the INSTR and SOCKET connection?

hgrecco, 2015-01-14:
My understanding is that the SOCKET devices is just sending messages to a socket without all the message wrapping done by INSTR. In a way the SOCKET is to INSTR in TCPIP, like the RAW is to INSTR in USB. So opening a socket in python and calling `send` from `write` and `recv` from `read` should be close to have it done.

cmarqu, 2015-02-06:
Am I right in assuming that the examples from https://lantz.readthedocs.org/en/0.3/tutorial/rich-app.html cannot work with pyvisa-py because they are also SOCKET devices?

hgrecco, 2015-02-06:
@cmarqu  You are right, right now you need to use NI-VISA backend for that. But we should make it work soon. Any volunteers?! I would be happy to assist but I cannot code it and test it right now.

cmarqu, 2015-02-09:
I could only test, but not code, sorry.

hgrecco, 2015-02-09:
@cmarqu With this type of libraries that depend on the OS, drivers, etc; testing is extremely important and difficult. So Thanks for keeping and eye and reporting

BernardoGO, 2015-03-15:
I was able to connect by removing the ::TCPIP and putting only the IP address on the connection string.

hgrecco, 2015-03-15:
@BernardoGO What do you mean? Can provide more info?

BernardoGO, 2015-03-15:
@hgrecco,

I've got it working using this code:
        import visa;
        rm = visa.ResourceManager("@py")
        inst = rm.open_resource("TCPIP::192.168.134.102")
        inst.write(cmd)

but if I use: "TCPIP::192.168.134.102:INS.......", it doesnt work

hgrecco, 2015-03-15:
@BernardoGO strange, because `TCPIP::192.168.134.102` and `TCPIP::192.168.134.102::INSTR` should yield the same resource after parsing.

Humberto4902, 2015-03-16:
It says not found if using ::INSTR

hgrecco, 2015-03-16:
I will take a look. Thanks for reporting

hgrecco, 2015-08-25:
This should be solved. Feel free to reopen.

closed 2015-08-25 by hgrecco

andisspire, 2016-02-03:
I have also the same problem works with telnet, but doesn't with pyvisa.
The pyvisa lists me 0 devices
The pyvisa-py lists me 32 devices, all /dev/tty*

andisspire, 2016-02-03:
I am on 64bit machine with 64bit drivers in place

zamora18, 2018-09-11:
> This should be solved. Feel free to reopen.

@hgrecco How was this solved? I didn't see a reference to a PR. I also see that #38 was addressing a similar issue.

Was it solved by doing [this](https://github.com/pyvisa/pyvisa-py/issues/12#issuecomment-81187145) as suggested by @BernardoGO? If so, I am not having success opening the resource. What is the proper way to connect to the instrument?

Thanks

MatthieuDartiailh, 2018-09-11:
@zamora18  could you post your installation details, the kind of instrument you are trying to interface to, and the script you use ? Without that it is hard to help ?

zamora18, 2018-09-11:
This is the output of `python -m visa info`:
```
Machine Details:
   Platform ID:    Linux-4.15.0-33-generic-x86_64-with-debian-stretch-sid
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /home/zamora18/anaconda2/bin/python
   Version:        2.7.15
   Compiler:       GCC 7.2.0
   Bits:           64bit
   Build:          May  1 2018 23:32:55 (#default)
   Unicode:        UCS4

PyVISA Version: 1.9.1

Backends:
   ni:
      Version: 1.9.1 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.3.0
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named serial
      TCPIP INSTR: Available 
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named usb
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named usb
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named gpib
      TCPIP SOCKET: Available
```
The instrument I am trying to connect to is a [Keysight FieldFox N9918A Microwave Analyzer](https://www.keysight.com/en/pdx-x201927-pn-N9918A/fieldfox-handheld-microwave-analyzer-265-ghz?cc=US&lc=eng). I have a direct ethernet connection from my computer to the instrument. 

This is the script I run:
```python
import visa
rm = visa.ResourceManager('@py')
inst_name = "TCPIP0::169.254.251.233::inst0::INSTR"
inst = rm.open_resource(inst_name)
inst.timeout = 5000 # ms
print (inst.query("*IDN?"))
inst.close()
```
When I run, I get this error:
```
Traceback (most recent call last):
  File "test_pyvisa_simple.py", line 4, in <module>
    inst = rm.open_resource(inst_name)
  File "/home/zamora18/anaconda2/lib/python2.7/site-packages/pyvisa/highlevel.py", line 1724, in open_resource
    res.open(access_mode, open_timeout)
  File "/home/zamora18/anaconda2/lib/python2.7/site-packages/pyvisa/resources/resource.py", line 209, in open
    self.session, status = self._resource_manager.open_bare_resource(self._resource_name, access_mode, open_timeout)
  File "/home/zamora18/anaconda2/lib/python2.7/site-packages/pyvisa/highlevel.py", line 1681, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "/home/zamora18/anaconda2/lib/python2.7/site-packages/pyvisa-py/highlevel.py", line 194, in open
    sess = cls(session, resource_name, parsed, open_timeout)
  File "/home/zamora18/anaconda2/lib/python2.7/site-packages/pyvisa-py/sessions.py", line 212, in __init__
    self.after_parsing()
  File "/home/zamora18/anaconda2/lib/python2.7/site-packages/pyvisa-py/tcpip.py", line 64, in after_parsing
    self.open_timeout)
  File "/home/zamora18/anaconda2/lib/python2.7/site-packages/pyvisa-py/protocols/vxi11.py", line 202, in __init__
    DEVICE_CORE_VERS, open_timeout)
  File "/home/zamora18/anaconda2/lib/python2.7/site-packages/pyvisa-py/protocols/rpc.py", line 715, in __init__
    pmap = TCPPortMapperClient(host, open_timeout)
  File "/home/zamora18/anaconda2/lib/python2.7/site-packages/pyvisa-py/protocols/rpc.py", line 691, in __init__
    open_timeout)
  File "/home/zamora18/anaconda2/lib/python2.7/site-packages/pyvisa-py/protocols/rpc.py", line 429, in __init__
    self.connect((open_timeout / 1000.0) + 1.0)
  File "/home/zamora18/anaconda2/lib/python2.7/site-packages/pyvisa-py/protocols/rpc.py", line 459, in connect
    raise RPCError('can\'t connect to server')
pyvisa-py.protocols.rpc.RPCError: can't connect to server
```
Let me know if you need more information.

MatthieuDartiailh, 2018-09-13:
Thanks for detailed answer. So first this RPCError should never bubble up to you so this should be fixed. Second can you ping the instrument using the command line ? Did you properly configure your network card (fixed IP address starting with 169.233) ? I am asking because it looks like the system does not find the instrument.
Also please update pyvisa-py to 0.3.1 to avoid a small bug that could show up later.