pyvisa/pyvisa-py #13: Problem connecting to TCP/IP INSTR class device

lmcculley, 2015-01-07:
I'm having a problem with INSTR devices, as well. I can confirm my INSTR device does work with NI backend. 

Working NI backend
```
C:\Users\lmcculley\Documents\libusb>python
Python 2.7.9 (default, Dec 10 2014, 12:28:03) [MSC v.1500 64 bit (AMD64)] on win
32
Type "help", "copyright", "credits" or "license" for more information.
>>> import visa; rm = visa.ResourceManager();
>>> inst = rm.open_resource("TCPIP::10.0.0.189::INSTR")
>>> inst.query("*IDN?")
u''
>>> inst.read_termination = '\n'
>>> inst.query("*IDN?")
u'B&K Precision Corporation;TDO3000B;123456;V2.02.0F.13 110315D'
>>>
```

Non-working PY backend
```
pi@raspberrypi ~/pyvisa-py $ sudo python
Python 2.7.8 (default, Oct 30 2014, 18:30:15)
[GCC 4.9.1] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import visa; rm = visa.ResourceManager("@py");
>>> inst = rm.open_resource("TCPIP::10.0.0.189::INSTR")
>>> inst.query("*IDN?")
u''
>>> inst.read_termination = "\n"
>>> inst.query("*IDN?")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/PyVISA-1.6.3.dev0-py2.7.egg/pyvisa/resources/messagebased.py", line 384, in query
    return self.read()
  File "/usr/local/lib/python2.7/dist-packages/PyVISA-1.6.3.dev0-py2.7.egg/pyvisa/resources/messagebased.py", line 309, in read
    message = self.read_raw().decode(enco)
  File "/usr/local/lib/python2.7/dist-packages/PyVISA-1.6.3.dev0-py2.7.egg/pyvisa/resources/messagebased.py", line 283, in read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "pyvisa-py/highlevel.py", line 297, in read
    return self.sessions[session].read(count)
  File "pyvisa-py/tcpip.py", line 106, in read
    self.lock_timeout, flags, term_char)
  File "pyvisa-py/protocols/vxi11.py", line 217, in device_read
    self.unpacker.unpack_device_read_resp)
  File "pyvisa-py/protocols/rpc.py", line 237, in make_call
    pack_func(args)
  File "pyvisa-py/protocols/vxi11.py", line 115, in pack_device_read_parms
    self.pack_int(term_char)
  File "/usr/lib/python2.7/xdrlib.py", line 46, in result
    raise ConversionError(e.args[0])
xdrlib.ConversionError: cannot convert argument to integer
>>> inst.read_termination = "\r"
>>> inst.query("*IDN?")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/PyVISA-1.6.3.dev0-py2.7.egg/pyvisa/resources/messagebased.py", line 384, in query
    return self.read()
  File "/usr/local/lib/python2.7/dist-packages/PyVISA-1.6.3.dev0-py2.7.egg/pyvisa/resources/messagebased.py", line 309, in read
    message = self.read_raw().decode(enco)
  File "/usr/local/lib/python2.7/dist-packages/PyVISA-1.6.3.dev0-py2.7.egg/pyvisa/resources/messagebased.py", line 283, in read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "pyvisa-py/highlevel.py", line 297, in read
    return self.sessions[session].read(count)
  File "pyvisa-py/tcpip.py", line 106, in read
    self.lock_timeout, flags, term_char)
  File "pyvisa-py/protocols/vxi11.py", line 217, in device_read
    self.unpacker.unpack_device_read_resp)
  File "pyvisa-py/protocols/rpc.py", line 237, in make_call
    pack_func(args)
  File "pyvisa-py/protocols/vxi11.py", line 115, in pack_device_read_parms
    self.pack_int(term_char)
  File "/usr/lib/python2.7/xdrlib.py", line 46, in result
    raise ConversionError(e.args[0])
xdrlib.ConversionError: cannot convert argument to integer
>>> inst.read_termination = "\r\n"
>>> inst.query("*IDN?")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/PyVISA-1.6.3.dev0-py2.7.egg/pyvisa/resources/messagebased.py", line 384, in query
    return self.read()
  File "/usr/local/lib/python2.7/dist-packages/PyVISA-1.6.3.dev0-py2.7.egg/pyvisa/resources/messagebased.py", line 309, in read
    message = self.read_raw().decode(enco)
  File "/usr/local/lib/python2.7/dist-packages/PyVISA-1.6.3.dev0-py2.7.egg/pyvisa/resources/messagebased.py", line 283, in read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "pyvisa-py/highlevel.py", line 297, in read
    return self.sessions[session].read(count)
  File "pyvisa-py/tcpip.py", line 106, in read
    self.lock_timeout, flags, term_char)
  File "pyvisa-py/protocols/vxi11.py", line 217, in device_read
    self.unpacker.unpack_device_read_resp)
  File "pyvisa-py/protocols/rpc.py", line 237, in make_call
    pack_func(args)
  File "pyvisa-py/protocols/vxi11.py", line 115, in pack_device_read_parms
    self.pack_int(term_char)
  File "/usr/lib/python2.7/xdrlib.py", line 46, in result
    raise ConversionError(e.args[0])
xdrlib.ConversionError: cannot convert argument to integer
>>>
```

hgrecco, 2015-02-19:
Is this still a problem?

thliebig, 2015-02-20:
I'm able to talk to my INSTR device over LAN without a problem.
- pyVisa 1.6.3 and pyvisa-py of today, running Kubuntu 14.04.
- I'm not using any termination.
```python
    import visa
    rm = visa.ResourceManager('@py')
    my_instrument = rm.open_resource('TCPIP::192.168.1.10::inst0::INSTR')
    my_instrument.write("*IDN?")
    print my_instrument.read()
```
Can be closed??

hgrecco, 2015-02-20:
Let's close it. @lmcculley Feel free to reopen if needed

closed 2015-02-20 by hgrecco