pyvisa/pyvisa-py #16: Support for non-decimal numbers in device string, issue in return value of usb/get_attribute

t-kopp, 2015-02-16:
Maybe I'm missing something but currently non-decimal device strings don't seem to be supported. This should be easily possible. Feel free to point out any issues with this.

the get_attribute() function in usb.py returns only one value if the attribute is unknown. This raises an error because it's obviously not iterable.
        try:
            return self._get_attribute(attribute), constants.StatusCode.success
        except UnknownAttribute as e:
            logger.exception(str(e))
            return constants.StatusCode.error_nonsupported_attribute

hgrecco, 2015-02-16:
You are completely right about the missing return value after `UnknownAttribute`. Please look in the whole code as this error repeats a few times.

Regarding the non-decimal device strings. I full agree that we should handle this situations gracefully. Your proposal sounds reasonable and would work if we keep the output of the resource name parser as strings (which is the case)

t-kopp, 2015-02-16:
Regarding the other possible suspects, I found:
highlevel.py:
line 332/334
line 351/353

Currently not sure what the correct fix for those is, append ", constants.sucess" to 334 and 353?

tcpip.py:
line 163/165

fix for that: "return NONE, StatusCode.error_timeout" in line 165 ?

Did you find any others?

hgrecco, 2015-02-17:
I think there are more in usb.py but fix those and I will revisit all of them.

t-kopp, 2015-02-17:
Ok, added them. Not sure if the fixes are correct though - could you please check?

hgrecco, 2015-02-19:
I think this is fixed. Feel free to reopen if needed

closed 2015-02-19 by hgrecco