pyvisa/pyvisa-py #18: USB Raw devices

t-kopp, 2015-02-18:
USB Raw devices don't seem to be supported at the moment. Is this just some bug I don't see or is there something missing? If the latter is the case I'll iron out a couple issues with my locally working solution and submit it as a PR

hgrecco, 2015-02-18:
They are not supported right now. It is a bug: lack of time of the developers :-D
It would be great if you can submit a PR. Remeber to register the Session class that you create using the following decorator:
```python
@Session.register(constants.InterfaceType.usb, 'RAW')
```

t-kopp, 2015-02-18:
yes, currently did it dirty by replacing INSTR support :S will clean that up and come back to you

hgrecco, 2015-02-19:
Mereged in #19

closed 2015-02-19 by hgrecco