pyvisa/pyvisa-py #21: No backend available on Win7 with latest versions

jonnojohnson, 2015-03-05:
```
C:\Users\marnj>python -m visa info
Machine Details:
   Platform ID:    Windows-7-6.1.7601-SP1
   Processor:      Intel64 Family 6 Model 58 Stepping 9, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Anaconda\python.exe
   Version:        2.7.7
   Compiler:       MSC v.1500 32 bit (Intel)
   Bits:           32bit
   Build:          Jun 11 2014 10:41:43 (#default)
   Unicode:        UCS2

PyVISA Version: 1.6.4.dev0

Backends:
   ni:
      Version: 1.6.4.dev0 (bundled with PyVISA)
      #1: C:\windows\system32\visa32.dll:
         found by: auto
         bitness: 32
         Vendor: Keysight Technologies
         Impl. Version: 1360312085
         Spec. Version: 5243136
      #2: C:\windows\system32\visa32.dll:
         found by: auto
         bitness: 32
         Vendor: Keysight Technologies
         Impl. Version: 1360312085
         Spec. Version: 5243136
   py:
      Version: 0.2.dev0
      ASRL INSTR: Available via PySerial (N/A)
      TCPIP INSTR: Available
      USB INSTR: Available via PyUSB (1.0.0b2). Backend: N/A
      USB RAW: Available via PyUSB (1.0.0b2). Backend: N/A
```

```
Python 2.7.7 |Anaconda 2.1.0 (32-bit)| (default, Jun 11 2014, 10:41:43) [MSC v.1
500 32 bit (Intel)] on win32
Type "help", "copyright", "credits" or "license" for more information.
Anaconda is brought to you by Continuum Analytics.
Please check out: http://continuum.io/thanks and https://binstar.org
>>> import visa
>>> rm = visa.ResourceManager('@py')
>>> rm.list_resources()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Anaconda\lib\site-packages\pyvisa\highlevel.py", line 1554, in list_r
esources
    find_list, return_counter, instrument_description, err = lib.find_resources(
self.session, query)
  File "C:\Anaconda\lib\site-packages\pyvisa-py\highlevel.py", line 240, in find
_resources
    for key, st in sessions.Session.iter_valid_session_classes()], [])
  File "C:\Anaconda\lib\site-packages\pyvisa-py\usb.py", line 148, in list_resou
rces
    for dev in usbtmc.find_tmc_devices():
  File "C:\Anaconda\lib\site-packages\pyvisa-py\protocols\usbtmc.py", line 69, i
n find_tmc_devices
    return find_devices(vendor, product, serial_number, is_usbtmc, **kwargs)
  File "C:\Anaconda\lib\site-packages\pyvisa-py\protocols\usbutil.py", line 200,
 in find_devices
    return usb.core.find(find_all=True, custom_match=cm, **kwargs)
  File "C:\Anaconda\lib\site-packages\usb\core.py", line 1199, in find
    raise ValueError('No backend available')
ValueError: No backend available
```

hgrecco, 2015-03-05:
Thanks for testing this. It would be great if you help me to debug this. A few questions
- Have you tried in other version of Windows? Which?
- It seems that pyserial and pyusb are installed. 
Can you actually try to import them in Python and do something with them. The "N/A" is weird: it should look like:
```
      Version: 0.1.dev0
      ASRL INSTR: Available via PySerial (14.1.0)
      TCPIP INSTR: Available
      USB INSTR: Available via PyUSB (1.0.0rc1). Backend: libusb0
```

jonnojohnson, 2015-03-05:
I only have the one Win7 machine right now. I did test in both 32 & 64 bit python installs with the same results.
I get no errors importing serial or usb.
```
In [4]: usb.version_info
Out[4]: (1, 0, 0, 'b2')
```
Not sure what else to try with serial or usb.

hgrecco, 2015-03-05:
can you also post the output of:
```
import serial
ver = serial.version
```

hgrecco, 2015-03-05:
I think that there are three issues here:
1.- The PySerial version is not reported (this is PyVISA-py responsibility)
2.- PyUSB does not have a backend. (this is PyUSB responsability and might be related to a lack of libusb or similar in your system. See https://github.com/walac/pyusb/issues/26)
3.- PyVISA-py still tries to use USB even if PyUSB has no backend (this is PyVISA-py responsibility)

I will fix 3, it would be great if you can look at 2.

And if you can help to understand 1.

jonnojohnson, 2015-03-06:
```
In [5]: serial.version
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-5-bc4648fda10c> in <module>()
----> 1 serial.version

AttributeError: 'module' object has no attribute 'version'
```
I have libusb1 installed, version 1.3.0

jonnojohnson, 2015-03-06:
Just realized this:
```
In [9]: serial.VERSION
Out[9]: '2.7'
```

hgrecco, 2015-03-06:
In my computer

```
>>> import serial
>>> serial.version
['14', '1', '0']
>>> serial.VERSION
'2.7'
```

Looking at the PySerial code, I think it is quite messy how the version is handled due to 2to3 usage.

Regarding PyUSB (point 2), I am convinced that the problem is not in PyVISA but in your PyUSB installation or in PyUSB. I would suggest that you try to make the `find` command work as described here: https://github.com/walac/pyusb/blob/master/docs/tutorial.rst

jonnojohnson, 2015-03-06:
I think my usb problem is the install (or lack of) libusb1.0. I had installed libusb1 using pip but hadn't realized it's just a wrapper for libusb1 and doesn't get me the dll. It looks like installing libusb1 itself is a royal pain on Windows that I don't have time to mess with right now.

hgrecco, 2015-03-06:
Then I will close this issue and open others por the pending stuff.

closed 2015-03-06 by hgrecco