pyvisa/pyvisa-py #24: Are GPIB instruments supported?

bessman, 2015-03-13:
README suggests yes:

> Requirements:
> - linux-gpib (to interface with gpib instruments, only on linux)

Readthedocs suggests no:

> Which resource types are supported?
> And later:
> - GPIB INSTR

hgrecco, 2015-03-13:
Not yet! The plan is to do it but we are out of coding hands right now. I will be happy to guide you if you want to implement it.

bessman, 2015-03-13:
I'm afraid I'm not much of a programmer, but I would love to try my hand at it. How newb friendly would you rate this task?

hgrecco, 2015-03-13:
It is not hard. It is basically wrapping linux-gpib within session. You can use SerialSession as a guide and should be relatively straight forward. As I said if you want to give it a shot I will be happy to help.

hgrecco, 2015-03-18:
Thanks for your work! Merged

closed 2015-03-18 by hgrecco

cmarqu, 2015-05-20:
The index.rst still needs updating with this info (great BTW!). I can provide a pull-request tomorrow unless someone beats me to it.

hgrecco, 2015-05-20:
you are right! Please do.

hgrecco, 2015-05-20:
I will make a blog post about PyVISA 1.7, PyVISA-sim 0.2 and (soon to be released) PyVISA-py 0.2

benjaminhwilliams, 2015-08-18:
Thank you all for this excellent hard work.  Could you please provide a link to the blog post?  And do you know whether PyVISA-py 0.2 is likely to be released soon?  As cmarqu pointed out in May, the docs (https://pyvisa-py.readthedocs.org/en/latest/) still refer to version 0.1 but the readme seems more up-to-date.  Let me know if I can be of any assistance in bringing the release forward.