pyvisa/pyvisa-py #29: PyVisa not using PyVisa-py ->NotImplementedError

sasa00, 2015-04-23:
I just started with the pyVisa-py Tutorial on https://pyvisa-py.readthedocs.org/en/latest/:

```Python
import visa

rm = visa.ResourceManager('@py')
print(rm)
rm.list_resources()
```
and get the following error message:

```
raceback (most recent call last):
  File "/src/usb/ATP_USB_Double/usbutu.py", line 5, in <module>
    rm.list_resources()
  File "/usr/local/lib/python2.7/dist-packages/pyvisa/highlevel.py", line 1554, in list_resources
    find_list, return_counter, instrument_description, err = lib.find_resources(self.session, query)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa/highlevel.py", line 500, in find_resources
    raise NotImplementedError
NotImplementedError
```
I am using a Raspi 2 with Linux:
Linux (none) 3.18.7-v7+ #755 SMP PREEMPT Thu Feb 12 17:20:48 GMT 2015 armv7l GNU/Linux

    python -m visa info
gives me:
```
Machine Details:
   Platform ID:    Linux-3.18.7-v7+-armv7l-with-debian-7.8
   Processor:

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        2.7.3
   Compiler:       GCC 4.6.3
   Bits:           32bit
   Build:          Mar 18 2014 05:13:23 (#default)
   Unicode:        UCS4

PyVISA Version: 1.6.3

Backends:
   ni:
      Version: 1.6.3 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.2.dev0
      ASRL INSTR:
         Please install PySerial to use this resource type.
         No module named tools.list_ports
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named gpib
      TCPIP INSTR: Available
      USB INSTR: Available via PyUSB (N/A). Backend: libusb10
      USB RAW: Available via PyUSB (N/A). Backend: libusb10
```

It seems to me that Python is searching inside the pyvisa/highlevel.py instead of the pyvisa-py/highlevel.py for the list_resources function.

sasa00, 2015-04-23:
I am using Python 2.7.3:
```
Python 2.7.3 (default, Mar 18 2014, 05:13:23)
[GCC 4.6.3] on linux2
```

hgrecco, 2015-04-23:
We are changing the way we implement the resource listing and you were caught in the middle. Install:
- pyvisa from pypi 
- pyvisa-py from pypi or from github but 0ea232aa9ba101fcb543fa3062956364908d97d0 (not the latest)

sasa00, 2015-04-24:
I installed the packages form https://pypi.python.org/pypi. I am now getting a different Errormessage:
```
Resource Manager of Visa Library at unset
Traceback (most recent call last):
  File "/src/usb/ATP_USB_Double/usbutu.py", line 6, in <module>
    rm.list_resources()
  File "/usr/local/lib/python2.7/dist-packages/PyVISA-1.6.3-py2.7.egg/pyvisa/highlevel.py", line 1554, in list_resources
    find_list, return_counter, instrument_description, err = lib.find_resources(self.session, query)
  File "/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.1-py2.7.egg/pyvisa-py/highlevel.py", line 236, in find_resources
    for key, st in sessions.Session.iter_valid_session_classes()], [])
  File "/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.1-py2.7.egg/pyvisa-py/usb.py", line 55, in list_resources
    serial_number=dev.serial_number,
AttributeError: 'Device' object has no attribute 'serial_number'
```
```
python -m visa info
```
gives me:

```
Machine Details:
   Platform ID:    Linux-3.18.7-v7+-armv7l-with-debian-7.8
   Processor:

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        2.7.3
   Compiler:       GCC 4.6.3
   Bits:           32bit
   Build:          Mar 18 2014 05:13:23 (#default)
   Unicode:        UCS4

PyVISA Version: 1.6.3

Backends:
   ni:
      Version: 1.6.3 (bundled with PyVISA)
      #1: /usr/local/lib/libvisa.so.0.0.0:
         found by: auto
         bitness: 32
         Could not get more info:
            VI_ERROR_NSUP_ATTR (-1073807331): The specified attribute is not defined or supported by the referenced object.
   py:
      Version: 0.1
      ASRL INSTR: Available via PySerial (N/A)
      TCPIP INSTR: Available
      USB INSTR: Available via PyUSB (N/A). Backend: libusb10
```

hgrecco, 2015-04-24:
Very likely you PyUSB is old, update to the latest beta version.

sasa00, 2015-04-24:
That was the trick! I downloaded the git master of PyUSB and now I can see my instrument in the list.

hgrecco, 2015-04-24:
As soon as PyUSB release a stable version 1 I will add a check.

closed 2015-04-24 by hgrecco