pyvisa/pyvisa-py #30: struct problems on python2.7

daritter, 2015-04-27:


hgrecco, 2015-04-27:
does it work on Python 3 as well? If not, please use the same solution as here:

https://github.com/hgrecco/pyvisa/blob/master/pyvisa/util.py#L28

daritter, 2015-04-27:
I don't have python3 on the machine where the device is connected so I cannot verify it directly. But `struct.pack(b">I", 5) == struct.pack(">I", 5)` seems to work fine in python 3.4 on my local machine. 

Of course the solution in utils.py would also be fine. Do you want me to prepare a pull request for the two occurences in protocols/rpc.py and the four in protocols/usbmtc.py ?

hgrecco, 2015-04-27:
I have the idea that there was a bug in certain versions of python (maybe in 3.2) that required us to do the solution coded in `utils.py`. We are putting common compatibility code in PyVISA compat package  to make it easier to maintain and test. Therefore, please:

1. The first against PyVISA, moving the struct functions (but removing the underscore) to an new module `compat\struct.py` and then use them directly from there in `utils.py`. 

2. Then a second PR against PyVISA-py using this module in `rpc.py` and `usbtmc.py`

Thanks for looking into this!

PR #31 (2015-04-29): Use pyvisa compat/struct.py for python < 2.7.8
Use pyvisa.compat.struct instead of standard library struct module

resolves #30
closed 2015-04-29 by hgrecco