pyvisa/pyvisa-py #33: ASRL error

mcoughlin, 2015-05-05:
Hi,

First of all, your code is awesome, and thank you so much for it.
I believe there may be a missing class for ASRL connections though. I call pyvisa (using the latest pyvisa-py and pyvisa) as follows:

    rm = visa.ResourceManager("@py")
    resource = u'ASRL1::/dev/ttyUSB0::INSTR'
    my_instrument = rm.open_resource(resource)

The traceback for the error is:

Traceback (most recent call last):
  File "takedata.py", line 51, in <module>
    my_instrument = rm.open_resource(resource)
  File "/usr/local/lib/python2.7/dist-packages/PyVISA-1.6.4.dev0-py2.7.egg/pyvisa/highlevel.py", line 1609, in open_resource
    info = self.resource_info(resource_name)
  File "/usr/local/lib/python2.7/dist-packages/PyVISA-1.6.4.dev0-py2.7.egg/pyvisa/highlevel.py", line 1576, in resource_info
    ret, err = self.visalib.parse_resource_extended(self.session, resource_name)
  File "/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.2.dev0-py2.7.egg/pyvisa-py/highlevel.py", line 275, in parse_resource_extended
    parsed = rname.parse_resource_name(resource_name)
  File "/usr/local/lib/python2.7/dist-packages/PyVISA-1.6.4.dev0-py2.7.egg/pyvisa/rname.py", line 187, in from_string
    raise InvalidResourceName.bad_syntax(subclass.syntax,
AttributeError: type object 'ASRLInstr' has no attribute 'syntax'

and my setup is as follows:
Machine Details:
   Platform ID:    Linux-3.18.0-trunk-rpi-armv6l-with-debian-7.8
   Processor:      

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        2.7.3
   Compiler:       GCC 4.6.3
   Bits:           32bit
   Build:          Mar 18 2014 05:13:23 (#default)
   Unicode:        UCS4

PyVISA Version: 1.6.4.dev0

Backends:
   ni:
      Version: 1.6.4.dev0 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.2.dev0
      ASRL INSTR: Available via PySerial (2.7)
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named gpib
      TCPIP INSTR: Available 
      USB INSTR: Available via PyUSB (1.0.0rc1). Backend: libusb1
      USB RAW: Available via PyUSB (1.0.0rc1). Backend: libusb1

My apologies if I simply misunderstand how to correctly deal with ASRL based instruments.

Thank you,
Michael

hgrecco, 2015-05-06:
I would say that both. You are using the ASRL Syntax wrong but you also found a bug in the PyVISA error handler. Please update to the latest PyVISA and you will see the following error.
```
Could not parse 'ASRL1::/dev/ttyUSB0::INSTR'. The syntax is 'ASRL[board][::INSTR]' (too many parts).
```
Which means that in your case you need to use the following resource name `ASRL/dev/ttyUSB0::INSTR`

mcoughlin, 2015-05-06:
Thank you sir, this makes sense. Reinstalling and using the correct syntax results in:

ASRL/dev/ttyUSB0::INSTR
Traceback (most recent call last):
  File "takedata.py", line 51, in <module>
    my_instrument = rm.open_resource(resource)
  File "/usr/local/lib/python2.7/dist-packages/PyVISA-1.6.4.dev1-py2.7.egg/pyvisa/highlevel.py", line 1631, in open_resource
    res.open(access_mode, open_timeout)
  File "/usr/local/lib/python2.7/dist-packages/PyVISA-1.6.4.dev1-py2.7.egg/pyvisa/resources/resource.py", line 170, in open
    self.session, status = self._resource_manager.open_bare_resource(self._resource_name, access_mode, open_timeout)
  File "/usr/local/lib/python2.7/dist-packages/PyVISA-1.6.4.dev1-py2.7.egg/pyvisa/highlevel.py", line 1595, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.2.dev0-py2.7.egg/pyvisa-py/highlevel.py", line 198, in open
    sess = cls(session, resource_name, parsed)
  File "/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.2.dev0-py2.7.egg/pyvisa-py/sessions.py", line 182, in __init__
    constants.VI_ATTR_RSRC_NAME: parsed['canonical_resource_name'],
TypeError: tuple indices must be integers, not unicode
pi@raspberrypi ~/python/keithley/bin $ 

This error occurs on both my Mac and Raspberry Pi.

hgrecco, 2015-05-06:
This is also due to the latest changes. I will fix it tonight.

hgrecco, 2015-05-10:
Can you try now?

mcoughlin, 2015-05-10:
Perfect! Thank you so much!

closed 2015-05-10 by mcoughlin