pyvisa/pyvisa-py #35: Fixed compatibility with struct

hgrecco, 2015-05-19:
See https://github.com/hgrecco/pyvisa/issues/149

hgrecco, 2015-05-19:
This might be fixed in the latest pyvisa (from pypi) and pyvisa-py (from github) Can you update both?

enzo96, 2015-05-19:
Hi, 

I found a solution which solves the problem. I installed python-ivi and its extension python-vxi11, which allows to control instruments over Ethernet. Now I can very simply communicate with my Analyzer. Thank you for your help, 
Bye

hgrecco, 2015-05-19:
Lets port that solution to pyvisa-py

hgrecco, 2015-05-19:
Fixed with #31

closed 2015-05-19 by hgrecco

twitst12, 2017-04-20:
Hi enzo96
I'm working on the same project that you. Have you been able to automate the antenna rotator in order to make radiance diagrams measurments using the raspberry pi ? Because I have some issues related with NI-VISA. And my problem is that I can't install it on raspberry. 
what have you done to install it ?