pyvisa/pyvisa-py #37: Problem with using "query" and 'write' method on raspberry pi2

FusionLi, 2015-05-19:
I am using raspberry pi 2 with pyvisa to control Tektronix signal oscilloscope. The issue I got is I cannot use 'write' and 'query' methods and both of them gave the same error:
>>> inst.write('measu:meas1:source1 ch1')
>>> inst.query('measu:meas1:value?')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/pyvisa/resources/messagebased.py", line 187, in write
    count = self.write_raw(message.encode(enco))
  File "/usr/local/lib/python2.7/dist-packages/pyvisa/resources/messagebased.py", line 165, in write_raw
    return self.visalib.write(self.session, message)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/highlevel.py", line 281, in write
    return self.sessions[session].write(data)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/usb.py", line 117, in write
    count = self.interface.write(data)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/protocols/usbtmc.py", line 286, in write
    bytes_sent += raw_write(data)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/protocols/usbtmc.py", line 202, in write
    return self.usb_send_ep.write(data)
File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 349, in write
    return self.device.write(self, data, timeout)
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 878, in write
    self.__get_timeout(timeout)
  File "/usr/local/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 760, in bulk_write
    timeout)
  File "/usr/local/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 856, in __write
    _check(retval)
  File "/usr/local/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 552, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
  File "/usr/local/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 541, in _strerror
    return _lib.libusb_strerror(errcode).decode('utf8')
  File "/usr/lib/python2.7/ctypes/__init__.py", line 378, in __getattr__
    func = self.__getitem__(name)
  File "/usr/lib/python2.7/ctypes/__init__.py", line 383, in __getitem__
    func = self._FuncPtr((name_or_ordinal, self))
AttributeError: /lib/arm-linux-gnueabihf/libusb-1.0.so.0: undefined symbol: libusb_strerror

hgrecco, 2015-05-19:
According to https://github.com/walac/pyusb/issues/57  your libusb version is not supported by pyusb. (Non python users have reported the same problem with older libusb https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=100607)

FusionLi, 2015-05-20:
I tried the solution above, but none of them worked for me. Any more suggestion? Thanks

hgrecco, 2015-05-20:
It would be really good to know if the problem is in pyvisa or pyusb. Till now, it seems to be a pyusb problem. Is there anything you can test using just pyusb?

FusionLi, 2015-05-20:
Not really, cuz most of my codes need to get measured data from the devices. If "query" does not work, it really hard for me to continue.

hgrecco, 2015-05-20:
I mean that you try another device. i.e. you use pyusb to get the value of a mouse or to exchange info between two computer. We would like to know if pyusb is working on your arm

FusionLi, 2015-05-20:
Oh, I see. I tried another device but same error appeared when I used 'query' method. Also, I tried everything on a brand-new raspberry pi 2 but it seems like they have the same problem

hgrecco, 2015-05-20:
try with a mouse. It involves only reading 
http://www.orangecoat.com/how-to/read-and-decode-data-from-your-mouse-using-this-pyusb-hack

FusionLi, 2015-05-20:
Hi, I tried the script but it gave me the error:
Traceback (most recent call last):
  File "mouse.py", line 14, in <module>
    if dev.is_kernel_driver_active(interface) is True:
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 992, in is_kernel_driver_active
    interface)
  File "/usr/local/lib/python2.7/dist-packages/usb/backend/__init__.py", line 363, in is_kernel_driver_active
    _not_implemented(self.is_kernel_driver_active)
  File "/usr/local/lib/python2.7/dist-packages/usb/backend/__init__.py", line 79, in _not_implemented
    raise NotImplementedError(func.__name__)
NotImplementedError: is_kernel_driver_active

hgrecco, 2015-05-20:
I am pretty sure you have a problem with your PyUSB installation. I would suggest that you post this last thing in the PyUSB issues, they might be able to help.

FusionLi, 2015-05-20:
Thanks a lot, I will do it definitely.

FusionLi, 2015-05-20:
By the way, I solved the mouse testing script problem via installing pyusb1.0.0b1 from https://github.com/arvydas/blinkstick-python/wiki/NotImplementedError-is_kernel_driver_active-on-some-Linux-systems. However, this package is not compatible with the one used by pyvisa because after I installed pyusb1.0.0b1, I cannot even use "open_resource() and list_resources()" methods.

hgrecco, 2015-05-20:
Yes, we use heavily some of the new features of pyusb. Let's try to work with the PyUSB people to solve it.

hgrecco, 2015-05-20:
Post this on PyUSB issues, I will jump in as well

FusionLi, 2015-05-20:
Thanks, I posted the issue already: https://github.com/walac/pyusb/issues/101.

hgrecco, 2015-05-21:
As discussed in the pyusb thread, it is not a pyvisa issue. Therefore I am closing it.

PR #383 (2023-10-09): fix(hislip): drop expected message id on read timeout
As per #376, current implementation was not resetting `expected_message_id` in case of timeout, potentially leading to stale responses being returned to the caller.

This is very much a bandaid while I'm working on both better integration tests and on a (hopefully) more robust hislip implementation.

- [x] Closes #37
closed 2015-05-21 by hgrecco