pyvisa/pyvisa-py #40: Unable to connect to a Keithley DMM 2701 through TCPIP/Ethernet

nktsaini, 2015-05-25:
__I am trying to use PyVISA-py as a backend on a BeagleBone Black which runs Debian 7.8 wheezy. I have installed the latest versions of both PyVISA and PyVISA-py available as of the time of this post.__ 

__When I run__

>python -m visa info

I get:
```
Machine Details:
   Platform ID:    Linux-3.8.13-bone70-armv7l-with-debian-7.8
   Processor:      

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        2.7.3
   Compiler:       GCC 4.6.3
   Bits:           32bit
   Build:          Mar 14 2014 17:55:54 (#default)
   Unicode:        UCS4

PyVISA Version: 1.8.dev0

Backends:
   ni:
      Version: 1.8.dev0 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.2.dev2
      ASRL INSTR:
         Please install PySerial to use this resource type.
         No module named tools.list_ports
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named gpib
      TCPIP INSTR: Available 
      USB INSTR: Available via PyUSB (1.0.0rc1). Backend: libusb1
      USB RAW: Available via PyUSB (1.0.0rc1). Backend: libusb1
```

__Then I go ahead and try communication with the Keithley using these commands:__

>import visa
>rm = visa.ResourceManager('@py')
>dmm = rm.open_resource('TCPIP::10.66.53.50::inst0:INSTR')

__I get:__
```
error                                     Traceback (most recent call last)
<ipython-input-3-478994839e1c> in <module>()
----> 1 dmm = rm.open_resource('TCPIP::10.66.53.50::inst0::INSTR')

/usr/local/lib/python2.7/dist-packages/PyVISA-1.8.dev0-py2.7.egg/pyvisa/highlevel.pyc in open_resource(self, resource_name, access_mode, open_timeout, **kwargs)
   1632                 raise ValueError('%r is not a valid attribute for type %s' % (key, res.__class__.__name__))
   1633 
-> 1634         res.open(access_mode, open_timeout)
   1635 
   1636         for key, value in kwargs.items():

/usr/local/lib/python2.7/dist-packages/PyVISA-1.8.dev0-py2.7.egg/pyvisa/resources/resource.pyc in open(self, access_mode, open_timeout)
    182         logger.debug('%s - opening ...', self._resource_name, extra=self._logging_extra)
    183         with self._resource_manager.ignore_warning(constants.VI_SUCCESS_DEV_NPRESENT):
--> 184             self.session, status = self._resource_manager.open_bare_resource(self._resource_name, access_mode, open_timeout)
    185 
    186             if status == constants.VI_SUCCESS_DEV_NPRESENT:

/usr/local/lib/python2.7/dist-packages/PyVISA-1.8.dev0-py2.7.egg/pyvisa/highlevel.pyc in open_bare_resource(self, resource_name, access_mode, open_timeout)
   1596         :return: Unique logical identifier reference to a session.
   1597         """
-> 1598         return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
   1599 
   1600     def open_resource(self, resource_name,

/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.2.dev2-py2.7.egg/pyvisa-py/highlevel.pyc in open(self, session, resource_name, access_mode, open_timeout)
    196         cls = sessions.Session.get_session_class(parsed.interface_type_const, parsed.resource_class)
    197 
--> 198         sess = cls(session, resource_name, parsed)
    199 
    200         return self._register(sess), constants.StatusCode.success

/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.2.dev2-py2.7.egg/pyvisa-py/sessions.pyc in __init__(self, resource_manager_session, resource_name, parsed)
    182                       constants.VI_ATTR_RSRC_CLASS: parsed.resource_class,
    183                       constants.VI_ATTR_INTF_TYPE: parsed.interface_type}
--> 184         self.after_parsing()
    185 
    186     def after_parsing(self):

/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.2.dev2-py2.7.egg/pyvisa-py/tcpip.pyc in after_parsing(self)
     43         # TODO: board_number not handled
     44         # TODO: lan_device_name not handled
---> 45         self.interface = vxi11.CoreClient(self.parsed.host_address)
     46 
     47         self.lock_timeout = 10000

/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.2.dev2-py2.7.egg/pyvisa-py/protocols/vxi11.pyc in __init__(self, host)
    195         self.packer = Vxi11Packer()
    196         self.unpacker = Vxi11Unpacker('')
--> 197         super(CoreClient, self).__init__(host, DEVICE_CORE_PROG, DEVICE_CORE_VERS)
    198 
    199     def create_link(self, id, lock_device, lock_timeout, name):

/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.2.dev2-py2.7.egg/pyvisa-py/protocols/rpc.pyc in __init__(self, host, prog, vers)
    580     """
    581     def __init__(self, host, prog, vers):
--> 582         pmap = TCPPortMapperClient(host)
    583         port = pmap.get_port((prog, vers, IPPROTO_TCP, 0))
    584         pmap.close()

/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.2.dev2-py2.7.egg/pyvisa-py/protocols/rpc.pyc in __init__(self, host)
    558 
    559     def __init__(self, host):
--> 560         RawTCPClient.__init__(self, host, PMAP_PROG, PMAP_VERS, PMAP_PORT)
    561         PartialPortMapperClient.__init__(self)
    562 

/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.2.dev2-py2.7.egg/pyvisa-py/protocols/rpc.pyc in __init__(self, host, prog, vers, port)
    323     def __init__(self, host, prog, vers, port):
    324         Client.__init__(self, host, prog, vers, port)
--> 325         self.connect()
    326 
    327     def connect(self):

/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.2.dev2-py2.7.egg/pyvisa-py/protocols/rpc.pyc in connect(self)
    327     def connect(self):
    328         self.sock = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
--> 329         self.sock.connect((self.host, self.port))
    330 
    331     def close(self):

/usr/lib/python2.7/socket.pyc in meth(name, self, *args)
    222 
    223 def meth(name,self,*args):
--> 224     return getattr(self._sock,name)(*args)
    225 
    226 for _m in _socketmethods:

error: [Errno 110] Connection timed out
```

__Any help is highly appreciated.__

__PS. I can ping the IP address 10.66.53.50 with zero packets lost.__

hgrecco, 2015-05-25:
Can you put a breakpoint to findout the host and port. And then do:
```python
>>> print((host, socket))
>>> sock = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
>>> sock.connect((host, post))
```
(
By the way. Not related but I will strongly suggest to upgrade your Python to the latest 2.7 . There are important bugfixes there.

hgrecco, 2015-05-25:
I added a few log call. Enable logging to check it out.

nktsaini, 2015-05-26:
I am new with python so I'm looking up how to put a breakpoint. I'll reply as soon as possible.

nktsaini, 2015-05-26:
I am using the python debugger (pdb) which automatically sets breakpoint after each line. But after the line:

```
dmm = rm.open_resource('TCPIP::10.66.53.50::inst0::INSTR')
```

when I ask it to print the host and socket, it says the name 'host' is not defined.

hgrecco, 2015-05-26:
Can you update to the latest pyvisa-py (in github) and then run your script prepending:
```
import visa
visa.log_to_screen()
```

nktsaini, 2015-05-26:
__I update the pyvisa-py on my system and added the line. This is what I get now:__

```
debian@beaglebone:~/Documents/Keithley PyVISA$ python comms.py
2015-05-26 00:00:23,467 - pyvisa - DEBUG - SerialSession was not imported No module named tools.list_ports.
2015-05-26 00:00:23,472 - pyvisa - DEBUG - USBSession was not imported No module named usb.
2015-05-26 00:00:23,476 - pyvisa - WARNING - (<InterfaceType.usb: 7>, u'INSTR') is already registered in the ResourceManager. Overwriting with unavailable Please install PyUSB to use this resource type.
No module named usb
2015-05-26 00:00:23,478 - pyvisa - WARNING - (<InterfaceType.usb: 7>, u'RAW') is already registered in the ResourceManager. Overwriting with unavailable Please install PyUSB to use this resource type.
No module named usb
2015-05-26 00:00:23,484 - pyvisa - DEBUG - USBRawSession was not imported No module named usb.
2015-05-26 00:00:23,569 - pyvisa - DEBUG - TCPIPSession was correctly imported.
2015-05-26 00:00:23,576 - pyvisa - DEBUG - GPIBSession was not imported No module named gpib.
2015-05-26 00:00:23,586 - pyvisa - DEBUG - Created library wrapper for unset
2015-05-26 00:00:23,593 - pyvisa - DEBUG - Created ResourceManager with session 7914990
2015-05-26 00:00:23,596 - pyvisa - DEBUG - TCPIP::10.66.53.50::inst0:INSTR - opening ...
2015-05-26 00:00:23,603 - pyvisa - DEBUG - RawTCPClient: connecting to socket at (10.66.53.50, 111)
Traceback (most recent call last):
  File "comms.py", line 7, in <module>
    dmm = rm.open_resource('TCPIP::10.66.53.50::inst0:INSTR')
  File "/usr/local/lib/python2.7/dist-packages/PyVISA-1.8.dev0-py2.7.egg/pyvisa/highlevel.py", line 1634, in open_resource
    res.open(access_mode, open_timeout)
  File "/usr/local/lib/python2.7/dist-packages/PyVISA-1.8.dev0-py2.7.egg/pyvisa/resources/resource.py", line 184, in open
    self.session, status = self._resource_manager.open_bare_resource(self._resource_name, access_mode, open_timeout)
  File "/usr/local/lib/python2.7/dist-packages/PyVISA-1.8.dev0-py2.7.egg/pyvisa/highlevel.py", line 1598, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.2.dev2-py2.7.egg/pyvisa-py/highlevel.py", line 198, in open
    sess = cls(session, resource_name, parsed)
  File "/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.2.dev2-py2.7.egg/pyvisa-py/sessions.py", line 184, in __init__
    self.after_parsing()
  File "/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.2.dev2-py2.7.egg/pyvisa-py/tcpip.py", line 45, in after_parsing
    self.interface = vxi11.CoreClient(self.parsed.host_address)
  File "/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.2.dev2-py2.7.egg/pyvisa-py/protocols/vxi11.py", line 197, in __init__
    super(CoreClient, self).__init__(host, DEVICE_CORE_PROG, DEVICE_CORE_VERS)
  File "/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.2.dev2-py2.7.egg/pyvisa-py/protocols/rpc.py", line 595, in __init__
    pmap = TCPPortMapperClient(host)
  File "/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.2.dev2-py2.7.egg/pyvisa-py/protocols/rpc.py", line 573, in __init__
    RawTCPClient.__init__(self, host, PMAP_PROG, PMAP_VERS, PMAP_PORT)
  File "/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.2.dev2-py2.7.egg/pyvisa-py/protocols/rpc.py", line 334, in __init__
    self.connect()
  File "/usr/local/lib/python2.7/dist-packages/PyVISA_py-0.2.dev2-py2.7.egg/pyvisa-py/protocols/rpc.py", line 339, in connect
    self.sock.connect((self.host, self.port))
  File "/usr/lib/python2.7/socket.py", line 224, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 110] Connection timed out
2015-05-26 00:02:31,317 - pyvisa - DEBUG - TCPIP::10.66.53.50::inst0:INSTR - closing
debian@beaglebone:~/Documents/Keithley PyVISA$ 2015-05-26 00:00:23,467 - pyvisa - DEBUG - SerialSession was not imported No module named tools.list_ports.
bash: 2015-05-26: command not found
debian@beaglebone:~/Documents/Keithley PyVISA$ 2015-05-26 00:00:23,472 - pyvisa - DEBUG - USBSession was not imported No module named usb.
bash: 2015-05-26: command not found
debian@beaglebone:~/Documents/Keithley PyVISA$ 2015-05-26 00:00:23,476 - pyvisa - WARNING - (<InterfaceType.usb: 7>, u'INSTR') is already registered in the ResourceManager. Overwriting with unavailable Please install PyUSB to use this resource type.
bash: syntax error near unexpected token `('
debian@beaglebone:~/Documents/Keithley PyVISA$ No module named usb
bash: No: command not found
debian@beaglebone:~/Documents/Keithley PyVISA$ 2015-05-26 00:00:23,478 - pyvisa - WARNING - (<InterfaceType.usb: 7>, u'RAW') is already registered in the ResourceManager. Overwriting with unavailable Please install PyUSB to use this resource type.
bash: syntax error near unexpected token `('
debian@beaglebone:~/Documents/Keithley PyVISA$ No module named usb
bash: No: command not found
debian@beaglebone:~/Documents/Keithley PyVISA$ 2015-05-26 00:00:23,484 - pyvisa - DEBUG - USBRawSession was not imported No module named usb.
bash: 2015-05-26: command not found
debian@beaglebone:~/Documents/Keithley PyVISA$ 2015-05-26 00:00:23,569 - pyvisa - DEBUG - TCPIPSession was correctly imported.
bash: 2015-05-26: command not found
debian@beaglebone:~/Documents/Keithley PyVISA$ 2015-05-26 00:00:23,576 - pyvisa - DEBUG - GPIBSession was not imported No module named gpib.
bash: 2015-05-26: command not found
[... cut ...]
debian@beaglebone:~/Documents/Keithley PyVISA$   File "/usr/lib/python2.7/socket.py", line 224, in meth
bash: File: command not found
debian@beaglebone:~/Documents/Keithley PyVISA$     return getattr(self._sock,name)(*args)
bash: syntax error near unexpected token `('
debian@beaglebone:~/Documents/Keithley PyVISA$ socket.error: [Errno 110] Connection timed out
bash: socket.error:: command not found
debian@beaglebone:~/Documents/Keithley PyVISA$ 2015-05-26 00:02:31,317 - pyvisa - DEBUG - TCPIP::10.66.53.50::inst0:INSTR - closing
bash: 2015-05-26: command not found
```

nktsaini, 2015-05-27:
What can I do now?

hgrecco, 2015-08-25:
Is this still a problem? I develop on Mac I have people using it on Windows and Mac. Feel free to reopen.

closed 2015-08-25 by hgrecco