pyvisa/pyvisa-py #41: Pipe error with controlling Tektronix MSO4034

FusionLi, 2015-05-25:
I got a very strange error in controlling the Tektronix oscilloscope via PyVISA. Right now, I can only run six lines of command after getting it connected with raspberry pi 2. What I mean connected is that after I write the command "inst = rm.open_resource(rm.list_resources()[0])" in python interrupter. After whatever six lines of command, the pipe error appears in the following:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/pyvisa/resources/messagebased.py", line 187, in write
    count = self.write_raw(message.encode(enco))
  File "/usr/local/lib/python2.7/dist-packages/pyvisa/resources/messagebased.py", line 165, in write_raw
    return self.visalib.write(self.session, message)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/highlevel.py", line 281, in write
    return self.sessions[session].write(data)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/usb.py", line 117, in write
    count = self.interface.write(data)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/protocols/usbtmc.py", line 286, in write
    bytes_sent += raw_write(data)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/protocols/usbtmc.py", line 204, in write
    raise ValueError(str(e))
ValueError: [Errno 32] Pipe error

I have to re-enable the USB connection on the signal oscilloscope to get it working again.

hgrecco, 2015-05-25:
Can you try 6 times the same query and post the full program?

FusionLi, 2015-05-26:
Hi, here is the code and error appeared after third query:
>>> import visa
>>> rm = visa.ResourceManager('@py')
>>> inst = rm.open_resource(rm.list_resources()[0])
>>> inst
<'USBInstrument'('USB0::1689::1025::C001713::0::INSTR')>
>>> inst.query('measu:meas1:value?')
u':MEASU:MEAS1:VAL 5.8876E-3\n'
>>> inst.query('measu:meas1:value?')
u''
>>> inst.query('measu:meas1:value?')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/pyvisa/resources/messagebased.py", line 381, in query
    self.write(message)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa/resources/messagebased.py", line 187, in write
    count = self.write_raw(message.encode(enco))
  File "/usr/local/lib/python2.7/dist-packages/pyvisa/resources/messagebased.py", line 165, in write_raw
    return self.visalib.write(self.session, message)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/highlevel.py", line 281, in write
    return self.sessions[session].write(data)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/usb.py", line 117, in write
    count = self.interface.write(data)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/protocols/usbtmc.py", line 286, in write
    bytes_sent += raw_write(data)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/protocols/usbtmc.py", line 204, in write
    raise ValueError(str(e))
ValueError: [Errno 110] Operation timed out

hgrecco, 2015-05-27:
My guess is that there is a bug in the usb library in rasberry py or in the pyusb wrapper. Because it is failing on writing. My suggestion is, just to be sure that it is not failing forever, that you wrap the call into a loop:
```python
import time
for n in range(10):
    try:
        r = inst.query('measu:meas1:value?')
        print(n, r)
    except Exception as e:
        print(n, e)
    time.sleep(1)
```

FusionLi, 2015-05-27:
Thanks for your reply, I tried your code and the output is:
(0, u':MEASU:MEAS1:VAL 13.0210E-3\n')
(1, u'')
(2, ValueError('[Errno 110] Operation timed out',))
(3, ValueError('[Errno 110] Operation timed out',))
(4, ValueError('[Errno 110] Operation timed out',))
(5, ValueError('[Errno 110] Operation timed out',))
(6, ValueError('[Errno 110] Operation timed out',))
(7, ValueError('[Errno 110] Operation timed out',))
(8, ValueError('[Errno 110] Operation timed out',))
(9, ValueError('[Errno 110] Operation timed out',))
Any more insight?

swirhun, 2015-12-07:
I am having the same problem with Agilent equipment. Basically, after connecting the instrument, you can get one query out, but all subsequent commands and queries fail until you power cycle the equipment. This seems to be a major bug in pyvisa-py.

hgrecco, 2015-12-09:
I have not found the same problem in my computer @swirhun and @FusionLi can you dump the output of `python -m visa info`

mbirthelmer, 2015-12-30:
I'm having a similar issue with a Rigol DM3058E. The RS232 interface on the same device seems to work (though it has some issues with timing between commands, which is why I would prefer to use the USB interface). My DP832A PSU from Rigol has no issues. All of this is running on a Raspberry Pi with the pyvisa-py backend.

In my case, it seems to fail 100% of the time immediately. Here is the code snippet I'm using:

```
#!/usr/bin/python

import sys
import visa
from time import sleep

rm = visa.ResourceManager('@py')
dmm = rm.open_resource("USB0::6833::2500::DM3R171400537::0::INSTR")

dmm.timeout = 3000
dmm.query_delay = 1

dmm.write("*rst")
```

Here is the output:
```Python
Traceback (most recent call last):
  File "test_dmm.py", line 15, in <module>
    dmm.write("*rst")
  File "/usr/local/lib/python2.7/dist-packages/pyvisa/resources/messagebased.py", line 207, in write
    count = self.write_raw(message.encode(enco))
  File "/usr/local/lib/python2.7/dist-packages/pyvisa/resources/messagebased.py", line 185, in write_raw
    return self.visalib.write(self.session, message)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/highlevel.py", line 271, in write
    return self.sessions[session].write(data)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/usb.py", line 117, in write
    count = self.interface.write(data)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/protocols/usbtmc.py", line 286, in write
    bytes_sent += raw_write(data)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/protocols/usbtmc.py", line 204, in write
    raise ValueError(str(e))
ValueError: [Errno 32] Pipe error
```

Here is the info output:
```
pi@raspberrypi ~/tm_projects/led_iv $ sudo python -m visa info
Machine Details:
   Platform ID:    Linux-4.1.7-v7+-armv7l-with-debian-8.0
   Processor:

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        2.7.9
   Compiler:       GCC 4.9.2
   Bits:           32bit
   Build:          Mar  8 2015 00:52:26 (#default)
   Unicode:        UCS4

PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.2
      ASRL INSTR: Available via PySerial (2.6)
      TCPIP INSTR: Available
      USB RAW: Available via PyUSB (1.0.0b2). Backend: libusb1
      USB INSTR: Available via PyUSB (1.0.0b2). Backend: libusb1
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named gpib
      TCPIP SOCKET: Available
```

Let me know what I can do help track this down - and thanks for all your work on pyvisa, pyvisa-py etc.

mbirthelmer, 2015-12-30:
In case it's helpful, here is the lsusb output for both Rigol instruments:

```
Bus 001 Device 017: ID 1ab1:0e11 Rigol Technologies
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x1ab1 Rigol Technologies
  idProduct          0x0e11
  bcdDevice            0.02
  iManufacturer           1 Rigol Technologies.
  iProduct                2 DP800 Serials
  iSerial                 3 DP8C161750589
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           39
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       254 Application Specific Interface
      bInterfaceSubClass      3 Test and Measurement
      bInterfaceProtocol      1 TMC
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval              11
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  bNumConfigurations      1
Device Status:     0x0001
  Self Powered

Bus 001 Device 018: ID 1ab1:09c4 Rigol Technologies
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x1ab1 Rigol Technologies
  idProduct          0x09c4
  bcdDevice            1.00
  iManufacturer           1 Rigol Technologies
  iProduct                2 DM3000 SERIES
  iSerial                 3 DM3R171400537
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           39
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xc0
      Self Powered
    MaxPower                2mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       254 Application Specific Interface
      bInterfaceSubClass      3 Test and Measurement
      bInterfaceProtocol      1 TMC
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
Device Status:     0x0001
  Self Powered
```

mbirthelmer, 2015-12-30:
I did a bit more digging and captured the USB stream via usbmon and tcpdump/wireshark . It seems as though in both cases, the packet containing the "*rst" command makes it past the kernel. I also can't see any difference between the two packets - except for the destination (1.17.3 and 1.18.1) they are identical.

After a bit more digging, it turns out that the DM3058E stalls the OUT endpoint, while the DP832A doesn't. Obviously this is caused by the device itself and thus may or may not be related to what the other users are reporting here.

hgrecco, 2016-08-22:
I am closing this for now. Feel free to reopen or open another if necessary.

closed 2016-08-22 by hgrecco