pyvisa/pyvisa-py #43: Struggling to get started with a NI GPIB-ENET/100

djanderson, 2015-06-09:
Hi, I'm on linux and was unable to get the official NI-VISA installed. I'm interested in using pyvisa-py to communicate with an Agilent power meter over an NI GPIB-ENET/100, which I believe is a VISA-compatible device.

NMAP is showing the device is responding:
```bash
$ nmap 192.168.1.232

Starting Nmap 6.40 ( http://nmap.org ) at 2015-06-09 12:33 MDT
Nmap scan report for 192.168.1.232
Host is up (0.024s latency).
Not shown: 998 closed ports
PORT     STATE SERVICE
5000/tcp open  upnp
5003/tcp open  filemaker

Nmap done: 1 IP address (1 host up) scanned in 13.47 seconds
```
However I can't seem to find the correct command string to open the resource.
```bash
enet100 = rm.open_resource('TCPIP::192.168.1.232::inst0::INSTR')
...
socket.error: [Errno 111] Connection refused
```
I could keep stabbing at it *ad nauseam* but I figured I'd check first and see: does pyvisa even work through a converter like the GPIB-ENET/100. If so, what kind of command string should I be testing?

hgrecco, 2015-06-09:
I have never used a GPIB-ENET device but my understanding is that it should provide you a virtual GPIB port on your computer once that your driver are installed, right? If you do not have the drivers for the OS we would need to now what is the protocol between the computer and the ENET so it can rout the message to the correct GPIB device.

hgrecco, 2015-06-09:
In other words, I am not sure thate the ENET100 (which mediates the communication with the Agilent) is a VISA INSTR.

djanderson, 2015-06-09:
I think you're right. I saw that NI-VISA was installed in the rpm they distribute for linux, but I think it's actually the NI-488.2 driver that handles the initial connection. Bummer. Thanks for the info and the great project.

closed 2015-06-09 by djanderson