pyvisa/pyvisa-py #44: serial device timeout being set to 2000 seconds

djanderson, 2015-06-12:
Hi,

In [serial.py](https://github.com/hgrecco/pyvisa-py/blob/master/pyvisa-py/serial.py#L68), you set a default read and write timeout value of "2000".

According to the [pyserial API docs](http://pyserial.sourceforge.net/pyserial_api.html), timeout is in seconds:

    Possible values for the parameter timeout:

    timeout = None: wait forever
    timeout = 0: non-blocking mode (return immediately on read)
    timeout = x: set timeout to x seconds (float allowed)

Did you mean to use a value of "2" instead?

hgrecco, 2015-06-13:
You are right. Would you like to submit a PR. You will also need to change the timeout function as the PyVISA uses milliseconds.

djanderson, 2015-06-15:
Sure, I'll take a look today.

commit 77cc30cd (2015-08-25): Fixed timeout in SerialSession

Close #44
closed 2015-08-25 by hgrecco