pyvisa/pyvisa-py #46: AttributeError: 'Device' object has no attribute 'serial_number' on Linux machine

jonnojohnson, 2015-06-29:
Just installed pyvisa and pyvisa-py today from github master zips on an Ubuntu machine.

Here's my python -m visa info output:
```
Machine Details:
   Platform ID:    Linux-3.13.0-43-generic-x86_64-with-debian-jessie-sid
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /home/mj-ubuntu/anaconda/bin/python
   Version:        2.7.9
   Compiler:       GCC 4.4.7 20120313 (Red Hat 4.4.7-1)
   Bits:           64bit
   Build:          Mar  9 2015 16:20:48 (#default)
   Unicode:        UCS4

PyVISA Version: 1.8.dev0

Backends:
   ni:
      Version: 1.8.dev0 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.2.dev2
      ASRL INSTR: Available via PySerial (2.7)
      TCPIP INSTR: Available 
      USB RAW: Available via PyUSB (1.0.0b1). Backend: libusb1
      USB INSTR: Available via PyUSB (1.0.0b1). Backend: libusb1
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named gpib
      TCPIP SOCKET: Available 
```

Here's the error I get:

```
In [4]: rm.list_resources()
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-4-a14d33fc6d00> in <module>()
----> 1 rm.list_resources()

/home/mj-ubuntu/anaconda/lib/python2.7/site-packages/pyvisa/highlevel.pyc in list_resources(self, query)
   1558         """
   1559 
-> 1560         return self.visalib.list_resources(self.session, query)
   1561 
   1562     def list_resources_info(self, query='?*::INSTR'):

/home/mj-ubuntu/anaconda/lib/python2.7/site-packages/pyvisa-py/highlevel.pyc in list_resources(self, session, query)
    230 
    231         resources = sum([st.list_resources()
--> 232                          for key, st in sessions.Session.iter_valid_session_classes()], [])
    233 
    234         query = query.replace('?*', '.*')

/home/mj-ubuntu/anaconda/lib/python2.7/site-packages/pyvisa-py/usb.pyc in list_resources()
    201                                   manufacturer_id=dev.idVendor,
    202                                   model_code=dev.idProduct,
--> 203                                   serial_number=dev.serial_number,
    204                                   usb_interface_number=intfc))
    205         return out

AttributeError: 'Device' object has no attribute 'serial_number'
```

hgrecco, 2015-07-02:
I think `serial_number` was added in beta2 or rc1 of PyUSB. Until PyUSB 1.0 is released I expect to find a lot of this problems.

hgrecco, 2015-08-25:
I am closing the issue.

closed 2015-08-25 by hgrecco