pyvisa/pyvisa-py #48: Very slow visalib.read() with instruments connected by USB

ayrodriguez, 2015-07-06:
The visalib.read() function takes about 6 seconds to return, whereas with usbtmc it's instantaneous.

pyvisa version 1.8.dev0
pyvisa-py version 0.2.dev2

hgrecco, 2015-08-25:
Sorry but I missed this, can you provide more info.

herm, 2015-10-09:
I'm having the same issue.
The problem is located here:
https://github.com/hgrecco/pyvisa-py/blob/master/pyvisa-py/usb.py#L98

USB communication never checks for the termination character but instead always uses the timeout.

hgrecco, 2015-10-09:
Please submit a PR, I will be happy to review it and merge it during the weekend.

herm, 2015-10-09:
I'm not sure how to fix it correctly. 
``lambda current: term_char_en and current.endswith(chr(term_char))`` 
works, but I think term_char_en should default to True for TMC but I don't know what the best place would be to set it.

herm, 2015-10-27:
This should be fixed now.

hgrecco, 2015-10-27:
@ayrodriguez Please confirm that this is fixed for you and close it! Thanks for the report and Thanks @herm  for the fix!

hgrecco, 2016-08-22:
I am closing this for now. Feel free to reopen or open another if necessary.

closed 2016-08-22 by hgrecco