pyvisa/pyvisa-py #49: Implementation error when using rm = visa.ResourceManager("@py")

bizarri, 2015-07-10:
I apologize in advance If I am in the wrong place but I am not sure if my problem is related to pyvisa or pyvisa-py. Under linux (Ubuntu 15.04), I am trying to play with pyvisa to communicate with a couple of instruments.

I have not problem using the backend '@sim" and when I query the list of resources:
import visa
rm = visa.ResourceManager("@sim")
print(rm.list_resources())

I have the following output:
('GPIB0::4::65535::INSTR', 'ASRL1::INSTR', 'ASRL2::INSTR', 'TCPIP0::localhost:4444::inst0::INSTR', 'GPIB0::8::65535::INSTR', 'USB0::0x1111::0x2222::0x1234::0::INSTR', 'GPIB0::9::65535::INSTR', 'ASRL3::INSTR', 'USB0::0x1111::0x2222::0x2468::0::INSTR', 'TCPIP0::localhost::10001::SOCKET', 'TCPIP0::localhost:3333::inst0::INSTR', 'ASRL4::INSTR', 'TCPIP0::localhost:1111::inst0::INSTR', 'USB0::0x1111::0x2222::0x4444::0::INSTR', 'TCPIP0::localhost:2222::inst0::INSTR', 'USB0::0x1111::0x2222::0x3692::0::INSTR', 'GPIB0::10::65535::INSTR')

However when I try to shift to '@py' I have an error about the implementation:
import visa
rm = visa.ResourceManager("@py")
print(rm.list_resources())

I have the error:
Traceback (most recent call last):
  File "Try-Pyvisa.py", line 3, in <module>
    print(rm.list_resources())
  File "/home/bizarri/bin/miniconda3/lib/python3.4/site-packages/pyvisa/highlevel.py", line 1560, in list_resources
    return self.visalib.list_resources(self.session, query)
  File "/home/bizarri/bin/miniconda3/lib/python3.4/site-packages/pyvisa/highlevel.py", line 669, in list_resources
    raise NotImplementedError
NotImplementedError

I am kind of lost...
Here is the output when I query the visa information (python3 -m visa info)

Python:
   Implementation: CPython
   Executable:     /home/bizarri/bin/miniconda3/bin/python3
   Version:        3.4.3
   Compiler:       GCC 4.4.7 20120313 (Red Hat 4.4.7-1)
   Bits:           64bit
   Build:          Jun  4 2015 15:29:08 (#default)
   Unicode:        UCS4

PyVISA Version: 1.7

Backends:
   ni:
      Version: 1.7 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.1
      USB INSTR: Available via PyUSB (1.0.0b2). Backend: libusb1
      TCPIP INSTR: Available 
      ASRL INSTR: Available via PySerial (N/A)
   sim:
      Version: 0.2
      Spec version: 1.1

Any idea where I have to look for?

Thanks,

Greg

hgrecco, 2015-07-12:
Try installing pyvisa-py from github and let me know if the problem still persists

bizarri, 2015-07-13:
Hi Herman,

Thanks for the help.
It solved my problem.

Greg

closed 2015-07-13 by bizarri