pyvisa/pyvisa-py #52: With logging, lots of UnicodeDecodeError

fcoiffie, 2015-07-31:
When I enable logging at DEBUG level, pyvisa-py generates many UnicodeDecodeError : 
```
Traceback (most recent call last):
  File "/usr/lib/python2.7/logging/__init__.py", line 859, in emit
    msg = self.format(record)
  File "/usr/lib/python2.7/logging/__init__.py", line 732, in format
    return fmt.format(record)
  File "/usr/lib/python2.7/logging/__init__.py", line 471, in format
    record.message = record.getMessage()
  File "/usr/lib/python2.7/logging/__init__.py", line 335, in getMessage
    msg = msg % self.args
UnicodeDecodeError: 'ascii' codec can't decode byte 0xde in position 35: ordinal not in range(128)
Logged from file rpc.py, line 324
```

Same thing at line 296 of rpc.py (in sendrecord()).

hgrecco, 2015-07-31:
I cannot reproduce, so my guess would be that there is a problem due to your locale and the logging module. Because the logging formatter and the message are unicode already. Can you run a simple logging example just to test.

hgrecco, 2015-08-28:
If think this is solved, can you try?

fcoiffie, 2015-09-10:
Unfortunately, it's not solved in 0.2.
But, if I change in rpc.py at 296 : 
```diff
-     logger.debug('Sending record through %s: %s', sock, record)
+     logger.debug('Sending record through %s: %r', sock, record)
```
I have no error.

But, I don't know if the output is really useful : 
```
2015-09-10 14:10:05,006 DEBUG:make_call(240):Make call 11, (7109344, 15, 10000, 136, ':OUTP:STAT ON\r\n'), <bound method Vxi11Packer.pack_device_write_parms of <pyvisa-py.protocols.vxi11.Vxi11Packer instance at 0xb6cc04cc>>, <bound method Vxi11Unpacker.unpack_device_write_resp of <pyvisa-py.protocols.vxi11.Vxi11Unpacker instance at 0xb6cc0a2c>>
2015-09-10 14:10:05,006 DEBUG:sendrecord(296):Sending record through <socket._socketobject object at 0xb6ce7dbc>: "\x00\x00\x00\x12\x00\x00\x00\x00\x00\x00\x00\x02\x00\x06\x07\xaf\x00\x00\x00\x01\x00\x00\x00\x0b\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00lz\xe0\x00\x00\x00\x0f\x00\x00'\x10\x00\x00\x00\x88\x00\x00\x00\x0f:OUTP:STAT ON\r\n\x00"
```

hgrecco, 2015-09-10:
This very low level is usefull to compare implentations, debug
communication problems.

please submit a PR with your change. I will merge it next week.

Hernán

luksan, 2017-05-09:
Please see pull request #97

skrchnavy, 2017-11-24:
#97 accepted, can be this closed?

MatthieuDartiailh, 2017-11-24:
Yes

closed 2017-11-24 by MatthieuDartiailh