pyvisa/pyvisa-py #53: TypeError: can't concat bytes to str (When using gpib)

Denhart, 2015-08-18:
Hi, I'm trying to connect pyvisa and pyvisa-py to a instrument using gpib (NI-USB-B adaptor), and I've manged to open a connection to the instrument, however when I try to query it, pyvisa-py breaks and throws me a traceback.  
```
Python 3.4.3 (default, Mar 25 2015, 17:13:50) 
[GCC 4.9.2 20150304 (prerelease)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import visa
>>> visa.log_to_screen()
>>> rm = visa.ResourceManager('@py')
2015-08-18 11:05:35,438 - pyvisa - DEBUG - SerialSession was not imported No module named 'serial'.
2015-08-18 11:05:35,440 - pyvisa - DEBUG - USBSession and USBRawSession were not imported No module named 'usb'.
2015-08-18 11:05:35,448 - pyvisa - DEBUG - TCPIPSession was not imported cannot import name 'TCPIPSession'.
2015-08-18 11:05:35,451 - pyvisa - DEBUG - GPIBSession was correctly imported.
2015-08-18 11:05:35,451 - pyvisa - DEBUG - Created library wrapper for unset
2015-08-18 11:05:35,452 - pyvisa - DEBUG - Created ResourceManager with session 5468318
>>> rm.list_resources()
('GPIB0::20::INSTR',)
>>> inst = rm.open_resource('GPIB0::20::INSTR')
2015-08-18 11:06:01,818 - pyvisa - DEBUG - GPIB0::20::INSTR - opening ...
2015-08-18 11:06:01,822 - pyvisa - DEBUG - GPIB0::20::INSTR - is open with session 8322128
>>> print(inst.query("*IDN?"))
2015-08-18 11:06:12,064 - pyvisa - DEBUG - GPIB.write b'*IDN?\r\n'
2015-08-18 11:06:12,074 - pyvisa - DEBUG - GPIB0::20::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.4/site-packages/pyvisa/resources/messagebased.py", line 407, in query
    return self.read()
  File "/usr/lib/python3.4/site-packages/pyvisa/resources/messagebased.py", line 332, in read
    message = self.read_raw().decode(enco)
  File "/usr/lib/python3.4/site-packages/pyvisa/resources/messagebased.py", line 306, in read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/usr/lib/python3.4/site-packages/pyvisa-py/highlevel.py", line 253, in read
    return self.sessions[session].read(count)
  File "/usr/lib/python3.4/site-packages/pyvisa-py/gpib.py", line 119, in read
    return self._read(reader, count, checker, False, None, False, gpib.GpibError)
  File "/usr/lib/python3.4/site-packages/pyvisa-py/sessions.py", line 318, in _read
    out += current
TypeError: can't concat bytes to str
```

hgrecco, 2015-08-19:
I do not have GPIB linux installed, but it seems that the read function from GPIB linux returns `str`, not `bytes`. I would be great if you can try this fix. In gpib.py, replace:
```
reader = lambda: self.interface.read(1)
```
by 
```
reader = lambda: self.interface.read(1).encode('ASCII')
```

I have to say that I am surprised that the `write` (which is executed in `query`  before `read`) has not failed.

It would be very nice linux-gpib has a way of returning directly the bytes, but I think is not available. It always interpret the buffer as string. @bessman: I have seen that you have recently contributed a Patch for linux-gpib. Any idea why they are interpreting the output as string, not bytes?

Please try it in Python 2 and 3. Thanks a lot.

Denhart, 2015-08-19:
Hi, your solution works for both python2 and 3, thanks!

closed 2015-08-19 by Denhart

bessman, 2015-08-19:
@hgrecco: This behavior is exactly what my patch to linux-gpib addresses. In the next release of linux-gpib, `read` will return bytes for python3 and string for python2, instead of string for both as is currently the case.

What surprises me is that @Denhart even managed to get as far as they did with python3. Prior to my patch to linux-gpib, pyvisa-py should be failing as early as `import gpib`.

hgrecco, 2015-08-19:
@bessman  Thanks for writing back. I agree, the `encode` solution is just temporary that people should use in the mean time. 

A question about this: I have not found a good, accesible, version number of the linux-gpib. The version given in `setup.py` does not seem to be in use. How about adding that, maybe using `ibvers`?

@Denhart Have you installed linux-gpib from sources?

Denhart, 2015-08-19:
@hgrecco Yes, everything is built from sources. And @bessman surprisingly everything works pretty well in python3 after the encode work-around.

bessman, 2015-08-20:
@Denhart: Which version of linux-gpib are your using, exactly? 3.2.21, 4.0-rc1, or trunk? If it's either of the latter two, the encode workaround not only shouldn't work, but more importantly shouldn't be needed.

Denhart, 2015-08-21:
@bessman I'm using version 3.2.21, so that is probably why :) I'll try to build 4.0 later today and see if it works.