pyvisa/pyvisa-py #56: PyVISA-py does not appear to work with VM of Ubuntu

dillharr, 2015-08-27:
Hi I am not sure if I'm having issues with my VM or PyVISA-py. I installed the latest PyVISA-py on my Ubuntu VM at work so I can start running all my test scripts in Linux instead of Windows, however somethings seem to not work correctly. Below is an output I get after checking to see what devices I have available to connect to.

Python 2.7.6 (default, Jun 22 2015, 17:58:13) 
[GCC 4.8.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import visa
>>> rm = visa.ResourceManager('@py')
>>> rm.list_resources()
(u'ASRL/dev/ttyS31::INSTR', u'ASRL/dev/ttyS30::INSTR', u'ASRL/dev/ttyS29::INSTR', u'ASRL/dev/ttyS28::INSTR', u'ASRL/dev/ttyS27::INSTR', u'ASRL/dev/ttyS26::INSTR', u'ASRL/dev/ttyS25::INSTR', u'ASRL/dev/ttyS24::INSTR', u'ASRL/dev/ttyS23::INSTR', u'ASRL/dev/ttyS22::INSTR', u'ASRL/dev/ttyS21::INSTR', u'ASRL/dev/ttyS20::INSTR', u'ASRL/dev/ttyS19::INSTR', u'ASRL/dev/ttyS18::INSTR', u'ASRL/dev/ttyS17::INSTR', u'ASRL/dev/ttyS16::INSTR', u'ASRL/dev/ttyS15::INSTR', u'ASRL/dev/ttyS14::INSTR', u'ASRL/dev/ttyS13::INSTR', u'ASRL/dev/ttyS12::INSTR', u'ASRL/dev/ttyS11::INSTR', u'ASRL/dev/ttyS10::INSTR', u'ASRL/dev/ttyS9::INSTR', u'ASRL/dev/ttyS8::INSTR', u'ASRL/dev/ttyS7::INSTR', u'ASRL/dev/ttyS6::INSTR', u'ASRL/dev/ttyS5::INSTR', u'ASRL/dev/ttyS4::INSTR', u'ASRL/dev/ttyS3::INSTR', u'ASRL/dev/ttyS2::INSTR', u'ASRL/dev/ttyS1::INSTR', u'ASRL/dev/ttyS0::INSTR')
>>> 
 and here is the output of python -m visa info on my VM. Any help would be much appreciated!! Thanks!

Machine Details:
   Platform ID:    Linux-3.13.0-61-generic-x86_64-with-Ubuntu-14.04-trusty
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        2.7.6
   Compiler:       GCC 4.8.2
   Bits:           64bit
   Build:          Jun 22 2015 17:58:13 (#default)
   Unicode:        UCS4

PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.2
      ASRL INSTR: Available via PySerial (2.7)
      TCPIP INSTR: Available 
      USB RAW: Available via PyUSB (1.0.0b2). Backend: libusb1
      USB INSTR: Available via PyUSB (1.0.0b2). Backend: libusb1
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named gpib
      TCPIP SOCKET: Available

hgrecco, 2015-08-27:
What is not working? I do not see any error messages?

Ps. I am using PyUSB rc1, maybe you can update to it as well.

dillharr, 2015-08-27:
So I don't know if something isn't working. I am just really confused on why I get the list of all the devices that I can supposedly use, when I only have one USB device connected to my computer. I thought it might be because I had an old version of PyUSB on my VM, but after I updated it was the same. I'll try the PyUSB rc1 to see if that helps. I was mainly curious if you have tried PyVISA-py out on a Linux VM, because it's always possible the list of usable device that is presented to me could just be devices the VM thinks it sees.

hgrecco, 2015-08-28:
In general VISA lists all conected devices except in the case of serial
where ir lists all ports. This is what you are seeing. This is true for
Ni-ViSA and pyvisa-py (albeit the first provides a shorter and cleaner
output as it filters the list to show only the real ports). So the output
you are getting is correct.

Regarding the usb device in the VM, it works. But you have to tell the VM
to grab it in the same way that you do it with a usb stick.

hope it helps. let us know.

dillharr, 2015-08-28:
Thank you for your help! I'm pretty sure my issues are stemming from my VM setup after doing some more testing, so I am going to try to tweak some things on my VM. Thanks again!

closed 2015-08-28 by dillharr

hgrecco, 2015-08-29:
Post your results here if you manage to make it work. I am sure other users are trying to achieve the same thing!