pyvisa/pyvisa-py #57: Support for multiple GPIB controllers

benjaminhwilliams, 2015-09-01:
At present, PyVISA-py (via Linux GPIB) only seems to support communication with a single GPIB controller.  Is control via multiple GPIB controllers currently supported in PyVISA-py?  If not, is there a consensus as to whether this would be a desirable feature?

  GPIB devices can be daisy-chained together so that a single controller can communicate with up to fifteen devices.  I would like to have multiple controllers connected (for example, a National Instruments GPIB PCI card and a National Instruments GPIB-USB-HS controller) in order to communicate with more than fifteen devices, if necessary.  When I do this, the output of `rm.list_resources()`, where `rm` is an instance of `visa.ResourceManager('@py')`, lists only the GPIB devices connected to the controller I have designated as the 'system controller' (also known as 'controller-in-charge', 'CiC' or 'master controller') in `/etc/gpib.conf` (the configuration file of Linux GPIB).  This would appear to be because `rm.list_resources()` calls `gpib.listener()`, which requires that the queried interface be the system controller.

  For clarity, here is the output of `python3 -m visa info`.  Note that the Linux GPIB version number is 4.0.0 (as you can see, I am using the latest stable release of PyVISA-py, 0.2, which doesn't report the Linux GPIB version number as the development version does):

```bash
Machine Details:
   Platform ID:    Linux-3.13.0-62-generic-x86_64-with-Ubuntu-14.04-trusty
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.4.0
   Compiler:       GCC 4.8.2
   Bits:           64bit
   Build:          Jun 19 2015 14:20:21 (#default)
   Unicode:        UCS4

PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.2
      TCPIP SOCKET: Available 
      USB RAW: Available via PyUSB (1.0.0b2). Backend: libusb1
      TCPIP INSTR: Available 
      USB INSTR: Available via PyUSB (1.0.0b2). Backend: libusb1
      GPIB INSTR: Available 
      ASRL INSTR: Available via PySerial (2.6)
```

Also, to avoid any confusion, here is `gpib.conf`:

```C
/***********************************************************************
                 GPIB.CONF IEEE488 library config file
                             -------------------

   copyright            : (C) 2002 by Frank Mori Hess
                          (C) 1994 by C.Schroeter
   email                : fmhess@users.sourceforge.net
 ***************************************************************************/
/***************************************************************************
 *    
 *   Syntax:
 *
 *         interface { ... } starts new interface board section
 *         device {...} device configuration
 *
 ***************************************************************************/

/* This section configures the configurable driver characteristics
 * for an interface board, such as board address, and interrupt level.
 * minor = 0 configures /dev/gpib0, minor = 1 configures /dev/gpib1, etc.
 */

interface {
	minor = 0	/* board index, minor = 0 uses /dev/gpib0, minor = 1 uses /dev/gpib1, etc. */
	board_type = "ni_pci"	/* type of interface board being used */
	pad = 0	/* primary address of interface             */
	master = yes	/* interface board is system controller */
}

interface {
	minor = 1	/* board index, minor = 0 uses /dev/gpib0, minor = 1 uses /dev/gpib1, etc. */
	board_type = "ni_usb_b"	/* type of interface board being used */
	pad = 0	/* primary address of interface             */
	master = no	/* interface board is system controller */
}
```

hgrecco, 2015-09-01:
This is indeed a desired feature. I will be happy to merge PR in relation to this. But I think that the problem is not (just) the `listener` call, but rather than the board is hardcoded to 0. We did because it works for most people and we were not sure how to find out the number of boards.

I could envision something like:
```python
def _find_listeners(board):
    """Find GPIB listeners.
    """
    for i in range(31):
        if gpib.listener(board, i) and gpib.ask(board, 1) != i:
            yield i

def _find_boards():
     """Return an iterable with board numbers
     """"
     return (0, 1) # This should be somehow queried from the system
[...]

    @staticmethod
    def list_resources():
        return ['GPIB%d::%d::INSTR' % (board, pad) 
                    for pad in _find_listeners(board) 
                    for board in _find_boards()]
```

Maybe you can try this solution (with hard coded number of boards). If it works we look how to query that from the system or at least we read it from the configuration file.

benjaminhwilliams, 2015-09-02:
Thanks Hernán, I'm glad I'm not just being an idiot.  I'd love to help out, particularly since I'll probably be making a lot of use of the software over the next couple of years and it would be nice to have chipped in, even in a small way.  Unfortunately, I'm currently trying to scrabble together a PhD thesis so I don't have as much free time as I'd like.  I'll look into this but I may consequently be very slow.  If someone beats me to it, so be it!

tpchuckles, 2019-11-21:
I'd love to see this; we have several of those NI GPIB-USB-HS controllers to various pieces of equipment in our lab and they work fine going into the same windows machine, but if you wanted to use them on a linux machine, you're going to need to rerun all the cables to spider/chain them together before the GPIB-usb adapter.
That said, while investigating this, I suspect the bigger issue is the linux-gpib driver itself (lower level issue than pyvisa itself). plugging both GPIB-USB-HSs into my laptop, no matter what I do to gpib.conf and how i run gpib_config, i can still only get one of them connected / mapped to /dev/gpib[n], as seen via dmesg. 

one potential workaround I came across: conceivable if you had separate linux boxes for each GPIB controller (raspberry pis are cheap), you could network them together to still funnel all the data into one place. 
one other thing i found: if the GPIB controllers _are_ of different types (eg USB and PCI, as you mentioned), you may be able to define them as devices rather than interfaces in your config file? (as seen here: http://www.anderswallin.net/2013/11/gpib-on-linux/)