pyvisa/pyvisa-py #60: Problem with connection to Instrument (Spectracom CNT91 Time Interval Counter)

RenierSF1, 2015-10-01:
Hi, I am trying to interface to a Spectracom CNT91 time interval counter using pyvisa-py on a Red Hat Linux machine.

When using print rm.list_resources() it prints the following:
>>>(u'USB0::5355::145::286688::0::INSTR',)

What it should be printing is:
>>>('USB0::0x14EB::0x0091::286688::INSTR')

It seems to be printing the correct Vendor and Product ID in Decimal. (5355 instead of 0x14EB and 145 instead of 0x0091). However, I can open_resource() using either one of the resource names and request the IDN number.
>>>PENDULUM, CNT-91, 286688, V1.30 24 May 2013 08:30
 
But I can only request the IDN number once. Any further requests after the first IDN read and the following error is displayed:

>>> print(inst.query('*IDN?'))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/pyvisa/resources/messagebased.py", line 401, in query
    self.write(message)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa/resources/messagebased.py", line 207, in write
    count = self.write_raw(message.encode(enco))
  File "/usr/local/lib/python2.7/dist-packages/pyvisa/resources/messagebased.py", line 185, in write_raw
    return self.visalib.write(self.session, message)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/highlevel.py", line 271, in write
    return self.sessions[session].write(data)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/usb.py", line 117, in write
    count = self.interface.write(data)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/protocols/usbtmc.py", line 286, in write
    bytes_sent += raw_write(data)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/protocols/usbtmc.py", line 204, in write
    raise ValueError(str(e))
ValueError: [Errno 110] Operation timed out

System info is as follows:
>>>[renier@tfrtest00 ~]$ python -m visa info
Machine Details:
   Platform ID:    Linux-2.6.32-279.el6.x86_64-x86_64-with-redhat-6.3-Santiago
   Processor:      x86_64

>>>Python:
   Implementation: CPython
   Executable:     /usr/local/bin/python
   Version:        2.7.0
   Compiler:       GCC 4.4.6 20120305 (Red Hat 4.4.6-4)
   Bits:           64bit
   Build:          Aug 24 2015 04:16:07 (#r27:82500)
   Unicode:        UCS2

>>>PyVISA Version: 1.8

>>>Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      #1: /usr/local/vxipnp/linux/lib64/libvisa.so:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 15728640
         Spec. Version: 5243904
   py:
      Version: 0.2
      ASRL INSTR:
         Please install PySerial to use this resource type.
         No module named serial
      TCPIP INSTR: Available
      USB RAW: Available via PyUSB (1.0.0rc1). Backend: libusb1
      USB INSTR: Available via PyUSB (1.0.0rc1). Backend: libusb1
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named gpib
      TCPIP SOCKET: Available

We have interfaced to the CNT91 using pyvisa and the NI Visa driver on a windows machine. The code is working and we are able to read back values from the unit. We now need to get this working on the Linux machine. 

Help will be appreciated. 
Thanks.

hgrecco, 2015-10-01:
Both decimal or hex values are accepted. They are used to find out the the resources (which is done by pyusb). 

My guess is that there is something stuck in the buffer of the instrument. It would be good to compare the messages sent by the py and ni backends. Can you take a look?

RenierSF1, 2015-10-02:
Below is the messages when using the NI backend. The problem is that it does not list the usb device, so we are unable to open it.

>>>[renier@tfrtest00 ~]$ python
Python 2.7 (r27:82500, Aug 24 2015, 04:16:07)
[GCC 4.4.6 20120305 (Red Hat 4.4.6-4)] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import visa
>>> rm=visa.ResourceManager('@ni')
>>> print rm.list_resources()
(u'ASRL1::INSTR', u'ASRL2::INSTR', u'ASRL3::INSTR', u'ASRL4::INSTR')
>>> rm=visa.ResourceManager('@py')
>>> print rm.list_resources()
(u'USB0::5355::145::286688::0::INSTR',)

hgrecco, 2015-10-09:
if it works with the NI backend, it should also work with the py backend (or we should improve it to make it happen). But if it does not work with the NI backend (i.e. VISA library), then the problem is different. 
Have you tried the device in another machine, maybe using windows?

RenierSF1, 2015-10-13:
It only works with the NI Backend on my Windows PC. The system we would like to run the scripts on is Red Hat Linux. The NI backend does not work on the Linux machine. 

We have tried a system running Ubuntu and Debian. We are getting similar results detailed in my first post.

hgrecco, 2015-10-14:
If it works on windows then we should make it work in Linux. (that is the whole point of pyvisa-py :smile: ) Please install the latest version from github, run the script again but with logging enabled as described [here](http://pyvisa.readthedocs.org/en/stable/faq.html#i-have-an-error-in-my-program-and-i-am-having-trouble-to-fix-it)

Then post the output here

RenierSF1, 2015-10-14:
Thanks for the feedback. Here is the code with the @py backend:

```python
[root@tfrtest00 ~]# python
Python 2.7 (r27:82500, Aug 24 2015, 04:16:07)
[GCC 4.4.6 20120305 (Red Hat 4.4.6-4)] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import visa
>>> visa.log_to_screen()
>>> rm = visa.ResourceManager('@py')
2015-10-13 13:47:55,977 - pyvisa - DEBUG - SerialSession was not imported No module named serial.
2015-10-13 13:47:56,002 - pyvisa - DEBUG - USBSession and USBRawSession were correctly imported.
2015-10-13 13:47:56,005 - pyvisa - DEBUG - TCPIPSession was not imported cannot import name TCPIPSession.
2015-10-13 13:47:56,005 - pyvisa - DEBUG - GPIBSession was not imported No module named gpib.
2015-10-13 13:47:56,005 - pyvisa - DEBUG - Created library wrapper for unset
2015-10-13 13:47:56,006 - pyvisa - DEBUG - Created ResourceManager with session 8484015
>>> print rm.list_resources()
(u'USB0::5355::145::286688::0::INSTR',)
>>> inst = rm.open_resource('USB0::5355::145::286688::0::INSTR')
2015-10-13 13:48:37,438 - pyvisa - DEBUG - USB0::5355::145::286688::0::INSTR - opening ...
2015-10-13 13:48:42,618 - pyvisa - DEBUG - USB0::5355::145::286688::0::INSTR - is open with session 9901799
>>> print inst.query('*IDN?')
2015-10-13 13:50:41,878 - pyvisa - DEBUG - USB0::5355::145::286688::0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
PENDULUM, CNT-91, 286688, V1.30 24 May 2013 08:30

>>> print inst.query('*IDN?')
2015-10-13 13:50:50,861 - pyvisa - DEBUG - USB0::5355::145::286688::0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)

>>> print inst.query('*IDN?')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 401, in query
    self.write(message)
  File "/usr/local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 207, in write
    count = self.write_raw(message.encode(enco))
  File "/usr/local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 185, in write_raw
    return self.visalib.write(self.session, message)
  File "/usr/local/lib/python2.7/site-packages/pyvisa-py/highlevel.py", line 271, in write
    return self.sessions[session].write(data)
  File "/usr/local/lib/python2.7/site-packages/pyvisa-py/usb.py", line 117, in write
    count = self.interface.write(data)
  File "/usr/local/lib/python2.7/site-packages/pyvisa-py/protocols/usbtmc.py", line 286, in write
    bytes_sent += raw_write(data)
  File "/usr/local/lib/python2.7/site-packages/pyvisa-py/protocols/usbtmc.py", line 204, in write
    raise ValueError(str(e))
ValueError: [Errno 110] Operation timed out
```

Also adding code for the @ni backend. This does not even read back the device.

```python
[root@tfrtest00 ~]# python
Python 2.7 (r27:82500, Aug 24 2015, 04:16:07)
[GCC 4.4.6 20120305 (Red Hat 4.4.6-4)] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import visa
>>> visa.log_to_screen()
>>> rm=visa.ResourceManager('@ni')
2015-10-13 13:56:46,046 - pyvisa - DEBUG - No user defined library files
2015-10-13 13:56:46,114 - pyvisa - DEBUG - Automatically found library files: ['/usr/local/vxi         pnp/linux/lib64/libvisa.so']
2015-10-13 13:56:46,119 - pyvisa - DEBUG - Library signatures: 90 ok, 0 failed
2015-10-13 13:56:46,120 - pyvisa - DEBUG - Created library wrapper for /usr/local/vxipnp/linux         /lib64/libvisa.so
2015-10-13 13:56:46,125 - pyvisa - DEBUG - viOpenDefaultRM('<ViObject object at 0x103d7a0>',)          -> 0
2015-10-13 13:56:46,125 - pyvisa - DEBUG - Created ResourceManager with session 4097
>>> print rm.list_resources()
2015-10-13 13:57:03,255 - pyvisa - DEBUG - viFindRsrc(4097L, u'?*::INSTR', '<ViObject object a         t 0x103d7a0>', 'c_uint(3L)', <ctypes.c_char_Array_256 object at 0x103d710>) -> 0
2015-10-13 13:57:03,255 - pyvisa - DEBUG - viFindNext(<ViObject object at 0x103d7a0>, <ctypes.         c_char_Array_256 object at 0x103d710>) -> 0
2015-10-13 13:57:03,256 - pyvisa - DEBUG - viFindNext(<ViObject object at 0x103d7a0>, <ctypes.         c_char_Array_256 object at 0x103d710>) -> 0
2015-10-13 13:57:03,256 - pyvisa - DEBUG - viClose(<ViObject object at 0x103d7a0>,) -> 0
(u'ASRL1::INSTR', u'ASRL2::INSTR', u'ASRL3::INSTR')
[... 2 lines cut ...]
2015-10-13 13:58:27,714 - pyvisa - DEBUG - USB0::0x14EB::0x0091::286688::INSTR - opening ...
2015-10-13 13:58:27,715 - pyvisa - DEBUG - viOpen(4097L, 'USB0::0x14EB::0x0091::286688::INSTR', <AccessModes.no_lock: 0>, 0, '<ViObject object at 0x103da70>') -> -1073807246
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/site-packages/pyvisa/highlevel.py", line 1644, in open_resource
    res.open(access_mode, open_timeout)
  File "/usr/local/lib/python2.7/site-packages/pyvisa/resources/resource.py", line 203, in open
    self.session, status = self._resource_manager.open_bare_resource(self._resource_name, access_mode, open_timeout)
  File "/usr/local/lib/python2.7/site-packages/pyvisa/highlevel.py", line 1601, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "/usr/local/lib/python2.7/site-packages/pyvisa/ctwrapper/functions.py", line 1211, in open
    ret = library.viOpen(session, resource_name, access_mode, open_timeout, byref(out_session))
[... cut ...]
2015-10-13 13:58:43,662 - pyvisa - DEBUG - USBSession and USBRawSession were correctly imported.
2015-10-13 13:58:43,666 - pyvisa - DEBUG - TCPIPSession was not imported cannot import name TCPIPSession.
2015-10-13 13:58:43,666 - pyvisa - DEBUG - GPIBSession was not imported No module named gpib.
2015-10-13 13:58:43,667 - pyvisa - DEBUG - Created library wrapper for unset
2015-10-13 13:58:43,667 - pyvisa - DEBUG - Created ResourceManager with session 5359076
>>> print rm.list_resources()
(u'USB0::5355::145::286688::0::INSTR',)
```

hgrecco, 2015-10-14:
The NI backend will not work in linux (but it would be nice to have the output of the windows machine running the same script, though). It looks as if the system is chocking. Have you configured correctly the read_termination?

RenierSF1, 2015-10-14:
The read_termination is correctly configured, no problems with reading back from the instrument on windows machine. The Windows machine does however use the @ni backend and not the @py backend. Different instrument on the windows machine - When running "python -m visa info" on windows it shows the the @py backend cannot be instantiated. This is with the development version of pyvisa and with pyusb installed.

hgrecco, 2015-10-14:
This is one of the most common problems. I think we should provide a better way to debug it and later  then write something about this to the pyvisa docs. 

As the first step on this path, I have just pushed to pyvisa (not pyvisa-py) a small but (hopefully) useful change: the debug log will now show the content of the buffer. Please update your pyvisa to the version on github run your script again and paste the log.

RenierSF1, 2015-10-15:
Updated updated pyvisa to the developer version on both Red Hat and Windows machines.

Red Hat Log:

```python
>>> import visa
>>> visa.log_to_screen()
>>> rm=visa.ResourceManager()
2015-10-14 14:50:38,871 - pyvisa - DEBUG - No user defined library files
2015-10-14 14:50:38,937 - pyvisa - DEBUG - Automatically found library files: ['/usr/local/vxipnp/linux/lib64/libvisa.so']
2015-10-14 14:50:38,949 - pyvisa - DEBUG - Library signatures: 90 ok, 0 failed
2015-10-14 14:50:38,949 - pyvisa - DEBUG - Created library wrapper for /usr/local/vxipnp/linux/lib64/libvisa.so
2015-10-14 14:50:38,953 - pyvisa - DEBUG - viOpenDefaultRM('<ViObject object at 0x1c917a0>',) -> 0
2015-10-14 14:50:38,954 - pyvisa - DEBUG - Created ResourceManager with session 4097
>>> print rm.list_resources()
2015-10-14 14:51:00,313 - pyvisa - DEBUG - viFindRsrc(4097L, u'?*::INSTR', '<ViObject object at 0x1c917a0>', 'c_uint(3L)', <ctypes.c_char_Array_256 object at 0x1c91710>) -> 0
2015-10-14 14:51:00,313 - pyvisa - DEBUG - viFindNext(<ViObject object at 0x1c917a0>, <ctypes.c_char_Array_256 object at 0x1c91710>) -> 0
2015-10-14 14:51:00,313 - pyvisa - DEBUG - viFindNext(<ViObject object at 0x1c917a0>, <ctypes.c_char_Array_256 object at 0x1c91710>) -> 0
2015-10-14 14:51:00,313 - pyvisa - DEBUG - viClose(<ViObject object at 0x1c917a0>,) -> 0
(u'ASRL1::INSTR', u'ASRL2::INSTR', u'ASRL3::INSTR')
>>> exit()
[root@tfrtest00 ~]# clear
[root@tfrtest00 ~]# python
Python 2.7 (r27:82500, Aug 24 2015, 04:16:07)
[GCC 4.4.6 20120305 (Red Hat 4.4.6-4)] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import visa
>>> visa.log_to_screen()
>>> rm=visa.ResourceManager("@py")
2015-10-14 14:52:42,391 - pyvisa - DEBUG - SerialSession was not imported No module named serial.
2015-10-14 14:52:42,409 - pyvisa - DEBUG - USBSession and USBRawSession were correctly imported.
2015-10-14 14:52:42,412 - pyvisa - DEBUG - TCPIPSession was not imported cannot import name TCPIPSession.
2015-10-14 14:52:42,412 - pyvisa - DEBUG - GPIBSession was not imported No module named gpib.
2015-10-14 14:52:42,412 - pyvisa - DEBUG - Created library wrapper for unset
2015-10-14 14:52:42,412 - pyvisa - DEBUG - Created ResourceManager with session 5000308
>>> print rm.list_resources()
(u'USB0::5355::145::286688::0::INSTR',)
>>> inst=rm.open_resource('USB0::5355::145::286688::0::INSTR')
2015-10-14 14:53:30,710 - pyvisa - DEBUG - USB0::5355::145::286688::0::INSTR - opening ...
2015-10-14 14:53:35,888 - pyvisa - DEBUG - USB0::5355::145::286688::0::INSTR - is open with session 5570462
>>> print inst.query("*IDN?")
2015-10-14 14:54:03,106 - pyvisa - DEBUG - USB0::5355::145::286688::0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
PENDULUM, CNT-91, 286688, V1.30 24 May 2013 08:30

>>> print inst.query("*IDN?")
2015-10-14 14:54:12,105 - pyvisa - DEBUG - USB0::5355::145::286688::0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)

>>> print inst.query("*IDN?")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 401, in query
    self.write(message)
  File "/usr/local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 207, in write
    count = self.write_raw(message.encode(enco))
  File "/usr/local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 185, in write_raw
    return self.visalib.write(self.session, message)
  File "/usr/local/lib/python2.7/site-packages/pyvisa-py/highlevel.py", line 271, in write
    return self.sessions[session].write(data)
  File "/usr/local/lib/python2.7/site-packages/pyvisa-py/usb.py", line 117, in write
    count = self.interface.write(data)
  File "/usr/local/lib/python2.7/site-packages/pyvisa-py/protocols/usbtmc.py", line 286, in write
    bytes_sent += raw_write(data)
  File "/usr/local/lib/python2.7/site-packages/pyvisa-py/protocols/usbtmc.py", line 204, in write
    raise ValueError(str(e))
ValueError: [Errno 110] Operation timed out
```

Windows Machine Log:
```python
>>> import visa
>>> visa.log_to_screen()
>>> rm=visa.ResourceManager("@py")
2015-10-15 08:08:47,470 - pyvisa - DEBUG - SerialSession was not imported No module named serial.

Traceback (most recent call last):
  File "<pyshell#2>", line 1, in <module>
    rm=visa.ResourceManager("@py")
  File "C:\Python27\lib\site-packages\pyvisa\highlevel.py", line 1488, in __new__
    visa_library = open_visa_library(visa_library)
  File "C:\Python27\lib\site-packages\pyvisa\highlevel.py", line 1457, in open_visa_library
    cls = get_wrapper_class(wrapper)
  File "C:\Python27\lib\site-packages\pyvisa\highlevel.py", line 1437, in get_wrapper_class
    pkg = __import__(pkgname)
  File "C:\Python27\lib\site-packages\pyvisa-py\__init__.py", line 26, in <module>
    from .highlevel import PyVisaLibrary
  File "C:\Python27\lib\site-packages\pyvisa-py\highlevel.py", line 26, in <module>
    class PyVisaLibrary(highlevel.VisaLibraryBase):
  File "C:\Python27\lib\site-packages\pyvisa-py\highlevel.py", line 49, in PyVisaLibrary
    from .usb import USBSession, USBRawSession
  File "C:\Python27\lib\site-packages\pyvisa-py\usb.py", line 31, in <module>
    _ = usb.core.find()
  File "C:\Python27\lib\site-packages\usb\core.py", line 1271, in find
    raise NoBackendError('No backend available')
NoBackendError: No backend available
>>> 
```

hgrecco, 2015-10-15:
The timeout occurs in the first read. Wrap the second query in a try except like this:
```python
try:
     print(inst.query("*IDN?"))
except:
     print(inst.visalib.read(inst.session, inst.bytes_in_buffer))
```

RenierSF1, 2015-10-15:
So if I understand you correctly, the code will be as follows:
```python

import visa
visa.log_to_screen()
rm = visa.ResourceManager("@py")
print(rm.list_resources())
inst = rm.open_resource('USB0::0x14EB::0x0091::286688::INSTR')

print(inst.query("*IDN?"))#first IDN Read

try:
        print(inst.query("*IDN?"))
        print(inst.visalib.read(inst.session, inst.bytes_in_buffer))
        print("IDN READ FAIL")

except:
        print(inst.visalib.read(inst.session, inst.bytes_in_buffer))
        print("IDN READ FAIL")

```
The result: 

```python
[root@tfrtest00 Downloads]# python TICread.py
2015-10-14 21:49:40,279 - pyvisa - DEBUG - SerialSession was not imported No module named serial.
2015-10-14 21:49:40,297 - pyvisa - DEBUG - USBSession and USBRawSession were correctly imported.
2015-10-14 21:49:40,300 - pyvisa - DEBUG - TCPIPSession was not imported cannot import name TCPIPSession.
2015-10-14 21:49:40,300 - pyvisa - DEBUG - GPIBSession was not imported No module named gpib.
2015-10-14 21:49:40,300 - pyvisa - DEBUG - Created library wrapper for unset
2015-10-14 21:49:40,300 - pyvisa - DEBUG - Created ResourceManager with session 6451763
(u'USB0::5355::145::286688::0::INSTR',)
2015-10-14 21:49:40,326 - pyvisa - DEBUG - USB0::0x14EB::0x0091::286688::INSTR - opening ...
2015-10-14 21:49:45,505 - pyvisa - DEBUG - USB0::0x14EB::0x0091::286688::INSTR - is open with session 2539739
2015-10-14 21:49:45,506 - pyvisa - DEBUG - USB0::0x14EB::0x0091::286688::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
PENDULUM, CNT-91, 286688, V1.30 24 May 2013 08:30

2015-10-14 21:49:46,520 - pyvisa - DEBUG - USB0::0x14EB::0x0091::286688::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)

Traceback (most recent call last):
  File "TICread.py", line 19, in <module>
    print(inst.visalib.read(inst.session, inst.bytes_in_buffer))
AttributeError: 'USBInstrument' object has no attribute 'bytes_in_buffer'
2015-10-14 21:49:47,534 - pyvisa - DEBUG - USB0::0x14EB::0x0091::286688::INSTR - closing
2015-10-14 21:49:47,534 - pyvisa - DEBUG - USB0::0x14EB::0x0091::286688::INSTR - is closed
```

I added the line "print(inst.visalib.read(inst.session, inst.bytes_in_buffer))" in the "try:" part because otherwise it would just exit at the end. 

I notice something that might be useful. So if i do an "inst =rm.open_resource('USB0::0x14EB::0x0091::286688::INSTR')" before every read or write instruction that I do to/from the device, everything seems to work without failure. The problem is that their is about a 6 second delay every time that I open the device. 

```python
import time
import visa
visa.log_to_screen()

rm = visa.ResourceManager("@py")

print(rm.list_resources())

inst = rm.open_resource('USB0::0x14EB::0x0091::286688::INSTR')

try:
        inst=rm.open_resource('USB0::0x14EB::0x0091::286688::INSTR')
        print(inst.query("*IDN?"))
        time.sleep(2)
        inst=rm.open_resource('USB0::0x14EB::0x0091::286688::INSTR')
        inst.write(':MEAS:TINT? (@2),(@1)')# set measurement interval to standard 1 to 2
        inst=rm.open_resource('USB0::0x14EB::0x0091::286688::INSTR')
        inst.write(':ROSC:SOUR EXT') #set reference oscillator to external
[... 42 lines cut ...]
2015-10-14 21:36:11,192 - pyvisa - DEBUG - USB0::0x14EB::0x0091::286688::INSTR - opening ...
2015-10-14 21:36:16,413 - pyvisa - DEBUG - USB0::0x14EB::0x0091::286688::INSTR - is open with session 4939854
2015-10-14 21:36:16,413 - pyvisa - DEBUG - USB0::0x14EB::0x0091::286688::INSTR - closing
2015-10-14 21:36:16,413 - pyvisa - DEBUG - USB0::0x14EB::0x0091::286688::INSTR - is closed
2015-10-14 21:36:16,414 - pyvisa - DEBUG - USB0::0x14EB::0x0091::286688::INSTR - opening ...
2015-10-14 21:36:21,591 - pyvisa - DEBUG - USB0::0x14EB::0x0091::286688::INSTR - is open with session 5804601
2015-10-14 21:36:21,592 - pyvisa - DEBUG - USB0::0x14EB::0x0091::286688::INSTR - closing
2015-10-14 21:36:21,592 - pyvisa - DEBUG - USB0::0x14EB::0x0091::286688::INSTR - is closed
2015-10-14 21:36:21,593 - pyvisa - DEBUG - USB0::0x14EB::0x0091::286688::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
21:36:23 10/14/15 PDT - +2.4151470160E-01

2015-10-14 21:36:23,129 - pyvisa - DEBUG - USB0::0x14EB::0x0091::286688::INSTR - opening ...
2015-10-14 21:36:28,305 - pyvisa - DEBUG - USB0::0x14EB::0x0091::286688::INSTR - is open with session 5967596
2015-10-14 21:36:28,306 - pyvisa - DEBUG - USB0::0x14EB::0x0091::286688::INSTR - closing
2015-10-14 21:36:28,306 - pyvisa - DEBUG - USB0::0x14EB::0x0091::286688::INSTR - is closed
2015-10-14 21:36:28,307 - pyvisa - DEBUG - USB0::0x14EB::0x0091::286688::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
21:36:30 10/14/15 PDT - +2.3305852494E-01

2015-10-14 21:36:30,121 - pyvisa - DEBUG - USB0::0x14EB::0x0091::286688::INSTR - opening ...
2015-10-14 21:36:35,297 - pyvisa - DEBUG - USB0::0x14EB::0x0091::286688::INSTR - is open with session 9662866
2015-10-14 21:36:35,297 - pyvisa - DEBUG - USB0::0x14EB::0x0091::286688::INSTR - closing
2015-10-14 21:36:35,297 - pyvisa - DEBUG - USB0::0x14EB::0x0091::286688::INSTR - is closed
2015-10-14 21:36:35,298 - pyvisa - DEBUG - USB0::0x14EB::0x0091::286688::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
21:36:37 10/14/15 PDT - +2.3655059951E-01
```

hgrecco, 2015-10-15:
Your observation about reopening the port supports the idea that it is chocking somewhere. Reopening clears the buffer. 

Two things:
1. I just realized that bytes in buffer is not implemented yet for PyVISA-py (my bad)
2. What it does not make sense to me is `I added the line "print(inst.visalib.read(inst.session, inst.bytes_in_buffer))" in the "try:" part because otherwise it would just exit at the end.``

The second query was failing before, so it should fail now leading to the except. Right?

It would be great if you can do a step by step debugging. I would like to know what is the inner state of the USBSession object before it fails.

RenierSF1, 2015-10-16:
Do you perhaps have some sample code on how to do the step by step debugging and checking the inner state of the USBsession?

I looked at the Pyusb "What's Wrong" section. Tried using the PYUSB_DEBUG=debug and PYUSB_LOG_FILENAME=pyusb.log but I cannot get it to log to the file when running the code.

https://github.com/walac/pyusb/blob/master/docs/tutorial.rst

hgrecco, 2015-10-16:
I use PyCharm. It is a Python IDE with a free version an very nice step by step debugger.

buckko, 2015-12-21:
I too have been having the same issue.  Hosed on a Linux using py as backend (libusb1), able to get working on a Windows (using ni as backend).  Once hosed, its hosed on both systems...however i can get it back to working on Windows by asserting "<inst handle>.clear()", but, when trying this command on Linux i get the exception:  "NotImplementedError".  How does one clear a resource?

buckko, 2015-12-23:
I'm assuming no one is able to talk to instruments via pyvisa-py...tis a shame, could have been really useful

doomroom, 2017-11-01:
Faced such a mistake.
Set the timeout 0. This solved the problem.

my_instrument = rm.open_resource ("USB0 :: 0000 :: 000 :: 000000 :: 0 :: INSTR")
my_instrument.timeout = 0
@buckko  @hgrecco