pyvisa/pyvisa-py #61: Deadlock on Read with TCPIP Socket

mluthi, 2015-10-08:
In debugging a problem with reading data from a device using the TCPIP socket connection I discovered a problem in the read method of the TCPIPSocketSession class (ln. 391ff in tcpip.py; version 0.3.dev0): the select function is called without a timeout and therefore may block indefinitely in case the VI_ATTR_TERMCHAR_EN attribute is not set. In my case the while loop is executed once in the beginning to receive the data and then blocks forever the second time it gets to the select call.

My environment: Windows 7 Enterprise, SP1, Python 3.4.1

Best Regards,
Matthias

petervago, 2016-09-02:
I experienced the same. (Ubuntu 14.04. 64bit)
The following solved the issue for me:

Replaced the following code:
/usr/local/lib/python3.4/dist-packages/pyvisa-py/tcpip.py line 391

`select.select([self.interface], [] , [])`

with this:

`try:
 select.select([self.interface], [] , [] , 0.01) # timeout 0.01sec
 last = read_fun(chunk_length)
except:
 last = ""`

skrchnavy, 2017-10-26:
in #107 there shall be implemented fix for this issue. Tested on windows only for now.

skrchnavy, 2017-11-24:
#107 merged, #113 improves implementation. When merged or closed, this issue can be also closed as resolved.

MatthieuDartiailh, 2017-11-27:
Closed by #113

closed 2017-11-27 by MatthieuDartiailh