pyvisa/pyvisa-py #69: UnicodeDecodeError with USB INSTR but not with TCPIP INSTR

in3otd, 2015-11-03:
Hello,
when trying to read the trace binary data from a Rigol scope using USB I get often errors like `UnicodeDecodeError: 'ascii' codec can't decode byte 0x80 in position 12: ordinal not in range(128)`. Sometimes I get this error even when the instrument should have sent a real ASCII string, but it might be actually sending invalid bytes due to scope FW bugs. Anyway, I think the `UnicodeDecodeError` should not happen in any case.
When I connect using Ethernet this error does not happen.

With the log enabled this is what I get:
```
2015-11-03 19:35:22,312 - pyvisa - DEBUG - SerialSession was correctly imported.
2015-11-03 19:35:22,342 - pyvisa - DEBUG - USBSession and USBRawSession were correctly imported.
2015-11-03 19:35:22,344 - pyvisa - DEBUG - TCPIPSession was correctly imported.
2015-11-03 19:35:22,345 - pyvisa - DEBUG - GPIBSession was correctly imported.
2015-11-03 19:35:22,345 - pyvisa - DEBUG - Created library wrapper for unset
2015-11-03 19:35:22,345 - pyvisa - DEBUG - Created ResourceManager with session 6034227
2015-11-03 19:35:22,345 - pyvisa - DEBUG - USB0::6833::1230::DS1ZC162943448::0::INSTR - opening ...
2015-11-03 19:35:22,508 - pyvisa - DEBUG - USB0::6833::1230::DS1ZC162943448::0::INSTR - is open with session 9064266
24000000
2015-11-03 19:35:24,013 - pyvisa - DEBUG - USB0::6833::1230::DS1ZC162943448::0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
act_md= u'24000000\n\x7f' md= 24000000
Memory depth set to 24000000
0
2015-11-03 19:35:25,016 - pyvisa - DEBUG - USB0::6833::1230::DS1ZC162943448::0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
Traceback (most recent call last):
  File "mytest.py", line 10, in <module>
    cd = s.read()
  File "/home/username/mylib.py", line 67, in read
    d = self.i.read_raw()
  File "/usr/lib/python2.7/site-packages/PyVISA-1.8-py2.7.egg/pyvisa/resources/messagebased.py", line 306, in read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/usr/lib/python2.7/site-packages/PyVISA_py-0.3.dev0-py2.7.egg/pyvisa-py/highlevel.py", line 253, in read
    return self.sessions[session].read(count)
  File "/usr/lib/python2.7/site-packages/PyVISA_py-0.3.dev0-py2.7.egg/pyvisa-py/usb.py", line 102, in read
    usb.USBError)
  File "/usr/lib/python2.7/site-packages/PyVISA_py-0.3.dev0-py2.7.egg/pyvisa-py/sessions.py", line 323, in _read
    current = reader().encode('ascii')
UnicodeDecodeError: 'ascii' codec can't decode byte 0x80 in position 12: ordinal not in range(128)
2015-11-03 19:35:25,222 - pyvisa - DEBUG - USB0::6833::1230::DS1ZC162943448::0::INSTR - closing
2015-11-03 19:35:25,222 - pyvisa - DEBUG - USB0::6833::1230::DS1ZC162943448::0::INSTR - is closed
```
I have also tried to use `query_binary_values()` instead of `read_raw()` but I get the same error
```
Traceback (most recent call last):
  File "mytest.py", line 10, in <module>
    cd = s.read()
  File "/home/username/mylib.py", line 68, in read
    d = self.i.query_binary_values("WAV:DATA?", datatype='b', is_big_endian=True)
  File "/usr/lib/python2.7/site-packages/PyVISA-1.8-py2.7.egg/pyvisa/resources/messagebased.py", line 481, in query_binary_values
    block = self.read_raw()
  File "/usr/lib/python2.7/site-packages/PyVISA-1.8-py2.7.egg/pyvisa/resources/messagebased.py", line 306, in read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/usr/lib/python2.7/site-packages/PyVISA_py-0.3.dev0-py2.7.egg/pyvisa-py/highlevel.py", line 253, in read
    return self.sessions[session].read(count)
  File "/usr/lib/python2.7/site-packages/PyVISA_py-0.3.dev0-py2.7.egg/pyvisa-py/usb.py", line 102, in read
    usb.USBError)
  File "/usr/lib/python2.7/site-packages/PyVISA_py-0.3.dev0-py2.7.egg/pyvisa-py/sessions.py", line 323, in _read
    current = reader().encode('ascii')
UnicodeDecodeError: 'ascii' codec can't decode byte 0x80 in position 12: ordinal not in range(128)
```
System information:
```
$ python -m visa info
Machine Details:
   Platform ID:    Linux-3.17.1-smp-i686-x86_64
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        2.7.5
   Compiler:       GCC 4.8.0
   Bits:           32bit
   Build:          May 29 2013 03:26:28 (#default)
   Unicode:        UCS2

PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.3.dev0
      ASRL INSTR: Available via PySerial (2.7)
      TCPIP INSTR: Available 
      USB RAW: Available via PyUSB (1.0.0b2). Backend: libusb1
      USB INSTR: Available via PyUSB (1.0.0b2). Backend: libusb1
      GPIB INSTR: Available via Linux GPIB (< 4.0)
      TCPIP SOCKET: Available 
```

skrchnavy, 2017-11-24:
related to PRs #66 / #70 / #86

MatthieuDartiailh, 2017-12-07:
Closed by #119

closed 2017-12-07 by MatthieuDartiailh