pyvisa/pyvisa-py #71: Issue with Keysight USBTMC devices

daniele-nicolodi, 2015-11-30:
I have troubles using pyvisa-py to communicate with Keysight USBTMC devices on Linux:
```
Python 2.7.6 (default, Jun 22 2015, 17:58:13) 
[GCC 4.8.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import visa; rm = visa.ResourceManager('@py'); l = rm.list_resources(); d = rm.open_resource(l[0])
>>> d.ask('*IDN?')
u'Agilent Technologies,34972A,MY49023315,1.17-1.12-02-02\n'
>>> d.ask('*IDN?')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/manip/src/test-pyvisa-py/local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 407, in query
    return self.read()
  File "/home/manip/src/test-pyvisa-py/local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 332, in read
    message = self.read_raw().decode(enco)
  File "/home/manip/src/test-pyvisa-py/local/lib/python2.7/site-packages/pyvisa/resources/messagebased.py", line 306, in read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/home/manip/src/test-pyvisa-py/local/lib/python2.7/site-packages/pyvisa-py/highlevel.py", line 253, in read
    return self.sessions[session].read(count)
  File "/home/manip/src/test-pyvisa-py/local/lib/python2.7/site-packages/pyvisa-py/usb.py", line 102, in read
    usb.USBError)
  File "/home/manip/src/test-pyvisa-py/local/lib/python2.7/site-packages/pyvisa-py/sessions.py", line 313, in _read
    current = reader()
  File "/home/manip/src/test-pyvisa-py/local/lib/python2.7/site-packages/pyvisa-py/usb.py", line 96, in <lambda>
    return self._read(lambda: self.interface.read(1),
  File "/home/manip/src/test-pyvisa-py/local/lib/python2.7/site-packages/pyvisa-py/protocols/usbtmc.py", line 307, in read
    raw_write(req)
  File "/home/manip/src/test-pyvisa-py/local/lib/python2.7/site-packages/pyvisa-py/protocols/usbtmc.py", line 204, in write
    raise ValueError(str(e))
ValueError: [Errno 110] Operation timed out
>>> 
```
as you can see, the second '*IDN?' command fails. I see some associated kernel messages:

```
[  716.703289] usb 1-10: new high-speed USB device number 11 using xhci_hcd
[  717.254758] usb 1-10: New USB device found, idVendor=0957, idProduct=2007
[  717.254768] usb 1-10: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  717.254772] usb 1-10: Product: 34972A
[  717.254777] usb 1-10: Manufacturer: Keysight Technologies
[  717.254780] usb 1-10: SerialNumber: MY49023315
[  717.255088] usb 1-10: ep 0x2 - rounding interval to 128 microframes, ep desc says 255 microframes
[  717.255098] usb 1-10: ep 0x81 - rounding interval to 128 microframes, ep desc says 255 microframes
[  721.411616] usb 1-10: ep 0x2 - rounding interval to 128 microframes, ep desc says 255 microframes
[  721.411621] usb 1-10: ep 0x81 - rounding interval to 128 microframes, ep desc says 255 microframes
[  721.565855] usb 1-10: reset high-speed USB device number 11 using xhci_hcd
[  721.731155] usb 1-10: ep 0x2 - rounding interval to 128 microframes, ep desc says 255 microframes
[  721.731169] usb 1-10: ep 0x81 - rounding interval to 128 microframes, ep desc says 255 microframes
[  728.383831] xhci_hcd 0000:00:14.0: WARN Event TRB for slot 10 ep 2 with no TDs queued?
[  730.379499] xhci_hcd 0000:00:14.0: WARN Event TRB for slot 10 ep 2 with no TDs queued?
```
where the last two warnings correspond to the two '*IDN?' commands I issued. After that all attemps to communicate with the USBTMC device (read or write) fail until I reboot the device.

I'm able to communicate correctly with the device using the Linux `usbtmc

daniele-nicolodi, 2015-11-30:
python-usbtmc 0.6 https://github.com/python-ivi/python-usbtmc works fine on the same system with the same device:
```
Python 2.7.6 (default, Jun 22 2015, 17:58:13) 
[GCC 4.8.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import usbtmc
>>> i = usbtmc.Instrument(0x0957, 0x2007)
>>> i.ask('*IDN?')
u'Agilent Technologies,34972A,MY49023315,1.17-1.12-02-02'
>>> i.ask('*IDN?')
u'Agilent Technologies,34972A,MY49023315,1.17-1.12-02-02'
>>> i.ask('*IDN?')
u'Agilent Technologies,34972A,MY49023315,1.17-1.12-02-02'
```

hgrecco, 2015-12-01:
`ask` is deprecated, please use `query`

My guess that something is staying in the buffer and therefore the instrument is chocking. Can you enable the debug?

hlipka, 2017-04-26:
I have a similar error using a Tektronix TBS1202B-EDU. I'm using this program:
```
import visa
visa.log_to_screen()
rm = visa.ResourceManager('@py')
rm.list_resources()
inst = rm.open_resource('USB0::1689::872::C010396::0::INSTR',)
print(inst.query("*IDN?"))
print(inst.query("*IDN?"))
```
And get this as output:

```
2017-04-26 23:57:48,371 - pyvisa - DEBUG - SerialSession was not imported No module named 'serial'.
2017-04-26 23:57:48,382 - pyvisa - DEBUG - USBSession and USBRawSession were correctly imported.
2017-04-26 23:57:48,384 - pyvisa - DEBUG - TCPIPSession was not imported cannot import name 'TCPIPSession'.
2017-04-26 23:57:48,385 - pyvisa - DEBUG - GPIBSession was not imported No module named 'gpib'.
2017-04-26 23:57:48,385 - pyvisa - DEBUG - Created library wrapper for unset
2017-04-26 23:57:48,386 - pyvisa - DEBUG - Created ResourceManager with session 3087558
2017-04-26 23:57:48,396 - pyvisa - DEBUG - USB0::1689::872::C010396::0::INSTR - opening ...
2017-04-26 23:57:48,613 - pyvisa - DEBUG - USB0::1689::872::C010396::0::INSTR - is open with session 9209035
2017-04-26 23:57:48,615 - pyvisa - DEBUG - USB0::1689::872::C010396::0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
TEKTRONIX,TBS 1202B-EDU,C010396,CF:91.1CT FV:v4.0

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pyvisa-py/protocols/usbtmc.py", line 202, in write
    return self.usb_send_ep.write(data)
  File "/usr/lib/python3/dist-packages/usb/core.py", line 387, in write
    return self.device.write(self, data, timeout)
  File "/usr/lib/python3/dist-packages/usb/core.py", line 948, in write
    self.__get_timeout(timeout)
  File "/usr/lib/python3/dist-packages/usb/backend/libusb1.py", line 824, in bulk_write
    timeout)
  File "/usr/lib/python3/dist-packages/usb/backend/libusb1.py", line 920, in __write
    _check(retval)
  File "/usr/lib/python3/dist-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 110] Operation timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "test.py", line 7, in <module>
    print(inst.query("*IDN?"))
  File "/usr/lib/python3/dist-packages/pyvisa/resources/messagebased.py", line 401, in query
    self.write(message)
  File "/usr/lib/python3/dist-packages/pyvisa/resources/messagebased.py", line 207, in write
    count = self.write_raw(message.encode(enco))
  File "/usr/lib/python3/dist-packages/pyvisa/resources/messagebased.py", line 185, in write_raw
    return self.visalib.write(self.session, message)
  File "/usr/lib/python3/dist-packages/pyvisa-py/highlevel.py", line 271, in write
    return self.sessions[session].write(data)
  File "/usr/lib/python3/dist-packages/pyvisa-py/usb.py", line 117, in write
    count = self.interface.write(data)
  File "/usr/lib/python3/dist-packages/pyvisa-py/protocols/usbtmc.py", line 286, in write
    bytes_sent += raw_write(data)
  File "/usr/lib/python3/dist-packages/pyvisa-py/protocols/usbtmc.py", line 204, in write
    raise ValueError(str(e))
ValueError: [Errno 110] Operation timed out
2017-04-26 23:57:50,668 - pyvisa - DEBUG - USB0::1689::872::C010396::0::INSTR - closing
2017-04-26 23:57:50,668 - pyvisa - DEBUG - USB0::1689::872::C010396::0::INSTR - is closed
2017-04-26 23:57:50,674 - pyvisa - DEBUG - Closing ResourceManager (session: 3087558)
```

When doing this in interactive mode, it help to close the instance ('inst.close()') and to re-open it, then I can send another command before I run into the timeout. I too can use the USB device directly, and also python-usbtmc.
How do I create a more detailed debug out?

mligare, 2017-07-10:
When using a Tektronix TBS 1052-EDU, I have had the same experience as @hlipka: I can only make one successful `read`/`query` before I get the timeout error noted above.  Closing and reopening the oscilloscope resource allows another single  `read`/`query`, but it doesn't work for transferring entire waveforms.  I presented the details of my experience in https://github.com/pyvisa/pyvisa/issues/236.

I have recently discovered, however, that **the same code works fine with two other Textronix models**: the TDS 1002B and the TDS 2024B.  At least for Tektronix scopes, behavior is model-dependent.

CompThing, 2019-11-12:
I get the error described in this bug with the attached code sample that runs
visa_instr.query("*IDN?")

PCAP traces added for successful operation using python_usbtmc and
failing operation using pyvisa-py
[USB_error_pcap_KeysightSigGen.zip](https://github.com/pyvisa/pyvisa-py/files/3837934/USB_error_pcap_KeysightSigGen.zip)

Once error occurs using pyvisa-py a reset of the instrument is required to restore connectivity.

MatthieuDartiailh, 2019-11-13:
@CompThing thanks for the detailed information. What version of pyvisa-py did you use ? if not the master branch, can you post the result you get with the master branch ? I have currently very little time for pyvisa-py so I am unlikely to act on this soon but hopefully I will come to it in time. If you want to investigate it more feel free to do so.