pyvisa/pyvisa-py #72: USB pipe error and connect problems with pyvisa-py 0.2

swirhun, 2015-12-07:
I suppose I should have filed a new bug intead of replying to an old thread. Sorry--I'm new to Github.

I'm using:
PyVISA==1.8
PyVISA-py==0.2

I'm having a problem with an Agilent 34411A multimeter. The stack trace looks like this. Last week I was able to get about one '*IDN?' query out of the machine and then all subsequent writes and queries failed until I unplugged/replugged/reconnected. It seems there is some bug deep in USB protocol handler. The same code works for a Keithley 2110 multimeter, so it must be some small difference in how they interact over USB which is tripping up pyvisa-py.

Traceback (most recent call last):
File "", line 1, in 
File "/usr/local/lib/python2.7/dist-packages/pyvisa/highlevel.py", line 1644, in open_resource
res.open(access_mode, open_timeout)
File "/usr/local/lib/python2.7/dist-packages/pyvisa/resources/resource.py", line 203, in open
self.session, status = self.resource_manager.open_bare_resource(self.resource_name, access_mode, open_timeout)
File "/usr/local/lib/python2.7/dist-packages/pyvisa/highlevel.py", line 1601, in open_bare_resource
return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/highlevel.py", line 191, in open
sess = cls(session, resource_name, parsed)
File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/sessions.py", line 184, in __init
self.after_parsing()
File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/usb.py", line 176, in after_parsing
self.parsed.serial_number)
File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/protocols/usbtmc.py", line 232, in init
super(USBTMC, self).init(vendor, product, serial_number, **kwargs)
File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/protocols/usbtmc.py", line 166, in init
raise Exception('failed to set configuration\n %s' % e)
Exception: failed to set configuration
[Errno 16] Resource busy

Here's an example where the first query succeeded but then it fails thereafter. Maybe there is a pyvisa-py problem in releasing the USB bus? Below, I query the DC range, I get 100mV as a response, but then I query it again and it fails due to a Pipe Error. Immediately prior, I set up i = rm.open_resource()

i.query('SENS:VOLT:DC:RANG?')
u'+1.00000000E-01\n'
i.query('SENS:VOLT:DC:RANG?')
Traceback (most recent call last):
File "", line 1, in 
File "/usr/local/lib/python2.7/dist-packages/pyvisa/resources/messagebased.py", line 407, in query
return self.read()
File "/usr/local/lib/python2.7/dist-packages/pyvisa/resources/messagebased.py", line 332, in read
message = self.read_raw().decode(enco)
File "/usr/local/lib/python2.7/dist-packages/pyvisa/resources/messagebased.py", line 306, in read_raw
chunk, status = self.visalib.read(self.session, size)
File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/highlevel.py", line 253, in read
return self.sessions[session].read(count)
File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/usb.py", line 102, in read
usb.USBError)
File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/sessions.py", line 313, in _read
current = reader()
File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/usb.py", line 96, in 
return self._read(lambda: self.interface.read(1),
File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/protocols/usbtmc.py", line 307, in read
raw_write(req)
File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/protocols/usbtmc.py", line 204, in write
raise ValueError(str(e))
ValueError: [Errno 32] Pipe error

buckko, 2015-12-27:
I have been having the same error for many weeks now.  I can only get one exchange of info (such as *IDN?) and the then usb device is "busy". Is there anything to do? Is this being looked at, or should I give up and forget trying to communicate from Linux using pyvisa-py?

mbirthelmer, 2016-01-04:
I have the same issue with a Rigol DM3058E, though in my case it always happens. I did some digging (see https://github.com/hgrecco/pyvisa-py/issues/41) and found that a "Pipe Error" corresponds to a USB stall from the device. This suggests that the issue is due to some miscommunication between the host and the device - the device being in a busy state or similar.

It might be helpful if you could capture the USB traffic between host and device and, optimally, do the same using the NI-VISA backend, to see if there are any differences.

hgrecco, 2016-01-04:
I am eager to fix it but I was not able to reproduce this. @mbirthelmer is indeed the way too go.

grvstick, 2022-04-22:
I have the same problem,

 The workaround for me was to..

1) increase the delay between commands or queries, this would tamper down how frequent it happens.
2) Not the best way but to reset the USB connection
https://gist.github.com/PaulFurtado/fce98aef890469f34d51

inferiorhumanorgans, 2026-01-02:
It's covered in the USB-TMC spec.  Most likely the issue is the command is either out of order or not recognized by the device.  The bus doesn't need to be reset but the stall condition needs to be cleared.