pyvisa/pyvisa-py #73: AttributeError: 'bytes' object has no attribute 'encode' when reading from GPIB

iliis, 2015-12-15:
Executing this code:

```python
import visa
visa.log_to_screen()
rm = visa.ResourceManager('@py')
spectrum_analyzer = rm.open_resource('GPIB0::20::INSTR')
spectrum_analyzer.write("*IDN?")
print(spectrum_analyzer.read_raw())
```

Fails with:

```
2015-12-15 16:01:40,178 - pyvisa - DEBUG - SerialSession was correctly imported.
2015-12-15 16:01:40,203 - pyvisa - DEBUG - USBSession and USBRawSession were correctly imported.
2015-12-15 16:01:40,209 - pyvisa - DEBUG - TCPIPSession was correctly imported.
2015-12-15 16:01:40,212 - pyvisa - DEBUG - GPIBSession was correctly imported.
2015-12-15 16:01:40,212 - pyvisa - DEBUG - Created library wrapper for unset
2015-12-15 16:01:40,213 - pyvisa - DEBUG - Created ResourceManager with session 4163272
2015-12-15 16:01:40,213 - pyvisa - DEBUG - GPIB0::20::INSTR - opening ...
2015-12-15 16:01:40,217 - pyvisa - DEBUG - GPIB0::20::INSTR - is open with session 9839346
2015-12-15 16:01:40,217 - pyvisa - DEBUG - GPIB.write b'*IDN?\r\n'
2015-12-15 16:01:40,226 - pyvisa - DEBUG - GPIB0::20::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
Traceback (most recent call last):
  File "./measure.py", line 21, in <module>
    print(spectrum_analyzer.read_raw())
  File "/usr/lib/python3.4/site-packages/pyvisa/resources/messagebased.py", line 306, in read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/usr/lib/python3.4/site-packages/PyVISA_py-0.3.dev0-py3.4.egg/pyvisa-py/highlevel.py", line 253, in read
    return self.sessions[session].read(count)
  File "/usr/lib/python3.4/site-packages/PyVISA_py-0.3.dev0-py3.4.egg/pyvisa-py/gpib.py", line 128, in read
    return self._read(reader, count, checker, False, None, False, gpib.GpibError)
  File "/usr/lib/python3.4/site-packages/PyVISA_py-0.3.dev0-py3.4.egg/pyvisa-py/sessions.py", line 323, in _read
    current = reader().encode('ascii')
AttributeError: 'bytes' object has no attribute 'encode'
2015-12-15 16:01:40,321 - pyvisa - DEBUG - GPIB0::20::INSTR - closing
2015-12-15 16:01:40,322 - pyvisa - DEBUG - GPIB0::20::INSTR - is closed
2015-12-15 16:01:40,332 - pyvisa - DEBUG - Closing ResourceManager (session: 4163272)
```

I "fixed" it by changing line https://github.com/hgrecco/pyvisa-py/blob/master/pyvisa-py/sessions.py#L323 to

```python
current = reader()

if not isinstance(current, bytes):
    current = current.encode('ascii')
```

------

My System:

```
python3 -m visa info
Machine Details:
   Platform ID:    Linux-4.2.6-201.fc22.x86_64+debug-x86_64-with-fedora-22-Twenty_Two
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.4.2
   Compiler:       GCC 5.1.1 20150618 (Red Hat 5.1.1-4)
   Bits:           64bit
   Build:          Jul  9 2015 17:24:30 (#default)
   Unicode:        UCS4

PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.3.dev0
      GPIB INSTR: Available via Linux GPIB (b'4.0.2')
      USB RAW: Available via PyUSB (1.0.0b2). Backend: libusb1
      USB INSTR: Available via PyUSB (1.0.0b2). Backend: libusb1
      ASRL INSTR: Available via PySerial (2.7)
      TCPIP SOCKET: Available 
      TCPIP INSTR: Available 
```
pyvisa-py is at current master (4c266c8bdfd021e2f0b6b1cd33556a759bcf982c) and linux-gpib is version `4.0.2`

gilbertf, 2016-11-08:
Have the same issue here using the ASRL interface. iliis workaround fixes the problem for me, too.

MatthieuDartiailh, 2017-12-07:
Closed by #119

closed 2017-12-07 by MatthieuDartiailh