pyvisa/pyvisa-py #77: getting error: Could not instantiate backend

jofus-t, 2016-01-06:
Similar to previously reported issue(s) but slightly different and previously suggested fixes didn't seem to work.  I'm relatively new to python and even more so to pyvisa/pyvisa-py .. here's what I've looked for so far:

'pip list' shows:
pyusb (1.0.0b2)
PyVISA (1.8)
pyserial (3.0)
pyusb (1.0.0b2)

Output of 'python -m visa info':
Machine Details:
   Platform ID:    Windows-7-6.1.7601-SP1
   Processor:      Intel64 Family 6 Model 61 Stepping 4, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Anaconda2\python.exe
   Version:        2.7.10
   Compiler:       MSC v.1500 64 bit (AMD64)
   Bits:           64bit
   Build:          Dec  1 2015 11:29:38 (#default)
   Unicode:        UCS2

PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      #1: C:\windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 15728640
         Spec. Version: 5243904
      #2: C:\windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 15728640
         Spec. Version: 5243904
   py:
      Could not instantiate backend
      -> No backend available

I am thinking that all the required pieces are there for the serial and USB backends, (and TCPIP as well?)  Any guidance is much appreciated.

hgrecco, 2016-01-06:
what command are you running?

jofus-t, 2016-01-06:
I eventually want to use a raw socket interface, but according to the results of running 'python -m visa info' there are no backends available.  All I am doing at this point is trying to verify the resources available from pyvisa-py.

hgrecco, 2016-01-06:
You pip list does not show pyvisa-py Is this installed?
The error that you are getting is when you do:
```
>>> import visa
>>> rm = visa.ResourceManager('@py')
```
?

jofus-t, 2016-01-07:
Neglected to show this in the original post:
from pip list:
PyVISA-py (0.2)
And per your last question:
>>> import visa
>>> rm = visa.ResourceManager('@py')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Anaconda2\lib\site-packages\pyvisa\highlevel.py", line 1488, in __new__
    visa_library = open_visa_library(visa_library)
  File "C:\Anaconda2\lib\site-packages\pyvisa\highlevel.py", line 1457, in open_visa_library
    cls = get_wrapper_class(wrapper)
  File "C:\Anaconda2\lib\site-packages\pyvisa\highlevel.py", line 1437, in get_wrapper_class
    pkg = __import__(pkgname)
  File "C:\Anaconda2\lib\site-packages\pyvisa-py\__init__.py", line 26, in <module>
    from .highlevel import PyVisaLibrary
  File "C:\Anaconda2\lib\site-packages\pyvisa-py\highlevel.py", line 26, in <module>
    class PyVisaLibrary(highlevel.VisaLibraryBase):
  File "C:\Anaconda2\lib\site-packages\pyvisa-py\highlevel.py", line 49, in PyVisaLibrary
    from .usb import USBSession, USBRawSession
  File "C:\Anaconda2\lib\site-packages\pyvisa-py\usb.py", line 31, in <module>
    _ = usb.core.find()
  File "C:\Anaconda2\lib\site-packages\usb\core.py", line 1199, in find
    raise ValueError('No backend available')
ValueError: No backend available

hgrecco, 2016-01-07:
I see now! What you are missing is the usb backend. PyUSB requires a usb library (like h libusb 0.1, libusb 1.0, libusbx, libusb-win32 ir OpenUSB). It seems to me that you have not that installed in your system.

gtatyous93, 2016-01-07:
do you have any instructions on how to install the libusb 1.0. it is not very intuitive and would really help if you can provide instructions on how to install this library. Thanks!

hgrecco, 2016-01-07:
Sorry but I don't have such instructions. In any case, the proper place for that would be in the PyUSB docs.

closed 2016-01-07 by hgrecco

gtatyous93, 2016-01-07:
Thanks for your time. I just have one more questions, right now I am not using pyUSB at all yet I am still getting the "Could not instantiate backend" error: 

 PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      #1: c:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 5243905
         Spec. Version: 5243136
      #2: c:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 5243905
         Spec. Version: 5243136
   py:
      Could not instantiate backend
      -> No backend available

and here is the actual error that I get when I use '@py' with the ResourceManager object:
     self.rm  = visa.ResourceManager('@py')#'@py'  c:\Windows\system32\visa32.dll

  File "C:\Python34\lib\site-packages\pyvisa\highlevel.py", line 1488, in __new_
_
    visa_library = open_visa_library(visa_library)
  File "C:\Python34\lib\site-packages\pyvisa\highlevel.py", line 1457, in open_v
isa_library
    cls = get_wrapper_class(wrapper)
  File "C:\Python34\lib\site-packages\pyvisa\highlevel.py", line 1437, in get_wr
apper_class
    pkg = __import__(pkgname)
  File "C:\Python34\lib\site-packages\pyvisa-py\__init__.py", line 26, in <modul
e>
    from .highlevel import PyVisaLibrary
  File "C:\Python34\lib\site-packages\pyvisa-py\highlevel.py", line 26, in <modu
le>
    class PyVisaLibrary(highlevel.VisaLibraryBase):
  File "C:\Python34\lib\site-packages\pyvisa-py\highlevel.py", line 49, in PyVis
aLibrary
    from .usb import USBSession, USBRawSession
  File "C:\Python34\lib\site-packages\pyvisa-py\usb.py", line 31, in <module>
    _ = usb.core.find()
  File "C:\Python34\lib\site-packages\usb\core.py", line 1271, in find
    raise NoBackendError('No backend available')
usb.core.NoBackendError: No backend available

even though I am not using USB, it is still complaining about that and I am not sure how to fix this?
Thanks!

hgrecco, 2016-01-07:
you can either uninstall pyvisa-py (if you use the ni backend) or upgrade to the latest version in github. We saw this problem in the past (also the not so usefull error message in `python -m visa info`) and we fixed. Install it and let me know

gtatyous93, 2016-01-07:
I uninstalled pyvisa-py, cloned this repo and ran "python setup.py install" to reinstall pyvisa-py but I am still getting the same not so usefull error message in python -m visa info. Should I delete the ni backend? if not, what else can I do?

gtatyous93, 2016-01-07:
if it helps, I am using Windows 10 on Macbook pro through bootcamp. Would that effect anything?

hgrecco, 2016-01-07:
They changed something in PyUSB please change this line https://github.com/hgrecco/pyvisa-py/blob/master/pyvisa-py/usb.py#L32 and instead of `ValueError` just put `Exception`

(I will push a fix later this week)

gtatyous93, 2016-01-08:
that did not fix the error btw. The NoBackendError is raised inside usb.core.py 
I am not sure why the resource manager checks for USB on construction time, but it seems like non of the code inside the directory backend seem to find any backends for usb

hgrecco, 2016-01-08:
Have you reinstalled after making that change?

jofus-t, 2016-01-08:
OP chiming in .. 
Question: would missing the USB backend cause all backends to disappear?  Is the USB backend the first one the is looked for?  I don't need the USB back end.  At this point I only need a raw socket interface.  It seems a bit excessive to install all of the USB back end when I don't intend to use it.

jofus-t, 2016-01-08:
Another question:
In the last post you ask 'did you re-install'?  I presume you mean re-installing pyvisa-py?  I'm not a developer and not versed in using git, but since I don't have a local repository, it seems that re-installing will just pull down the original source and not have the desired effect.  I can modify the local copy of usb.py and have done so, but it doesn't fix the error.

Thanks-

jofus-t, 2016-01-08:
The OP (me) is asking that this issue be re-opened since it has not been solved, and several questions posed by me went unanswered.  Replies to the other poster having a similar problem have been helpful, but this issue should still be open.

Thanks -

hgrecco, 2016-01-08:
I guess you did the change in your cloned version of the repository. If that is the case, Python will not be able to see this change. You need to uninstall pyvisa-py and then reinstall it. The best way to install it is using pip. Just move inside the repo and do 
```
pip install .
```

Regarding your questions:
You do not need pyusb/pyserial and linux-gpib if you want to use an ethernet socket (TCPIP RAW resource in the VISA jargon). PyVISA-py will raise an exception only when you try to open a resource for which the backend is not available.

jofus-t, 2016-01-10:
It appears that beyond your last statement, it is not possible to use *any* resource (in this case a RAW socket resource) when an exception is raised trying to open a different resource for which the backend (in this case USB) is not available.  In fact this exception prevents *any* backends from being seen.  My near-term solution is simply to uninstall pyUSB.  Then the socket backend becomes available.
In the long term I will need a better solution which, it appears, consists of:
- uninstalling pyvisa-py
- install libusb 1.0 and whatever is needed to support it
- re-install libUSB
- re-install pyvisa-py

Being that I'm not trying to develop, but rather am just using these packages I don't have a cloned repository available to fix and re-install pyvisa-py from.

Thanks -

hgrecco, 2016-01-11:
You do not need to clone the repository, you can just download the zip file using the "Download zip" button (it is usually on the right).

In any case, I have just pushed a fix for this issue. Just uninstall pyvisa-py and reinstall using
```
pip install https://github.com/hgrecco/pyvisa-py/archive/master.zip
```

jofus-t, 2016-01-12:
Yes.  Much better.  This allows the serial, socket backends to be available even when USB backend is not present.

Thank You

hgrecco, 2016-01-12:
Can you confirm that you get a nice error message using `python -m visa info` and when trying open an USB resource

gtatyous93, 2016-01-12:
The new update introduces a different error. I tried to open a resource and I got this:

>>> inst = im.open_resource(port)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Python34\lib\site-packages\pyvisa-1.8-py3.4.egg\pyvisa\highlevel.py", line 1644, in open_resource
    res.open(access_mode, open_timeout)
  File "C:\Python34\lib\site-packages\pyvisa-1.8-py3.4.egg\pyvisa\resources\resource.py", line 203, in open
    self.session, status = self._resource_manager.open_bare_resource(self._resource_name, access_mode, open_timeout)
  File "C:\Python34\lib\site-packages\pyvisa-1.8-py3.4.egg\pyvisa\highlevel.py", line 1601, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "C:\Python34\lib\site-packages\pyvisa_py-0.3.dev0-py3.4.egg\pyvisa-py\highlevel.py", line 191, in open
    sess = cls(session, resource_name, parsed)
  File "C:\Python34\lib\site-packages\pyvisa_py-0.3.dev0-py3.4.egg\pyvisa-py\sessions.py", line 184, in __init__
    self.after_parsing()
  File "C:\Python34\lib\site-packages\pyvisa_py-0.3.dev0-py3.4.egg\pyvisa-py\serial.py", line 68, in after_parsing
    self.interface = cls(port=self.parsed.board, timeout=2000, writeTimeout=2000)
  File "C:\Python34\lib\site-packages\serial\serialwin32.py", line 31, in __init__
    SerialBase.__init__(self, *args, **kwargs)
TypeError: __init__() got an unexpected keyword argument 'writeTimeout'
>>>

hgrecco, 2016-01-12:
This **NOT** due to the latest change. I guess that in the middle you upgraded PySerial to version 3.X. PySerial 3.X introduced several backward incompatible changes in the naming.

jofus-t, 2016-01-12:
I am not able to confirm since I have no USB INSTR class devices connected at the moment.

hgrecco, 2016-01-12:
@jofus-t But if you do not have libusb or similar installed, I would actually like to see the output of:
```
python -m visa info
```
and in python:
```python
>>> import visa
>>> rm = visa.ResourceManager('@py')
>>> rm.open_resource('ASRL1::INSTR')
```

gtatyous93, 2016-01-12:
here is the output of python -m visa info without using pyusb:

Machine Details:
   Platform ID:    Darwin-15.2.0-x86_64-i386-64bit
   Processor:      i386

Python:
   Implementation: CPython
   Executable:     /Library/Frameworks/Python.framework/Versions/3.4/bin/python3.4
   Version:        3.4.3
   Compiler:       GCC 4.2.1 (Apple Inc. build 5666) (dot 3)
   Bits:           64bit
   Build:          Feb 23 2015 02:52:03 (#v3.4.3:9b73f1c3e601)
   Unicode:        UCS4

PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      #1: /Library/Frameworks/visa.framework/visa:
         found by: auto
         bitness: 32, 64
         Vendor: National Instruments
         Impl. Version: 14680064
         Spec. Version: 5243904
   py:
      Version: 0.2
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb.util'
      ASRL INSTR: Available via PySerial (3.0.1)
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'
      TCPIP INSTR: Available 
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb.util'
      TCPIP SOCKET: Available 

However, when I install pyusbe and the backend libusb, I still get the 'no backend available' error. I know that I said I am not using USB before but actually now I need to work with an instrument that has usb interface only. Do you have any pointers on how I can fix that error?

hgrecco, 2016-01-13:
I created a small script to test the absence of libusb or similar and test the output for `python -m visa info` and opening a resource.

```python
import usb
def raise_exc():
    raise Exception('Just a simulated exception')

from usb import core
core.find = raise_exc

from pyvisa.util import get_debug_info

print("--- This is the output of python -m visa info ---")

get_debug_info()

print("--- This is the output of trying to open an USB resource ---")

import visa
rm = visa.ResourceManager('@py')
rm.open_resource("USB::12345::12345::12345::INSTR")
```

When I run it, I get a a nice error message (in both cases):
```
--- This is the output of python -m visa info ---
Machine Details:
   Platform ID:    Darwin-14.5.0-x86_64-i386-64bit
   Processor:      i386

Python:
   Implementation: CPython
   Executable:     /Users/grecco/anaconda/envs/pyvisadev/bin/python
   Version:        3.4.3
   Compiler:       GCC 4.2.1 (Apple Inc. build 5577)
   Bits:           64bit
   Build:          Mar  6 2015 12:07:41 (#default)
   Unicode:        UCS4

PyVISA Version: 1.7

Backends:
   ni:
      Version: 1.7 (bundled with PyVISA)
      #1: /Library/Frameworks/VISA.framework/Versions/Current/VISA:
         found by: user
         bitness: 32, 64
         Vendor: National Instruments
         Impl. Version: 14680064
         Spec. Version: 5243904
      #2: /Library/Frameworks/visa.framework/visa:
         found by: auto
         bitness: 32, 64
         Vendor: National Instruments
         Impl. Version: 14680064
         Spec. Version: 5243904
   py:
      Version: 0.3.dev0
      ASRL INSTR: Available via PySerial (2.7)
      USB RAW:
         PyUSB does not seem to be properly installed.
         Please refer to PyUSB documentation and
         install a suitable backend like
         libusb 0.1, libusb 1.0, libusbx,
         libusb-win32 or OpenUSB.
         Just a simulated exception
      TCPIP INSTR: Available
      TCPIP SOCKET: Available
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'
      USB INSTR:
         PyUSB does not seem to be properly installed.
         Please refer to PyUSB documentation and
         install a suitable backend like
         libusb 0.1, libusb 1.0, libusbx,
         libusb-win32 or OpenUSB.
         Just a simulated exception
   sim:
      Version: unknown
      Spec version: 1.1

--- This is the output of trying to open an USB resource ---
Traceback (most recent call last):
  File "test.py", line 18, in <module>
    rm.open_resource("USB::12345::12345::12345::INSTR")
  File "/Users/grecco/anaconda/envs/pyvisadev/lib/python3.4/site-packages/pyvisa/highlevel.py", line 1644, in open_resource
    res.open(access_mode, open_timeout)
  File "/Users/grecco/anaconda/envs/pyvisadev/lib/python3.4/site-packages/pyvisa/resources/resource.py", line 203, in open
    self.session, status = self._resource_manager.open_bare_resource(self._resource_name, access_mode, open_timeout)
  File "/Users/grecco/anaconda/envs/pyvisadev/lib/python3.4/site-packages/pyvisa/highlevel.py", line 1601, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "/Users/grecco/anaconda/envs/pyvisadev/lib/python3.4/site-packages/pyvisa-py/highlevel.py", line 191, in open
    sess = cls(session, resource_name, parsed)
  File "/Users/grecco/anaconda/envs/pyvisadev/lib/python3.4/site-packages/pyvisa-py/sessions.py", line 154, in _internal
    raise ValueError(msg)
ValueError: PyUSB does not seem to be properly installed.
Please refer to PyUSB documentation and
install a suitable backend like
libusb 0.1, libusb 1.0, libusbx,
libusb-win32 or OpenUSB.
Just a simulated exception
```

hgrecco, 2016-01-13:
Regarding getting PyUSB to work on windows (that is installing libusb or similar) I would suggest that you open an issue in the PyUSB tracker which is the most appropriate place.

jofus-t, 2016-01-14:
python -m visa info returns what appears to be something reasonable:

Machine Details:
   Platform ID:    Windows-7-6.1.7601-SP1
   Processor:      Intel64 Family 6 Model 61 Stepping 4, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Anaconda2\python.exe
   Version:        2.7.10
   Compiler:       MSC v.1500 64 bit (AMD64)
   Bits:           64bit
   Build:          Dec  1 2015 11:29:38 (#default)
   Unicode:        UCS2

PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      #1: C:\windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 15728640
         Spec. Version: 5243904
      #2: C:\windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 15728640
         Spec. Version: 5243904
   py:
      Version: 0.3.dev0
      ASRL INSTR: Available via PySerial (3.0)
      TCPIP INSTR: Available
      USB RAW:
         PyUSB does not seem to be properly installed.
         Please refer to PyUSB documentation and install a suitable backend like libusb 0.1, libusb 1.0, libusbx, libusb-win32 or OpenUSB\No
 backend available
      USB INSTR:
         PyUSB does not seem to be properly installed.
         Please refer to PyUSB documentation and install a suitable backend like libusb 0.1, libusb 1.0, libusbx, libusb-win32 or OpenUSB\No
 backend available
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named gpib
      TCPIP SOCKET: Available
------------------------------------------------------------------------------
Trying to open serial INSTR class device returns an error (since I have no such device connected).
*But* I do have a device listening to a raw socket so I can do this without error:

>>> rm.open_resource('TCPIP::127.0.0.1::5025::SOCKET')
<'TCPIPSocket'('TCPIP0::127.0.0.1::5025::SOCKET')>

But I can not just list the available resources - although I would expect to at least get a list with the raw socket device as the only member I get this error instead:
>>> print(rm.list_resources())
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Anaconda2\lib\site-packages\pyvisa\highlevel.py", line 1560, in list_resources
    return self.visalib.list_resources(self.session, query)
  File "C:\Anaconda2\lib\site-packages\pyvisa-py\highlevel.py", line 238, in list_resources
    raise errors.VisaIOError(errors.StatusCode.error_resource_not_found.value)
pyvisa.errors.VisaIOError: VI_ERROR_RSRC_NFOUND (-1073807343): Insufficient location information or the requested device or resource is not
present in the system.

I also can not do a query on the instrument :
>>> inst = rm.open_resource('TCPIP::127.0.0.1::5025::SOCKET')
>>> print(inst.query("*IDN?"))
But it just hangs.
Now if I close the socket from the instrument's side, I get the correct response to the query.  Maybe pyvisa is expecting a EOL sequence that is different than whatever the instrument is producing and it is waiting?  

I can open NI MAX and get the expected query response from this instrument over the raw socket. Querying through NI MAX shows a newline at the end of the string.

BTW

>>> inst.write("*IDN?")
(7, <StatusCode.success: 0>)
Note: 7 chars are sent.  NI MAX appears to send 6 (counting the \n for EOL)

If I query a little differently I get this error:
>>> print(inst.query_ascii_values("*IDN?"))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Anaconda2\lib\site-packages\pyvisa\resources\messagebased.py", line 449, in query_ascii_values
    self.write(message)
  File "C:\Anaconda2\lib\site-packages\pyvisa\resources\messagebased.py", line 207, in write
    count = self.write_raw(message.encode(enco))
  File "C:\Anaconda2\lib\site-packages\pyvisa\resources\messagebased.py", line 185, in write_raw
    return self.visalib.write(self.session, message)
  File "C:\Anaconda2\lib\site-packages\pyvisa-py\highlevel.py", line 271, in write
    return self.sessions[session].write(data)
  File "C:\Anaconda2\lib\site-packages\pyvisa-py\tcpip.py", line 436, in write
    size = self.interface.send(block)
socket.error: [Errno 10053] An established connection was aborted by the software in your host machine

My questions:
- Why can't I list avaiable resources and see the raw socket interface?
- How can I figure out what is expected for end-of-line on write and read?

Thanks -

hgrecco, 2016-01-14:
> But I can not just list the available resources - although I would expect to at least get a list with the raw socket device as the only member.

This is not true. In many cases VISA is not able to see the Ethernet, gpib, serial resources. In those cases VISA tries to connect to them and see if something comes back. For pyvisa-py, the general rule is: if it works with the NI backend, then we would try to make it work.

> How can I figure out what is expected for end-of-line on write and read?

In the manual of the instrument

jofus-t, 2016-01-14:
To your first point - NI can list the raw socket by discovery, but pyvisa-py doesn't.

To your second response - The issue is not knowing what the instrument expects.  I can figure that out using NI MAX or assume that a simple '\n' is expected on the instrument side (which turns out to be the case).
The issue is that pyvisa appears to be sending an extra character to the instrument on write, and possibly expecting an extra character on read.  I can't change what the instrument expects, but I should be able to change the EOL sequence sent/expected by pyvisa.
- Can this be done with the visa query method?
-  Or do I have to use separate reads and writes and try to send the termination sequence there?  When I tried that it appears pyvisa-py\tcpip.py threw an error.

hgrecco, 2016-01-14:
> NI can list the raw socket by discovery, but pyvisa-py doesn't.
How does it work? How does it distinguishes between a device and any other open port in the network?

> The issue is not knowing what the instrument expects.
That was not clear from your question. The end of line characters can be specified as it is explained in the documentation: http://pyvisa.readthedocs.org/en/stable/resources.html#termination-characters

jofus-t, 2016-01-15:
I did that, per the documentation.  I specified the same end of line sequence that is sent by NI MAX ('\n') and the same sequence that the instrument sends on a read (again '\n').  I can read, write, query from NI MAX and get the expected behavior.
If I do this through python it just hangs (after setting the read and write terminations as you suggest).
This is one reason I think pyvisa is failing.  The other is that even without the proper termination characters, the operation should eventually time out - but it does not.  (Yes - I did check the timeout value.)

hgrecco, 2016-01-15:
Thanks for the info. Can you enable logging and put the log here?

jofus-t, 2016-01-15:
This is long, but here it is:
The response to python -m visa inf is the same as above:

>>> import visa
>>> visa.log_to_screen()
>>> rm = visa.ResourceManager('@py')
2016-01-15 13:41:09,401 - pyvisa - DEBUG - SerialSession was correctly imported.
2016-01-15 13:41:09,421 - pyvisa - DEBUG - USBSession and USBRawSession were not imported No backend available.
2016-01-15 13:41:09,428 - pyvisa - DEBUG - TCPIPSession was correctly imported.
2016-01-15 13:41:09,433 - pyvisa - DEBUG - GPIBSession was not imported No module named gpib.
2016-01-15 13:41:09,434 - pyvisa - DEBUG - Created library wrapper for unset
2016-01-15 13:41:09,437 - pyvisa - DEBUG - Created ResourceManager with session 1515264
>>> inst = rm.open_resource('TCPIP::127.0.0.1::5025::SOCKET')
2016-01-15 13:42:25,466 - pyvisa - DEBUG - TCPIP::127.0.0.1::5025::SOCKET - opening ...
2016-01-15 13:42:25,482 - pyvisa - DEBUG - TCPIP::127.0.0.1::5025::SOCKET - is open with session 8147123
>>> inst.read_termination = '\n'
>>> inst.write_termination = '\n'
>>> inst.write("*IDN?")
(6, <StatusCode.success: 0>)
>>> inst.read()
2016-01-15 13:52:45,332 - pyvisa - DEBUG - TCPIP::127.0.0.1::5025::SOCKET - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)

--------------------------
At this point it is hung.  I can get a prompt back only after closing the socket from the instrument end.  Sometimes I actually get the correct response to the *IDN? query after I close the instrument side of the connection.  I presume this is because pyvisa it flushes the data on close.  

If I reconnect the instrument and then try to close and re-open the resource manager, it is unable to create the socket connection again - giving this response:
>>> inst = rm.open_resource('TCPIP::127.0.0.1::5025::SOCKET')
2016-01-15 14:44:19,391 - pyvisa - DEBUG - TCPIP::127.0.0.1::5025::SOCKET - opening ...
2016-01-15 14:44:19,394 - pyvisa - DEBUG - TCPIP::127.0.0.1::5025::SOCKET - is open with session 5876509
2016-01-15 14:44:19,428 - pyvisa - DEBUG - TCPIP::127.0.0.1::5025::SOCKET - closing
2016-01-15 14:44:19,434 - pyvisa - DEBUG - TCPIP::127.0.0.1::5025::SOCKET - is closed

The only way to recover from this (where it will actually re-open the raw socket and leave it open) is to exit the python interpreter completely and start over.

jofus-t, 2016-01-21:
Follow-up from my last post:
1) By setting read_termination = u'\n' (from the default of '') and leaving write_termination set to the (odd) value of u'\r\n', and making no mistakes, I can get a query response from my instrument.  Otherwise it hangs and the only recovery is to close all visa resources, close the socket from the instrument end, kill the instrument app owning the socket, and then re-start everything.
2) The timeout appears to have no effect at all.  I've never seen it return, regardless of the value I set (which has been anywhere from 0 to 1000000).  We took a short glance through the code and wondered if the second vs. millisecond issue is at work here.  There seems to be quite a bit of "/1000" going on.  Is the timeout being rounded down to nothing?