pyvisa/pyvisa-py #78: PYVISA-py visa.ResourceManager('@py') not working.

iitmrohan, 2016-01-21:


dalonsoa, 2016-01-22:
Same problem here:

>>> import visa
>>> rm = visa.ResourceManager('@py')
>>> rm.list_resources()
Traceback (most recent call last):
  File "<pyshell#2>", line 1, in <module>
    rm.list_resources()
  File "/usr/lib/python3/dist-packages/pyvisa/highlevel.py", line 1560, in list_resources
    return self.visalib.list_resources(self.session, query)
  File "/usr/lib/python3/dist-packages/pyvisa/highlevel.py", line 669, in list_resources
    raise NotImplementedError
NotImplementedError
>>> 

I am afraid I don't have any known device address to test if it works directly. 

Best wishes

hgrecco, 2016-01-22:
Have you installed from github or PyPI?

dalonsoa, 2016-01-22:
In my case, from the Ubuntu repositories. They should be pretty updated, but I admit it might be a bug solved in newer versions. The one installed is 0.1-2.

iitmrohan, 2016-01-22:
Hi everyone,
@hgrecco : I installed with pip ()
The issue has been resolved for now. I uninstalled the linux-gpib library which I had compiled from source. I'm using a Tektronix TDS2024B DSO which I can connect via USB so pyVisa works fine if pyUSB is present. I'll try installing linux - gpib via pip and then observe what happens. Also I observed that my python packages were 'fragemented' . Used to use apt-get earlier, so there were some conflicts in some cases where pip wouldn't correctly id some packages installed via apt-get and vice versa. I removed everything related to pyVisa like pyVisa, pyVisa-py, pyVisa-sim, pyserial, pyusb and reinstalled via pip. People may try this solution. For the most part I followed the steps listed here by @hgrecco :
http://python-in-the-lab.blogspot.in/2014/10/communicating-with-instruments-using.html

dalonsoa, 2016-01-28:
Solved! 

Un-installing the version of pyvisa and pyvisa-py in the repositories and re-installing them with pip solved the problem. Definitely it looks as a bug in the older version available in Ubuntu repositories.  

Many thanks

manuelgodoy, 2016-05-27:
I have a similar problem when running the `list_resources()` function:

```python
import visa
rm = visa.ResourceManager('@py')
rm.list_resources()
(u'ASRL/dev/cu.Bluetooth-Incoming-Port::INSTR',)
##
rm2 = visa.ResourceManager()
rm2.list_resources()
(u'ASRL1::INSTR', u'GPIB0::12::INSTR', u'GPIB1::7::INSTR', u'GPIB2::8::INSTR')
```

As you can see, if I run the `ResourceManager()` using the NI backend, I can read all the instruments. However if I use the py backend I am out of luck.

rakansha, 2016-06-08:
Hi everyone!!
I installed and upgraded py visa as well as its python -py backend and Py USB all through pip.
When I try to run rm=visa.REsourceManager('@py'), I get

**Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/PyVISA-1.8-py2.7.egg/pyvisa/highlevel.py", line 1488, in __new__
    visa_library = open_visa_library(visa_library)
  File "/usr/local/lib/python2.7/dist-packages/PyVISA-1.8-py2.7.egg/pyvisa/highlevel.py", line 1457, in open_visa_library
    cls = get_wrapper_class(wrapper)
  File "/usr/local/lib/python2.7/dist-packages/PyVISA-1.8-py2.7.egg/pyvisa/highlevel.py", line 1441, in get_wrapper_class
    raise ValueError('Wrapper not found: No package named pyvisa-%s' % backend_name)
ValueError: Wrapper not found: No package named pyvisa-py
**
Thanks for your time

hgrecco, 2016-08-22:
I cannot reproduce this. Any others getting the same thing?

manuelgodoy, 2016-08-23:
Are you on a linux? @hgrecco I am on Mac

rakansha, 2016-08-23:
@hgrecco and @manuelgodoy Its sorted now. Thanks. Yes I am on Linux.

closed 2017-11-27 by MatthieuDartiailh

Harshit1622, 2018-01-02:
Hello rakansha,

I am also facing the same error (Wrapper not found: No package named pyvisa-py).

Can you guide us for solution? How you sorted out?

Regards,
Harshit