pyvisa/pyvisa-py #84: Device has no langid

amias-channer, 2016-07-06:
I am trying to talk to ICS2313 via python3 with a rs-232 -> usb serial adapter on my ubuntu linux laptop.

I have installed all the python libs using pip3 and not using the debs.

I get this error message and can't see why this might be , i have tried two different types of usb serial adapter with no difference. I was expecting to get a list containing one device

>>> import visa
>>> rm = visa.ResourceManager('@py')
>>> rm.list_resources()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/amias/.local/lib/python3.5/site-packages/pyvisa/highlevel.py", line 1560, in list_resources
    return self.visalib.list_resources(self.session, query)
  File "/home/amias/.local/lib/python3.5/site-packages/pyvisa-py/highlevel.py", line 231, in list_resources
    for key, st in sessions.Session.iter_valid_session_classes()], [])
  File "/home/amias/.local/lib/python3.5/site-packages/pyvisa-py/highlevel.py", line 231, in <listcomp>
    for key, st in sessions.Session.iter_valid_session_classes()], [])
  File "/home/amias/.local/lib/python3.5/site-packages/pyvisa-py/usb.py", line 203, in list_resources
    serial_number=dev.serial_number,
  File "/home/amias/.local/lib/python3.5/site-packages/usb/core.py", line 830, in serial_number
    self._serial_number = util.get_string(self, self.iSerialNumber)
  File "/home/amias/.local/lib/python3.5/site-packages/usb/util.py", line 314, in get_string
    raise ValueError("The device has no langid")
ValueError: The device has no langid

here is my config 
amias@rome:~$ python3 -m visa info
Machine Details:
   Platform ID:    Linux-4.4.0-28-generic-x86_64-with-Ubuntu-16.04-xenial
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.5.1+
   Compiler:       GCC 5.3.1 20160330
   Bits:           64bit
   Build:          Mar 30 2016 22:46:26 (#default)
   Unicode:        UCS4

PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      #1: /usr/lib/x86_64-linux-gnu/libvisa.so.0.0.0:
         found by: auto
         bitness: 64
         Could not get more info:
            VI_ERROR_NSUP_ATTR (-1073807331): The specified attribute is not defined or supported by the referenced object.
   py:
      Version: 0.3.dev0
      ASRL INSTR: Available via PySerial (3.1.1)
      USB RAW: Available via PyUSB (1.0.0). Backend: libusb1
      USB INSTR: Available via PyUSB (1.0.0). Backend: libusb1
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 

There is a device connected to /dev/ttyUSB0 and i can open it with minicom

amias@rome:~$ setserial -g /dev/ttyUSB0 
/dev/ttyUSB0, UART: unknown, Port: 0x0000, IRQ: 0, Flags: low_latency

olafc, 2016-07-26:
Similar error here. Detecting devices connected via USB->Serial (1 port and 4 port) adapters works. But if I connect any USB->Ethernet adapters (tested 2 different models), I get the "ValueError: The device has no langid" as above...

edouardrosset, 2016-08-19:
Hi there,
looks like I have the same :

```
Python 3.4.3+ (default, Oct 14 2015, 16:03:50) 
[GCC 5.2.1 20151010] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import visa
>>> rm = visa.ResourceManager('@py')
>>> rm.list_resources()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.4/dist-packages/PyVISA-1.8-py3.4.egg/pyvisa/highlevel.py", line 1560, in list_resources
    return self.visalib.list_resources(self.session, query)
  File "/usr/local/lib/python3.4/dist-packages/PyVISA_py-0.3.dev0-py3.4.egg/pyvisa-py/highlevel.py", line 231, in list_resources
    for key, st in sessions.Session.iter_valid_session_classes()], [])
  File "/usr/local/lib/python3.4/dist-packages/PyVISA_py-0.3.dev0-py3.4.egg/pyvisa-py/highlevel.py", line 231, in <listcomp>
    for key, st in sessions.Session.iter_valid_session_classes()], [])
  File "/usr/local/lib/python3.4/dist-packages/PyVISA_py-0.3.dev0-py3.4.egg/pyvisa-py/usb.py", line 203, in list_resources
    serial_number=dev.serial_number,
  File "/usr/local/lib/python3.4/dist-packages/usb/core.py", line 830, in serial_number
    self._serial_number = util.get_string(self, self.iSerialNumber)
  File "/usr/local/lib/python3.4/dist-packages/usb/util.py", line 314, in get_string
    raise ValueError("The device has no langid")
ValueError: The device has no langid
```

About my config : 
```
python3 -m visa info
Machine Details:
   Platform ID:    Linux-4.2.0-34-generic-x86_64-with-Ubuntu-15.10-wily
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.4.3+
   Compiler:       GCC 5.2.1 20151010
   Bits:           64bit
   Build:          Oct 14 2015 16:03:50 (#default)
   Unicode:        UCS4

PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.3.dev0
      USB RAW: Available via PyUSB (1.0.0). Backend: libusb1
      USB INSTR: Available via PyUSB (1.0.0). Backend: libusb1
      TCPIP SOCKET: Available 
      ASRL INSTR: Available via PySerial (3.1.1)
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'
      TCPIP INSTR: Available 
```

Thanks for any thought

hgrecco, 2016-08-22:
This appears quite frequently. I think is a problem related to how pyusb handles missing langid. I would like to have a way to work around it. @walac any ideas?

closed 2016-08-22 by hgrecco

walac, 2016-08-22:
hrm, unless there is a bug in reading the langid, the error means what it says, the device has no langid, and afaik, no string can be read without a langid. Have you retried with langid value, say, 0 when it raises the error? It might be a firmware bug, and if this is the case, I doubt they are checking langid to return the string.

hgrecco, 2016-08-22:
@walac  Thanks for joining. I have not tried that, maybe we should. But would there be a way to do this directly in pyusb? i.e. have a flag in pyusb that defaults to langid 0 if no langid

edouardrosset, 2016-08-23:
I would be more than happy to help if I can.
Here is some more context : I am trying to communicate with a NI USB 6210.
lsusb -v shows :
```
Bus 003 Device 006: ID 3923:7269 National Instruments Corp. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass       255 Vendor Specific Subclass
  bDeviceProtocol       255 Vendor Specific Protocol
  bMaxPacketSize0        64
  idVendor           0x3923 National Instruments Corp.
  idProduct          0x7269 
  bcdDevice            0.00
  iManufacturer           0 
  iProduct                0 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
  [...]
```

To go further, I installed NI Visa on another machine, I was able to get visaconf work, so I have the serial number : 
[image]

Some information about this installation :
```
python3 -m visa info
Machine Details:
   Platform ID:    Linux-3.13.0-45-generic-i686-with-Ubuntu-14.04-trusty
   Processor:      i686

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.4.3
   Compiler:       GCC 4.8.4
   Bits:           32bit
   Build:          Oct 14 2015 20:33:09 (#default)
   Unicode:        UCS4

PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      #1: /usr/lib/i386-linux-gnu/libvisa.so.0.0.0:
         found by: auto
         bitness: 32
         Could not get more info:
            VI_ERROR_NSUP_ATTR (-1073807331): The specified attribute is not defined or supported by the referenced object.
   py:
      Version: 0.2
      TCPIP SOCKET: Available 
      TCPIP INSTR: Available 
      ASRL INSTR: Available via PySerial (2.7)
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'
      USB RAW: Available via PyUSB (1.0.0). Backend: libusb1
      USB INSTR: Available via PyUSB (1.0.0). Backend: libusb1
```

Trying to use the lib :
```
import visa
rm=visa.ResourceManager('@py')
rm.list_resources()

('ASRL/dev/ttyS4::INSTR', 'ASRL/dev/ttyS31::INSTR', 'ASRL/dev/ttyS30::INSTR', 'ASRL/dev/ttyS29::INSTR', 'ASRL/dev/ttyS28::INSTR', 'ASRL/dev/ttyS27::INSTR', 'ASRL/dev/ttyS26::INSTR', 'ASRL/dev/ttyS25::INSTR', 'ASRL/dev/ttyS24::INSTR', 'ASRL/dev/ttyS23::INSTR', 'ASRL/dev/ttyS22::INSTR', 'ASRL/dev/ttyS21::INSTR', 'ASRL/dev/ttyS20::INSTR', 'ASRL/dev/ttyS19::INSTR', 'ASRL/dev/ttyS18::INSTR', 'ASRL/dev/ttyS17::INSTR', 'ASRL/dev/ttyS16::INSTR', 'ASRL/dev/ttyS15::INSTR', 'ASRL/dev/ttyS14::INSTR', 'ASRL/dev/ttyS13::INSTR', 'ASRL/dev/ttyS12::INSTR', 'ASRL/dev/ttyS11::INSTR', 'ASRL/dev/ttyS10::INSTR', 'ASRL/dev/ttyS9::INSTR', 'ASRL/dev/ttyS8::INSTR', 'ASRL/dev/ttyS7::INSTR', 'ASRL/dev/ttyS6::INSTR', 'ASRL/dev/ttyS5::INSTR', 'ASRL/dev/ttyS3::INSTR', 'ASRL/dev/ttyS2::INSTR', 'ASRL/dev/ttyS1::INSTR', 'ASRL/dev/ttyS0::INSTR')
```

The device does not appear, but a full list which I believe is linked to another issue.
When trying to connect to the device directly with the address found via visaconf :

```
rm.open_resource("USB0::0x3923::0x7269::NI-VISA-3006::RAW")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.4/dist-packages/pyvisa/highlevel.py", line 1644, in open_resource
    res.open(access_mode, open_timeout)
  File "/usr/local/lib/python3.4/dist-packages/pyvisa/resources/resource.py", line 203, in open
    self.session, status = self._resource_manager.open_bare_resource(self._resource_name, access_mode, open_timeout)
  File "/usr/local/lib/python3.4/dist-packages/pyvisa/highlevel.py", line 1601, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "/usr/local/lib/python3.4/dist-packages/pyvisa-py/highlevel.py", line 191, in open
    sess = cls(session, resource_name, parsed)
  File "/usr/local/lib/python3.4/dist-packages/pyvisa-py/sessions.py", line 184, in __init__
    self.after_parsing()
  File "/usr/local/lib/python3.4/dist-packages/pyvisa-py/usb.py", line 210, in after_parsing
    self.parsed.serial_number)
  File "/usr/local/lib/python3.4/dist-packages/pyvisa-py/protocols/usbraw.py", line 45, in __init__
    super(USBRawDevice, self).__init__(vendor, product, serial_number, **kwargs)
  File "/usr/local/lib/python3.4/dist-packages/pyvisa-py/protocols/usbtmc.py", line 146, in __init__
    devices = list(self.find_devices(vendor, product, serial_number, None, **device_filters))
  File "/usr/local/lib/python3.4/dist-packages/usb/core.py", line 1249, in device_iter
    if _interop._all(tests) and (custom_match is None or custom_match(d)):
  File "/usr/local/lib/python3.4/dist-packages/pyvisa-py/protocols/usbutil.py", line 194, in cm
    if not fnmatch(getattr(dev, attr).lower(), pattern.lower()):
AttributeError: 'NoneType' object has no attribute 'lower'
```

Maybe I am facing another issue at this stage ?

walac, 2016-08-24:
@hgrecco hrm, the way to achieve that is by checking is `Device.langids` length is zero, if so, provide a langid parameter to `get_string`, if not, pass `None`.

hgrecco, 2016-08-26:
@edouardrosset  NI USB 6210 is a DAQ card so it does not fall within the scope of VISA.

hgrecco, 2016-08-26:
@walac Please notice that is not pyvisa-py who is calling get_string but pyusb:
```
File "/home/amias/.local/lib/python3.5/site-packages/usb/core.py", line 830, in serial_number
self._serial_number = util.get_string(self, self.iSerialNumber)
```

So I think it would be great if you can provide a fallback. For example [here](https://github.com/walac/pyusb/blob/9094c9b1ec2ac761dddce3c7d050fc4cd02e063d/usb/util.py#L311)

```python
 langids = dev.langids
```
to:
```python
 langids = dev.langids or util.FALLBACK_LANGIDS
```

By default `FALLBACK_LANGIDS` defined in `util` is `()` but users can change it. (You could  provide an exported setter if you would like to check the value)
```python
global FALLBACK_LANGIDS

def set_fallback_langid(langid):
    global FALLBACK_LANGIDS
    if langid is None:
        FALLBACK_LANGIDS = ()
    elif isinstance(langid, int):
        FALLBACK_LANGIDS = (langid, )
    else:
        raise ValueError('Invalid langid %s' % langid)

set_fallback_langid(None)
```

The other solutions that I see are:

1. monkey patch `dev._langids` if empty
2. create my own (safe) function to get the serial number and related.

iliis, 2017-03-13:
Just as a tip for whomever stumbles upon this issue: **It may simply be a permission issue**. Use `lsusb` to check wheter you can read any information from the device. If you get something like the following, but `sudo lsusb` shows the correct USB descriptor, then all you have to do is allow your user access to the device (e.g. by writing a new udev rule) ;)

```
$ lsusb -v -s 2:11
Bus 002 Device 011: ID 2a8d:0101  
Couldn't open device, some information will be missing
...
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
...
```

danielcrowley, 2017-08-19:
Cheers Iliis, solved it for me.

jshep321, 2019-03-11:
My rules (for an FTDI FT232 chip), fyi.  Added to/etc/udev/rules.d/99-com.rules 
SUBSYSTEM=="usb", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6014", SYMLINK+="ttyUSB0", GROUP="gpio", MODE=="0660"
SUBSYSTEM=="usb", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6014", SYMLINK+="ttyAMA0", GROUP="gpio", MODE=="0660"