pyvisa/pyvisa-py #88: opening TCP/IP connection to invalid IP blocks forever

iliis, 2016-09-23:
While `resource_manager.open_resource('TCPIP::127.0.0.1')` immediately raises a `ConnectionRefusedError`, connecting to an IP that doesn't exist (e.g. doing  `resource_manager.open_resource('TCPIP::192.255.255.42')` or something) will **block forever**.

This is an issue in https://github.com/hgrecco/pyvisa-py/blob/master/pyvisa-py/protocols/rpc.py#L339: `socket.connect()` blocks as the socket is in blocking mode. Adding for example `socket.settimeout(5)` immediately before this puts the socket into timeout mode and the `connect()` will timeout after five seconds.

I haven't checked, but as long as the socket is in blocking mode other operations might behave similarly and block forever (or timeout after a very long time).

I would suggest to at least set some default timeout. Pyvisa actually has a parameter for that in `open_resource()` called `open_timeout` (see http://pyvisa.readthedocs.io/en/stable/api/resourcemanager.html#pyvisa.highlevel.ResourceManager.open_resource).

See also https://docs.python.org/3/library/socket.html#socket-timeouts

skrchnavy, 2017-11-27:
Different approach implemented in TCPIP INSTR and TPCIP SOCKET.
* TCPIP INSTR:
```
    def connect(self):
        logger.debug('RawTCPClient: connecting to socket at (%s, %s)', self.host, self.port)
        self.sock = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
        self.sock.connect((self.host, self.port))
 ```
* TPCIP SOCKET:
```
        self.interface = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
        self.interface.setblocking(0)

        try:
            self.interface.connect_ex((self.parsed.host_address, int(self.parsed.port)))
        except Exception as e:
            raise Exception("could not create socket: %s" % e)
```

None is correct, handling of connect timeout is not provided. In SOCKET mode, also there is no check, if connect was succesful.

skrchnavy, 2017-11-29:
#115 implemented connect timeout in TCPIP SOCKET session.
@iliis Is same expected also in TCPIP INSTR session? (I don't have access to such device so can't test).

iliis, 2017-11-29:
@skrchnavy Uhm, I'm not sure. What's the difference? I'm sending GPIB commands to the device via TCP, so its INSTR?
Does it actually matter? Connecting should *never* block indefinitely, so I would say, yes, the same behaviour is expected in both cases.
And you don't need a device to test the timeout-behaviour itself for a non-existing devce ;) If you tell me what to test I can do that too.

skrchnavy, 2017-11-30:
@iliis :

There are 2 types of devices:
* `TCPIP0::1.2.3.4::999::SOCKET` \
 Raw TCP/IP access to port 999 at the specified IP address.
* `TCPIP::dev.company.com::INSTR` \
A TCP/IP device using VXI-11 or LXI located at the specified address. This uses the default LAN Device Name of inst0.
_when `TCPIP::1.2.3.4` is asked then it is equivalent to `TCPIP::1.2.3.4::INSTR`._

Device shall not block in case, when timeout != `VI_TMO_INFINITE` (http://zone.ni.com/reference/en-XX/help/370131S-01/ni-visa/vi_attr_tmo_value/).
But handling in both cases is different, SOCKET only initializes connection but INSTR is more complex protocol and it communicates during open resource phase.

Anyway I have INSTR changes in progress. Do you have access to a device for testing?