pyvisa/pyvisa-py #90: write command does not take affect

memanj, 2016-10-19:
Hi I am trying to automate setting voltage and querying ccurrent from Keithley 2220-31 DC power supply.
I am getting success for all the commands trying to write but nothing is changing on the power supply display.

>>> import visa
>>> rm = visa.ResourceManager('@py')
>>> list_re = rm.list_resources()
>>> my_instrument = rm.open_resource(list_re[0])
>>> my_instrument.query('*IDN?')
u'Keithley instruments, 2220-30-1, 9102079, 1.15-1.04\n'
>>> 
>>> 
>>> my_instrument.write('Output 1')
(24, <StatusCode.success: 0>)
>>> my_instrument.write('VOltage 12')
(24, <StatusCode.success: 0>)

**python -m visa info**
Machine Details:
   Platform ID:    Linux-3.13.0-24-generic-x86_64-with-Ubuntu-14.04-trusty
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        2.7.6
   Compiler:       GCC 4.8.2
   Bits:           64bit
   Build:          Mar 22 2014 22:59:56 (#default)
   Unicode:        UCS4

PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      #1: /usr/lib/x86_64-linux-gnu/libvisa.so.0.0.0:
         found by: auto
         bitness: 64
         Could not get more info:
            VI_ERROR_NSUP_ATTR (-1073807331): The specified attribute is not defined or supported by the referenced object.
   py:
      Version: 0.2
      ASRL INSTR:
         Please install PySerial to use this resource type.
         No module named serial
      TCPIP INSTR: Available 
      USB RAW: Available via PyUSB (1.0.0). Backend: libusb1
      USB INSTR: Available via PyUSB (1.0.0). Backend: libusb1
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named gpib
      TCPIP SOCKET: Available

samdejong86, 2018-06-25:
Were you able to solve this? I seem to be having the same issue.

MatthieuDartiailh, 2018-06-25:
Some quick questions:
- could you post the output of `python -m visa info` ? And the model of the instrument you are trying to communicate with ?
- can you communicate with your instrument using the ni backend ?
- did you set the termination character (write and read) ?

samdejong86, 2018-06-25:
`
$ python -m visa info
Machine Details:
   Platform ID:    Linux-4.4.0-128-generic-x86_64-with-debian-stretch-sid
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /hepuser/srdejong/miniconda3/bin/python
   Version:        3.6.3
   Compiler:       GCC 7.2.0
   Bits:           64bit
   Build:          Nov 20 2017 20:41:42 (#default)
   Unicode:        UCS4

PyVISA Version: 1.9.0

Backends:
   ni:
      Version: 1.9.0 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.2
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
      USB RAW: Available via PyUSB (1.0.2). Backend: libusb1
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named 'gpib'
`

I am trying to connect to a keithley 2220-30-1 DC power supply. I haven't communicated using the ni backend, since I'm on ubuntu and haven't found a way to install the ni visa.

I have not explicitly set the termination character.

MatthieuDartiailh, 2018-06-26:
Great you solved your issue !

closed 2018-06-26 by MatthieuDartiailh