pyvisa/pyvisa-py #92: Cannot use list_resources() if gpib board in not connected

KevinNadaud, 2016-11-08:
Hi,
Firslty thanks a lot for pyvsa-py which is usefull and powerfull. 
I use pyvisa-py on linux with linux-gpib. 

There are a small bug when using list_resources() when the gpib board is not connected (I want to use only usb or lan in the present case). I have the following error:
`>>> import visa
>>> rm=visa.ResourceManager('@py')
>>> rm.list_resources()
libgpib: ibBoardOpen failed to open device file /dev/gpib0
libgpib: No such file or directory
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.7/dist-packages/pyvisa/highlevel.py", line 1560, in list_resources
    return self.visalib.list_resources(self.session, query)
  File "/usr/lib/python2.7/dist-packages/pyvisa-py/highlevel.py", line 231, in list_resources
    for key, st in sessions.Session.iter_valid_session_classes()], [])
  File "/usr/lib/python2.7/dist-packages/pyvisa-py/gpib.py", line 57, in list_resources
    return ['GPIB0::%d::INSTR' % pad for pad in _find_listeners()]
  File "/usr/lib/python2.7/dist-packages/pyvisa-py/gpib.py", line 35, in _find_listeners
    if gpib.listener(BOARD, i) and gpib.ask(BOARD, 1) != i:
gpib.GpibError: listener() error: Bad file descriptor (errno: 9)
`
If I connected the gpib board, list_ressources() work well. 
If I uninstall linux-gpib, it also works but I sometimes need gpib to control old instruments, so I prefere to keep it.

Regards

abhishek123963, 2020-12-21:
hii , connected with instrument but still getting this error plz see

 import visa
>>> rm = visa.ResourceManager('@py')
>>> rm.list_resources()
libgpib: ibBoardOpen failed to open device file /dev/gpib0
libgpib: No such file or directory
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.7/dist-packages/pyvisa/highlevel.py", line 1560, in list_resources
    return self.visalib.list_resources(self.session, query)
  File "/usr/lib/python2.7/dist-packages/pyvisa-py/highlevel.py", line 231, in list_resources
    for key, st in sessions.Session.iter_valid_session_classes()], [])
  File "/usr/lib/python2.7/dist-packages/pyvisa-py/gpib.py", line 57, in list_resources
    return ['GPIB0::%d::INSTR' % pad for pad in _find_listeners()]
  File "/usr/lib/python2.7/dist-packages/pyvisa-py/gpib.py", line 35, in _find_listeners
    if gpib.listener(BOARD, i) and gpib.ask(BOARD, 1) != i:
gpib.GpibError: listener() error: Bad file descriptor (errno: 9)
>>> 
here is information -m visa info
 python -m visa info
Machine Details:
   Platform ID:    Linux-4.4.0-197-generic-x86_64-with-Ubuntu-16.04-xenial
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        2.7.12
   Compiler:       GCC 5.4.0 20160609
   Bits:           64bit
   Build:          Oct  5 2020 13:56:01 (#default)
   Unicode:        UCS4

PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      #1: /usr/lib/x86_64-linux-gnu/libvisa.so.0.0.0:
         found by: auto
         bitness: 64
         Could not get more info:
            VI_ERROR_NSUP_ATTR (-1073807331): The specified attribute is not defined or supported by the referenced object.
   py:
      Version: 0.2
      ASRL INSTR: Available via PySerial (3.0.1)
      TCPIP INSTR: Available 
      USB RAW: Available via PyUSB (1.0.0b2). Backend: libusb1
      USB INSTR: Available via PyUSB (1.0.0b2). Backend: libusb1
      GPIB INSTR: Available 
      TCPIP SOCKET: Available

MatthieuDartiailh, 2020-12-24:
Python 2.7 is not supported anymore. Please update to Python 3 and latest Pyvisa and Pyvisa-py.