pyvisa/pyvisa-py #93: problem when using rm.list_resources()

osama989, 2016-11-11:
Here is the details of my machine:

Machine Details:
   Platform ID:    Linux-3.10.0-327.36.1.el7.x86_64-x86_64-with-redhat-7.2-Nitrogen
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        2.7.5
   Compiler:       GCC 4.8.5 20150623 (Red Hat 4.8.5-4)
   Bits:           64bit
   Build:          Sep 14 2016 10:12:39 (#default)
   Unicode:        UCS4

PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.3.dev0
      ASRL INSTR:
         Please install PySerial to use this resource type.
         No module named serial
      TCPIP INSTR: Available 
      USB RAW: Available via PyUSB (1.0.0b1). Backend: libusb1
      USB INSTR: Available via PyUSB (1.0.0b1). Backend: libusb1
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named gpib
      TCPIP SOCKET: Available

The issue is when I try to see my devices using the regular commands:

>>> import visa
>>> rm = visa.ResourceManager('@py')
>>> rm.list_resources()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.7/site-packages/PyVISA-1.8-py2.7.egg/pyvisa/highlevel.py", line 1560, in list_resources
    return self.visalib.list_resources(self.session, query)
  File "/usr/lib/python2.7/site-packages/pyvisa-py/highlevel.py", line 231, in list_resources
    for key, st in sessions.Session.iter_valid_session_classes()], [])
  File "/usr/lib/python2.7/site-packages/pyvisa-py/usb.py", line 222, in list_resources
    serial_number=dev.serial_number,
AttributeError: 'Device' object has no attribute 'serial_number'

I get this error. 

I am trying to connect a KEITHLEY picoammeter 6485 via PROLOGIX GPIB-USB converter?

Any Idea of what is happening?

KevinNadaud, 2016-11-14:
Hi,
I don't why you get this error but to control instrument using a GPIB-USB converter I need to use linux-gpib. Even if I use the converter, the device is considered to be plugged in GPIB and I cannot control him with USB tools. 
Have-you tried with linux-gpib ?

osama989, 2016-11-14:
@KevinNadaud Thank you very much for your answer.

I have multiple platforms to try, I could not download linux-gpib on Linux scientific, some sort of premission issue, so I switch to an Ubuntu 16.04.

I was able to download gpib-linux so here is my device information:

Machine Details:
   Platform ID:    Linux-4.4.0-45-generic-x86_64-with-Ubuntu-16.04-xenial
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        2.7.12
   Compiler:       GCC 5.4.0 20160609
   Bits:           64bit
   Build:          Jul  1 2016 15:12:24 (#default)
   Unicode:        UCS4

PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      #1: /usr/lib/x86_64-linux-gnu/libvisa.so.0.0.0:
         found by: auto
         bitness: 64
         Could not get more info:
            VI_ERROR_NSUP_ATTR (-1073807331): The specified attribute is not defined or supported by the referenced object.
   sim:
      Version: 0.3
      Spec version: 1.1
   py:
      Version: 0.2
      ASRL INSTR:
         Please install PySerial to use this resource type.
         No module named serial
      TCPIP INSTR: Available 
      USB RAW: Available via PyUSB (1.0.0). Backend: libusb1
      USB INSTR: Available via PyUSB (1.0.0). Backend: libusb1
      GPIB INSTR: Available 
      TCPIP SOCKET: Available 

However, I get a different kind of error when I use the same commands in pyvisa:

libgpib: ibBoardOpen failed to open device file /dev/gpib0
libgpib: No such file or directory
Traceback (most recent call last):
  File "test.py", line 3, in <module>
    rm.list_resources_info()
  File "/usr/lib/python2.7/dist-packages/pyvisa/highlevel.py", line 1572, in list_resources_info
    for resource in self.list_resources(query))
  File "/usr/lib/python2.7/dist-packages/pyvisa/highlevel.py", line 1560, in list_resources
    return self.visalib.list_resources(self.session, query)
  File "/usr/lib/python2.7/dist-packages/pyvisa-py/highlevel.py", line 231, in list_resources
    for key, st in sessions.Session.iter_valid_session_classes()], [])
  File "/usr/lib/python2.7/dist-packages/pyvisa-py/gpib.py", line 57, in list_resources
    return ['GPIB0::%d::INSTR' % pad for pad in _find_listeners()]
  File "/usr/lib/python2.7/dist-packages/pyvisa-py/gpib.py", line 35, in _find_listeners
    if gpib.listener(BOARD, i) and gpib.ask(BOARD, 1) != i:
gpib.GpibError: listener() error: Bad file descriptor (errno: 9)

I am trying hard to figure out what is going on here, but I cannot, so any help would be appreciated.

Thank you

KevinNadaud, 2016-11-14:
Hi,
It seems you have the same that I have when the GPIB converter is not connected. 
[https://github.com/hgrecco/pyvisa-py/issues/92]
To verify if the board is detected, can you give the output of `lsusb -v` ?
You should see the converter in the list. If you see it, test to use `ibtest` to verify you can contact the board without pyvisa.

If it does not work, try to do `sudo gpib-config` and retry `ibtest`. If now it works, relaunch pyvisa and your script should works. To launch automatically gpib-config at starting, you can add the lines in the file /etc/udev/rules.d/linux-gpib_usb.rules
̀`SUBSYSTEM=="usb", ACTION=="add", ENV{DEVTYPE}=="usb_device", ATTR{idVendor}=="3923", ATTR{idProduct}=="709b", RUN+="/usr/local/sbin/gpib_config"`
`KERNEL=="gpib[0-9]*", ACTION=="add", MODE="660", GROUP="gpib"`

idVendor and idProduct must correspond to what your obtain with `lsusb -v`
I hope this help you !

DirtyJerz, 2017-08-07:
@osama989 
I was just going through the same issues as you. I forgot to add the user to the GPIB group. It was a permission error. 

`
libgpib: ibBoardOpen failed to open device file /dev/gpib0
libgpib: Permission denied
libgpib: error in is_cic()!
Traceback (most recent call last):
  File "run.py", line 9, in <module>
    inst = Gpib.Gpib(0,14) # Instrument GPIB Address = 14
  File "/home/pi/venv/local/lib/python2.7/site-packages/Gpib.py", line 26, in __init__
    self.id = gpib.dev(name, pad, sad, timeout, send_eoi, eos_mode)
gpib.GpibError: dev() error: Bad file descriptor (errno: 9)
`

slightlynybbled, 2019-04-23:
Issue appears to be dated and may need to be closed from a maintenance standpoint.  Does the author have any further information?

osama989, 2019-04-23:
Hi, I ended up using py-serial for this. I think I was having issues with compiling py-visa also. In any case, the issue is solved, so you can close this if you wish. Thanks a lot for everyone’s comments.

closed 2019-04-23 by MatthieuDartiailh