pyvisa/pyvisa-py #94: write_binary_values fails with Python 2x

phookit, 2017-03-21:
Calling write_binary_values with a string fails with struct conversion error on Python 2

write_binary_values(':system:setup ", "<foo>bar</foo>", datatype='B', is_big_endian=False)

In `to_ieee_block` the call to struct.pack fails (not integer) in the branch for Python < 3.

The data needs to be ord()'d before being packed.

skrchnavy, 2017-12-07:
bytes or bytearray shall be used for such data.
```
write_binary_values(':system:setup ", b'bar', datatype='B', is_big_endian=False)
```

Anyway, this is probably issue of [pyvisa](https://github.com/pyvisa/pyvisa) project

bilderbuchi, 2021-03-04:
I don't think Python 2 is still supported, so I think this can be closed.

MatthieuDartiailh, 2021-03-04:
Indeed

closed 2021-03-04 by MatthieuDartiailh