pyvisa/pyvisa-py #95: ni vs py backend GPIB speed tests

nikhartman, 2017-04-07:
Has anyone done this test in any rigorous way? 

I recently put together a Linux machine to run as a sort of instrument server using the 'py' backend and linux-gpib. Sending a single query takes about 50ms compared to about 20ms on a windows machine running the 'ni' backend. I'd like to speed this up, but it is not clear whether I am limited by my hardware (totally possible, the Linux machine is very minimal) or by the linux-gpib backend.

DavidLutton, 2017-05-21:
What timings/comparisons are you after?
The time between starting Python process and retrieving data like**\*IDN?**,  single data point, a trace from a scope/analyser, or the turnaround time to make a single query

A scope like Agilent DSO5052A with USBTMC, GPIB, Ethernet and plenty of trace points, if you are after trace retrieval time comparisons

I have been using NI VISA on Windows
I got Linux-GPIB and Python + PyVISA '@py' + PyVISA-py working today

This [timer block in Python](https://gist.github.com/DavidLutton/9a1ab6efc78b457159cfe0b6977d601b) could be used for basic timing

nikhartman, 2017-05-22:
I should have left a note here. After leaving that comment, I dual booted my Windows machine and tested the same code on the Ubuntu (py backend + linux-gpib) and Windows (ni backend) sides of the same machine. 

It was really the single query times I was looking at, like writing "*IDN?" and reading the response. The results aren't on the computer I'm writing from, but the response times were pretty much identical (~20ms to query an SRS830) on the Windows/Ubuntu tests.

closed 2017-05-22 by nikhartman