pyvisa/pyvisa-py #96: Two major pyvisa-py bugs

swirhun, 2017-04-20:
I have spent all day trying to get pyvisa-py to work with common instruments on Ubuntu, about a year after giving up and using NI-VISA on a mac. I believe I have tracked down multiple issues. I am testing using a Keithley 2450, but out of the box pyvisa-py also fails to communicate with a Keysight 34411A and a Keithley 2230-30-1.

Basically problem #1 is that the Keithleys (and possibly many other instruments?) don't comply with the USBTMC standard v1.0 (http://sdpha2.ucsd.edu/Lab_Equip_Manuals/USBTMC_1_00.pdf). In particular, in usbtmc.py, they don't like receiving the extra "padding to multiple of 4 bytes" that you have [correctly] included according to the standard. The solution to getting reliable writes, as far as I can tell, is simply to comment out that padding as shown below:

        s =  struct.pack('BBBx', MsgID.dev_dep_msg_out, btag, ~btag & 0xFF) + \
                         struct.pack("<LBxxx", size, eom) + \
                         chunk # + \
                         # b'\0' * ((4 - size) % 4)

Problem #2 is that pyvisa-py is reading multiple times when it shouldn't be. After much trouble, I found that usbtmc.py -> def read() loop is being executed twice even though it consumes all data and correctly detects 'eom' on the first iteration of the while loop. The bug is how you're using a lambda function in usb.py when you pass the "reader" function to superclass Session:

        return self._read(lambda: self.interface.read(1),
                          count,
                          lambda current: False,
                          supress_end_en,
                          term_char,
                          term_char_en,
                          usb.USBError)

I think the lambda: self.interface.read(1) actually calls the function once when the arguments are generated, since self.interface.read(1) doesn't depend on the [omitted] lambda variable. Then, when Session._read() uses "reader", a second read occurs. This screws up most instruments since the first read already consumed all their data, and the second read generates an "Error -420: Query UNTERMINATED (from USBTMC)" (as an example on the Keithley 2450 which has helpful log messages).

I'm not 100% sure this is all that's broken, but I hope I described it clearly enough. Feel free to contact me for details; I would really love for pyvisa-py to be as good as NI-VISA. You must have some very lenient instruments that you're testing on, since none of the 3 that I have on my desk work with pyvisa-py out of the box. My suggestion for problem #2 is to make usbtmc.read() have a default argument (count=1), and then change the above code to just pass the function directly:

        return self._read(self.interface.read,
                          count,
                          lambda current: False,
                          supress_end_en,
                          term_char,
                          term_char_en,
                          usb.USBError)

swirhun, 2017-04-20:
I should add, I haven't actually fixed the problem yet, so the above may be a little inaccurate, but I thought I would post where I suspect the problems are occurring.

swirhun, 2017-04-20:
I also question whether the complex logic at the bottom of sessions.py is necessary. Can't it simply be this?

            if current:
                out += current
                end_indicator_received = end_indicator_checker(current)
                if end_indicator_received and not suppress_end_en:
                    # RULE 6.1.1
                    return out, constants.StatusCode.success
                if current.endswith(chr(termination_char)) and termination_char_en:
                    # RULE 6.1.2
                    return out, constants.StatusCode.success_termination_character_read
                if len(out) == count:
                    # RULE 6.1.3
                    return out, constants.StatusCode.success_max_count_read

swirhun, 2017-04-20:
The above three fixes (bulk-out padding removed, reader lambda removed, and end-indicator logic changed) now seem to work on the Keithley 2450, and so far all -420 Query UNTERMINATED errors are fixed. I'll also check whether the 2230-3-1 and 34411A work now, possibly tomorrow morning.

swirhun, 2017-04-21:
Argh, the Agilent/Keysight 34411A is not happy about the missing zero-padding, so I guess I was wrong about that section...

Ziggles1, 2017-07-03:
I agree about the end_indicator_checker. This currently appears to default to False which causes an additional read. This is also fouling communications with my Rigol DS1052E which often doesn't return the length of data that is expected.
I would have thought that the end_indicator_checker function should do something other than just pass back a False default. The current implementation appears to cause the read functions to keep repeating until a timeout occurs, and only then is the test for EOM in usbtmc.py allowed to operate. This also results in reads being unnecessarily long (as the timeout has to run to completion for every read).
Thoughts?

majid-khan, 2017-07-13:
I had the same problem communicating with USB instruments and receiving a "query unterminated" error.  The problem is with the line:
`lambda current: False,` 
which should actually be:
`lambda current: True,` 
The maintainers seem to have made this change in the code but the packages where most users install from have not been updated.  After making this change, USB communication is working for me.

swirhun, 2017-07-18:
Thanks majid-khan. It seems like there are still some small open issues in sessions.py and its subclasses (in serial.py / usb.py) to make it as robust as NI-VISA. I wanted to add for others who may be struggling with the Keithley 2230-3 that we found in some decade-old code of ours two lines of specialized initialization required to get it to work with pyvisa-py, but curiously not required when using NI-VISA. Here is a simplified snippet:

```python
rm = manager.ResourceManager()
inst = rm.open_resource(r)
inst.timeout = 3000

if keithley_2230:
  inst.write('SYST:REM')
inst.write('*CLS')
if keithley_2230:
  inst.write('*ESE 0')

// After this, everything works as expected.
desc = inst.query('*IDN?').strip()
```