pyvisa/pyvisa-py #101: Problem with TCPIP read() when more than "chunk_length" is returned

rriggs, 2017-07-10:
I am attempting to download a screenshot from a Siglent DSO using the SCDP command.  This just dumps the binary data for a BMP file.  The problem appears to be that it sends all of the data in one shot.

```python
n, status = scope.write('SCDP')
if status == StatusCode.success:
    data = scope.read_raw()
    open("test.bmp", "w").write(data)
```
The code just hangs at read_raw().

Adding some diagnostics to TCPIPInstrSession.read(), this is what I see:

```
wanted 4096 bytes, read 1152057 bytes
want -1131577 more bytes
```

Upon return from this function, count is negative, the excess data read and returned, but the count is lost in the upper layers.  I don't understand the VXI-11 protocol well enough to know if this is a problem with the device not conforming to spec (it seems to work with NI-VISA and their native app), with PyVISA or with PyVISA-py.

I bodged the read() function to buffer the data it receives and only return the requested chunk_size.  That seems to work for me (I successfully stored a screenshot), but I don't know if that is a reasonable fix.  If it is, I am happy to work up a pull request.

rriggs, 2017-07-10:
It appears I can work around the issue by telling read_raw() to read 2MB of data.  Somehow I missed that read_raw() accepts an optional length argument.

skrchnavy, 2017-12-01:
@rriggs are you able test this isse when #116 merged?

skrchnavy, 2017-12-07:
#116 merged, is there a chance to verify if this issue is fixed?