pyvisa/pyvisa-py #102: query_binary_values on TCPIP-Socket stops after encountering termination character

jeras, 2017-07-26:
after some debugging it appears TCPIPSocketSession.read.enabled == True

MatthieuDartiailh, 2017-11-05:
Can you confirm this is still an issue, we fixed some issues with TCPIP recently ? Also did you check you are using the proper format for your data ?

jeras, 2017-11-05:
I probably used the released version from PyPI. I did not check this issue recently, and I am switching jobs, so I will not have the time to check this issue with recent code, sorry.

MatthieuDartiailh, 2017-11-05:
No problem.

MatthieuDartiailh, 2017-11-27:
@skrchnavy as you work on TCPIP could you check if this is still an issue ?

skrchnavy, 2017-11-27:
OK

skrchnavy, 2017-11-27:
@jeras please provide more info related to this issue. I am not sure if I understand, what is reported. 

Be aware, that `enabled` in read function is from attribute `VI_ATTR_TERMCHAR_EN` (http://zone.ni.com/reference/en-XX/help/370131S-01/ni-visa/vi_attr_termchar_en/) so when `True` it shall terminate.

skrchnavy, 2017-12-07:
This issue could be fixed in #116, is it possible to test?