pyvisa/pyvisa-py #103: Rigol DP832A +pyvisa-py+windows, "The parameter is incorrect."

kimble71, 2017-09-01:
Trying to build python control for Rigol DP832A PSU with x86 tablet running windows10.

(done before:
-got it working with Win7+NI-VISA+C#
-got it working on x86 tablet with Win10 +NI-VISA+pyvisa+python2)

Now my goal is to get it working by using opensource tools on Win10.

So far:
-clean Win10 installation
-installed python2-7.x
-cmd/ python -m pip install -U pip
-cmd/ pip install pyusb
-cmd/ pip install pyvisa
-cmd/ pip install pyvisa-py

Then python nagged about no backend... (python -m visa info)
so,I:
-installed usblib for DB800 via zadig GUI

I should now have all things in place.

did in python console:
>>> import visa
>>> resources = visa.ResourceManager('@py')
>>> print(resources.list_resources())
(u'USB0::6833::3601::DP8B185350299::0::INSTR',)

So far all works as expected, Rigol is seen on the USB connection.

But this fails:
>>> rigolpsu = resources.open_resource('USB0::6833::3601::DP8B185350299::0::INSTR')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Python27\lib\site-packages\pyvisa\highlevel.py", line 1644, in open_resource
    res.open(access_mode, open_timeout)
  File "C:\Python27\lib\site-packages\pyvisa\resources\resource.py", line 203, in open
    self.session, status = self._resource_manager.open_bare_resource(self._resource_name, access_mode, open_timeout)
  File "C:\Python27\lib\site-packages\pyvisa\highlevel.py", line 1601, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "C:\Python27\lib\site-packages\pyvisa-py\highlevel.py", line 191, in open
    sess = cls(session, resource_name, parsed)
  File "C:\Python27\lib\site-packages\pyvisa-py\sessions.py", line 184, in __init__
    self.after_parsing()
  File "C:\Python27\lib\site-packages\pyvisa-py\usb.py", line 176, in after_parsing
    self.parsed.serial_number)
  File "C:\Python27\lib\site-packages\pyvisa-py\protocols\usbtmc.py", line 236, in __init__
    self.usb_dev.set_configuration()
  File "C:\Python27\lib\site-packages\usb\core.py", line 869, in set_configuration
    self._ctx.managed_set_configuration(self, configuration)
  File "C:\Python27\lib\site-packages\usb\core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "C:\Python27\lib\site-packages\usb\core.py", line 148, in managed_set_configuration
    self.backend.set_configuration(self.handle, cfg.bConfigurationValue)
  File "C:\Python27\lib\site-packages\usb\backend\libusb0.py", line 493, in set_configuration
    _check(_lib.usb_set_configuration(dev_handle, config_value))
  File "C:\Python27\lib\site-packages\usb\backend\libusb0.py", line 431, in _check
    raise USBError(errmsg, ret)
usb.core.USBError: [Errno None] libusb0-dll:err [set_configuration] could not set config 1: win error: The parameter is incorrect.

((my third day using python, so, sorry if it's something obvious...)) 
(seeking also if some Rigol (super)user can help: http://www.eevblog.com/forum/testgear/controlling-rigol-dp832a-psu-with-python/ )

kimble71, 2017-09-05:
I did not manage to get DP832A working with pyvisa-py & USB.
Also LAN interface did not work straight out of the box, but...

For some reason pyvisa-py is not able to find RIGOL from LAN interface "automagically".

But this works...
[quote]
import visa
rm = visa.ResourceManager("@py")
inst = rm.open_resource("TCPIP::192.168.1.5")
inst.query('*IDN?')
-> RIGOL ...
inst.query(':MEAS:CURRent? CH1')
->0.0000
inst.close()
[/quote]

Using pyvisa-py & python2.

skrchnavy, 2017-12-04:
this is from [tcpip.py](../blob/master/pyvisa-py/tcpip.py#L41)
```
    @staticmethod
    def list_resources():
        # TODO: is there a way to get this?
        return []
```