pyvisa/pyvisa-py #104: Exception: error creating link: 4

mbe9a, 2017-09-08:
I am trying to open an ethernet connection to a Keysight GPIB/USB/LAN Gateway, and I'm getting this error.

---------------------------------------------------------------------------
Exception                                 Traceback (most recent call last)
<ipython-input-4-9bb51beec674> in <module>()
----> 1 rm.open_resource('TCPIP::FIRGPIBGateway.phys.virginia.edu::INSTR')

/usr/local/anaconda2/lib/python2.7/site-packages/PyVISA-1.8-py2.7.egg/pyvisa/highlevel.pyc in open_resource(self, resource_name, access_mode, open_timeout, resource_pyclass, **kwargs)
   1642                 raise ValueError('%r is not a valid attribute for type %s' % (key, res.__class__.__name__))
   1643 
-> 1644         res.open(access_mode, open_timeout)
   1645 
   1646         for key, value in kwargs.items():

/usr/local/anaconda2/lib/python2.7/site-packages/PyVISA-1.8-py2.7.egg/pyvisa/resources/resource.pyc in open(self, access_mode, open_timeout)
    201         logger.debug('%s - opening ...', self._resource_name, extra=self._logging_extra)
    202         with self._resource_manager.ignore_warning(constants.VI_SUCCESS_DEV_NPRESENT):
--> 203             self.session, status = self._resource_manager.open_bare_resource(self._resource_name, access_mode, open_timeout)
    204 
    205             if status == constants.VI_SUCCESS_DEV_NPRESENT:

/usr/local/anaconda2/lib/python2.7/site-packages/PyVISA-1.8-py2.7.egg/pyvisa/highlevel.pyc in open_bare_resource(self, resource_name, access_mode, open_timeout)
   1599         :return: Unique logical identifier reference to a session.
   1600         """
-> 1601         return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
   1602 
   1603     def open_resource(self, resource_name,

/usr/local/anaconda2/lib/python2.7/site-packages/pyvisa-py/highlevel.pyc in open(self, session, resource_name, access_mode, open_timeout)
    189         cls = sessions.Session.get_session_class(parsed.interface_type_const, parsed.resource_class)
    190 
--> 191         sess = cls(session, resource_name, parsed)
    192 
    193         return self._register(sess), constants.StatusCode.success

/usr/local/anaconda2/lib/python2.7/site-packages/pyvisa-py/sessions.pyc in __init__(self, resource_manager_session, resource_name, parsed)
    182                       constants.VI_ATTR_RSRC_CLASS: parsed.resource_class,
    183                       constants.VI_ATTR_INTF_TYPE: parsed.interface_type}
--> 184         self.after_parsing()
    185 
    186     def after_parsing(self):

/usr/local/anaconda2/lib/python2.7/site-packages/pyvisa-py/tcpip.pyc in after_parsing(self)
     60 
     61         if error:
---> 62             raise Exception("error creating link: %d" % error)
     63 
     64         self.link = link

Exception: error creating link: 4

__________________________________________________________

Machine Details:
   Platform ID:    Linux-4.4.0-93-generic-x86_64-with-debian-stretch-sid
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/local/anaconda2/bin/python
   Version:        2.7.11
   Compiler:       GCC 4.4.7 20120313 (Red Hat 4.4.7-1)
   Bits:           64bit
   Build:          Dec  6 2015 18:08:32 (#default)
   Unicode:        UCS4

PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.2
      ASRL INSTR:
         Please install PySerial to use this resource type.
         No module named serial
      TCPIP INSTR: Available 
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named usb
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named usb
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named gpib
      TCPIP SOCKET: Available 

Any help would be greatly appreciated.

closed 2017-09-08 by mbe9a

MatthieuDartiailh, 2017-09-08:
Why did you close ?

mbe9a, 2017-09-08:
So the device itself can’t be connected to through py-visa. I tried hooking up a glib device to the gateway and connected to that successfully -- Just me being stupid.