pyvisa/pyvisa-py #105: GPIB Instrument Support for Windows

tkzilla, 2017-09-22:
Is there a plan to support GPIB instruments on Windows? linux-gpib support is nice, but there are some instruments/software that only run on Windows and they use GPIB loopback addresses.

MatthieuDartiailh, 2017-09-22:
I am not aware of any library other than visa providing GPIB support on windows, if you do please let us know.

tivek, 2018-01-01:
GPIB drivers on Windows sit "below" the Visa layer. As far as I know, the GPIB C API exposed by gpib-32.dll/obj on Windows is the same as the C API of linux-gpib. So the question then becomes whether there are Python bindings for GPIB that work both on Linux and Windows.

There is an attempt to build the linux-gpib Python bindings on Windows: https://sourceforge.net/p/linux-gpib/mailman/message/36149135/

Instead of building a Python module in C like linux-gpib does, another way would be to use ctypes and call the GPIB C libraries. The old gpib-devices library does that quite successfully on both Windows and Linux, see [libgpib.py](http://bazaar.launchpad.net/~jordens/gpib-devices/trunk/view/head:/gpib/libgpib.py). However, that is only a straightforward wrap of the C API and some effort is required to have the same interface as linux-gpib.

MatthieuDartiailh, 2018-01-01:
This sounds very interesting @tivek. Please keep us posted on your progress. As soon as you have something that could be supported by pyvisa-py please let us know and we will try to provide some guidance on how to proceed from there.

tivek, 2018-01-03:
The rough first commit of linux-gpib-like bindings that also works on Windows is here: https://github.com/tivek/gpib_ctypes

I implemented the low-level `gpib` module from scratch by referring to linux-gpib Python bindings docstrings, NI GPIB C docs as well as linux-gpib docs. However, the nice, object-oriented, capital-letter `Gpib` module that pyvisa-py is using is copied directly from the linux-gpib project. Since linux-gpib is GPL2, that means for the time being gpib_ctypes is given under GPL2.

So far I tried out the code on Windows with NI GPIB-USB-HS, driver version 3.1.0.49154. I plan to also test it on Debian with linux-gpib in the coming days.

How to try it out:
- install pyvisa, pyvisa-py, gpib_ctypes

```python
from gpib_ctypes import make_default_gpib
make_default_gpib() # monkey-patch us to take precedence over eg. linux-gpib's implementation

import visa
rm = visa.ResourceManager('@py')
rm.list_resources()
# output on my Windows machine: ('GPIB0::14::INSTR', 'GPIB0::23::INSTR')
```

Bugs are to be expected. Testers welcome!

tivek, 2018-01-03:
Update on Linux tests: the above snippet works fine with linux-gpib on 201801 Arch Linux and the NI GPIB-USB-HS adapter (I only had to put board_type = "ni_usb_b" in /etc/gpib.conf to make the hardware work).

MatthieuDartiailh, 2018-02-23:
@tivek any plan on publishing a first release soon ?

tivek, 2018-03-01:
@MatthieuDartiailh apologies for the late reply, was offline and out of town.

Basically, https://github.com/tivek/gpib_ctypes works for us (TM) in the lab. It can be installed and tested right now using:
`
pip install git+https://github.com/tivek/gpib_ctypes.git
`

Of course I can make an alpha release on pypi, but I was hoping for a bit more testing by others before my first ever release there. Open for any advice you can give.

MatthieuDartiailh, 2018-03-01:
I can try to give it a go perhaps sometimes next week. I will keep you posted. After, as release go, I tend to hold back quite a bit myself, but actually releasing early usually makes sense. I do not find any large bug I think it would be worth releasing that way I will document it in PyVISA-py for the next release  (and make the manual patching superfluous).

tivek, 2018-03-01:
Sounds great. Please let me know what you find. For now the monkeypatching method above works ok so maybe start with that.

And thank you for the advice on releasing early. More eyes will catch more bugs. I will do so in the coming days.

kpckpc, 2018-04-02:
If linux-gpib is already supported, you might want to have a look at the script I created for octave
[https://wiki.octave.org/Instrument_control_package#Windows_.28cygwin.29](https://wiki.octave.org/Instrument_control_package#Windows_.28cygwin.29)
This allowed me to use software designed for linux-gpib wihtout any changes on Windows.

mb711, 2018-10-25:
Tivek,

I'm not sure if I found a bug, but the error I'm seeing is stumping me. I posted to the below link

https://github.com/tivek/gpib_ctypes/issues/1

closed 2019-05-29 by MatthieuDartiailh