pyvisa/pyvisa-py #108: No resources detected

labs-stellios, 2017-10-27:
0. Install dependencies

```
sudo apt-get install python3 pyvisa
pip3 install pyvisa
```

1. Download and install National Instruments’s VISA library (link may change)
```
pushd ~/Downloads
wget http://ftp.ni.com/support/softlib/visa/NI-VISA/5.4.1/Linux/NI-VISA-5.4.1.iso
sudo mkdir /media/iso
sudo mount -o loop ./NI-VISA-5.4.1.iso /media/iso
pushd /media/iso
sudo ./INSTALL
# Accept the license agreement
popd
sudo umount /media/iso
rm ./NI-VISA-5.4.1.iso
popd
```

2. Test it

`lsusb` shows my USB-to-serial adapter and my DAQ:

```
(venv) me@vm:~/daq$ lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 006: ID 0403:6001 Future Technology Devices International, Ltd FT232 USB-Serial (UART) IC
Bus 002 Device 007: ID 09db:0113 Measurement Computing Corp. 
Bus 002 Device 004: ID 0e0f:0008 VMware, Inc. 
Bus 002 Device 003: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 002 Device 002: ID 0e0f:0003 VMware, Inc. Virtual Mouse
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
```

but I get nothing out of the resource manager:

```
(venv) me@vm:~/daq$ python
Python 3.5.2 (default, Sep 14 2017, 22:51:06) 
[GCC 5.4.0 20160609] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import visa
>>> rm = visa.ResourceManager()
>>> print(rm.list_resources())
()
```

VISA info shows that I have consistent python and VISA bitnesses:

```
(venv) me@vm:~/daq$ python -m visa info
Machine Details:
   Platform ID:    Linux-4.10.0-37-generic-x86_64-with-Ubuntu-16.04-xenial
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /home/me/virtualenvs/venv/bin/python
   Version:        3.5.2
   Compiler:       GCC 5.4.0 20160609
   Bits:           64bit
   Build:          Sep 14 2017 22:51:06 (#default)
   Unicode:        UCS4

PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      #1: /usr/lib/x86_64-linux-gnu/libvisa.so.0.0.0:
         found by: auto
         bitness: 64
         Could not get more info:
            VI_ERROR_NSUP_ATTR (-1073807331): The specified attribute is not defined or supported by the referenced object.
```

MatthieuDartiailh, 2017-10-29:
Since you have NI VISA, could you check whether or not your instrument show up into NI MAX ? If it does not, something is wrong at a deeper level than PyVISA. In which case I would recommend you to check the manual of your instrument and possibly to contact the vendor.

In your infos, the part about the VISA library appear to be incomplete and I do not know why. I will try to look into it.

labs-stellios, 2017-11-03:
I'm sorry I can no longer support this.

closed 2017-11-03 by labs-stellios