pyvisa/pyvisa-py #109: Unsupported attributes for PyVISA-py TCPIP SOCKET communication

MatthieuDartiailh, 2017-11-05:
First reported on PyVisa: https://github.com/pyvisa/pyvisa/issues/274

skrchnavy, 2017-11-26:
@cwgamma 
I checked documentation for NI: http://zone.ni.com/reference/en-XX/help/370131S-01/ni-visa/vi_attr_send_end_en/
`VI_ATTR_SEND_END_EN` is not valid for TCPIP SOCKET only for TCPIP INSTR.

Anyway, it works as you have written, I tested it using pyvisa shell with backend pyvisa-py, attribute is supported but not listed and can't be changed.

MatthieuDartiailh, 2017-11-26:
Should we close then ?

skrchnavy, 2017-11-26:
no, this is real issue, VI_ATTR_SEND_END_EN shall be ignored but ['VI_ATTR_SUPPRESS_END_EN'](http://zone.ni.com/reference/en-XX/help/370131S-01/ni-visa/vi_attr_suppress_end_en/) shall be implemented. I am working on it for TCPIP.

MatthieuDartiailh, 2017-11-26:
Cool

MatthieuDartiailh, 2017-11-27:
Closed by #114

closed 2017-11-27 by MatthieuDartiailh