pyvisa/pyvisa-py #122: Wrapper not found: No package named pyvisa-py

Harshit1622, 2018-01-02:
Hello Anyone,

Please guide us  on below error. 
**Wrapper not found: No package named pyvisa-py**

Code:
import visa
rm = visa.ResourceManager('@py')

Thanks,
Harshit

MatthieuDartiailh, 2018-01-02:
How did you install pyvisa and pyvisa-py ?

Harshit1622, 2018-01-02:
Thanks for replying..

I have run below commands:
Pip install pyvisa-py
Pip install -U pyvisa

MatthieuDartiailh, 2018-01-03:
Could you post the output of  `python -m visa info` ? I am not sure what is happening here.

natevillaume, 2018-05-25:
I have the same problem. I cannot specify the pyvisa-py library.
I used the same install commands... 

output from python -m visa info:

 python3 -m visa info
Machine Details:
   Platform ID:    Darwin-16.7.0-x86_64-i386-64bit
   Processor:      i386

Python:
   Implementation: CPython
   Executable:     /Library/Frameworks/Python.framework/Versions/3.6/bin/python3
   Version:        3.6.3
   Compiler:       GCC 4.2.1 (Apple Inc. build 5666) (dot 3)
   Bits:           64bit
   Build:          Oct  3 2017 00:32:08 (#v3.6.3:2c5fed86e0)
   Unicode:        UCS4

PyVISA Version: 1.9.0

Backends:
   ni:
      Version: 1.9.0 (bundled with PyVISA)
      #1: /Library/Frameworks/visa.framework/visa:
         found by: auto
         bitness: 32, 64
         Vendor: National Instruments
         Impl. Version: 14680064
         Spec. Version: 5243904
   py:
      Could not instantiate backend
      -> No backend available

MatthieuDartiailh, 2018-05-26:
Could you post the output of pip list ?

natevillaume, 2018-06-01:
$ pip list

Package          Version

---------------- ---------

asn1crypto       0.23.0

aws-xray-sdk     0.94

boto             2.48.0

boto3            1.4.7

botocore         1.7.46

certifi          2017.11.5

cffi             1.11.2

chardet          3.0.4

cookies          2.2.1

cryptography     2.1.3

docker           2.6.1

docker-pycreds   0.2.1

docutils         0.14

et-xmlfile       1.0.1

idna             2.6

jdcal            1.3

Jinja2           2.10

jmespath         0.9.3

jsonpickle       0.9.5

MarkupSafe       1.0

mock             2.0.0

moto             1.1.24

numpy            1.13.3

openpyxl         2.4.9

pandas           0.21.0

pbr              3.1.1

pip              10.0.1

py               1.5.2

pyaml            17.10.0

pycparser        2.18

pytest           3.2.5

python-dateutil  2.6.1

pytz             2017.3

pyusb            1.0.2

PyVISA           1.9.0

PyVISA-py        0.2

PyYAML           3.12

requests         2.18.4

s3transfer       0.1.11

setuptools       36.7.2

six              1.11.0

urllib3          1.22

websocket-client 0.44.0

Werkzeug         0.12.2

wrapt            1.10.11

xmltodict        0.11.0

MatthieuDartiailh, 2018-06-01:
I really don't understand what causes this issue. Can you try to install the latest develoment version ?
```
pip uninstall pyvisa-py
pip install https://github.com/pyvisa/pyvisa-py/tarball/master
```
And post the output of ``pyvisa-info` (equivalent to `python -m visa info`).

natevillaume, 2018-06-02:
$ pip uninstall pyvisa-py

Uninstalling PyVISA-py-0.2:

  Would remove:

/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/PyVISA_py-0.2-py3.6.egg-info

/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/pyvisa-py/*

Proceed (y/n)? y

  Successfully uninstalled PyVISA-py-0.2

$ pip install https://github.com/pyvisa/pyvisa-py/tarball/master

Collecting https://github.com/pyvisa/pyvisa-py/tarball/master

  Downloading https://github.com/pyvisa/pyvisa-py/tarball/master

     \ 61kB 1.0MB/s

Requirement already satisfied: pyvisa>=1.8 in
/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages
(from PyVISA-py==0.3.dev0) (1.9.0)

Installing collected packages: PyVISA-py

  Running setup.py install for PyVISA-py ... done

Successfully installed PyVISA-py-0.3.dev0

LMC-062271:~$ pyvisa-info

Machine Details:

   Platform ID:    Darwin-16.7.0-x86_64-i386-64bit

   Processor:      i386

Python:

   Implementation: CPython

   Executable:
/Library/Frameworks/Python.framework/Versions/3.6/bin/python3.6

   Version:        3.6.3

   Compiler:       GCC 4.2.1 (Apple Inc. build 5666) (dot 3)

   Bits:           64bit

   Build:          Oct  3 2017 00:32:08 (#v3.6.3:2c5fed86e0)

   Unicode:        UCS4

PyVISA Version: 1.9.0

Backends:

   ni:

      Version: 1.9.0 (bundled with PyVISA)

      #1: /Library/Frameworks/visa.framework/visa:

         found by: auto

         bitness: 32, 64

         Vendor: National Instruments

         Impl. Version: 14680064

         Spec. Version: 5243904

   py:

      Version: 0.3.dev0

      ASRL INSTR:

         Please install PySerial (>=3.0) to use this resource type.

         No module named 'serial'

      USB INSTR:

         PyUSB does not seem to be properly installed.

         Please refer to PyUSB documentation and

         install a suitable backend like

         libusb 0.1, libusb 1.0, libusbx,

         libusb-win32 or OpenUSB.

         No backend available

      USB RAW:

         PyUSB does not seem to be properly installed.

         Please refer to PyUSB documentation and

         install a suitable backend like

         libusb 0.1, libusb 1.0, libusbx,

         libusb-win32 or OpenUSB.

         No backend available

      TCPIP INSTR: Available

      TCPIP SOCKET: Available

      GPIB INSTR:

         Please install linux-gpib to use this resource type.

         No module named 'gpib'

MatthieuDartiailh, 2018-06-04:
Looks like re-installing fixed your issue. To enable any backend in addition to the TCPIP one please have a look at the documentation.
Feel free to re-open if needed.

closed 2018-06-04 by MatthieuDartiailh

HITTIH, 2018-12-29:
hello, I encountered this problem too,and I tried the above methods,and it didn't work ,But once I run it the way below, it works.
>>> import visa
>>> rm = visa.ResourceManager('@py')
>>> eload = rm.open_resource('TCPIP0::10.44.90.19::INSTR')
>>> eload.query('*IDN?')
'Chroma,63210A-150-1000,63210AL00087,1.25,1.01,1.01\n'
if I run it in this way "python xxxx.py"(which has the same code ), it didn't work

a1abhisheks, 2019-07-27:
> I really don't understand what causes this issue. Can you try to install the latest develoment version ?
> 
> ```
> pip uninstall pyvisa-py
> pip install https://github.com/pyvisa/pyvisa-py/tarball/master
> ```
> 
> And post the output of `pyvisa-info` (equivalent to `python -m visa info`).

Hi MatthieuDartiailh,

this solution worked greatly just 1 point i wants to add that after this i have to install this also

pip install gpib-ctypes

after this i can connect my system with Agilent 8960 using NI GPIB Cable and perform RF measurement for TX Power of mobile phone on EGSM900 and DCS1800