pyvisa/pyvisa-py #123: AttributeError: 'module' object has no attribute 'ResourceManager'

Sakulas, 2018-01-10:
Hey I'm using pyvisa on a raspberry pi with python 2.7.9 and pip 1.5.6. I installed and uninstalled pyvisa and pyvisa-py several times, but the problems stay. I connected the KEITHLEY Multimeter 2000 per R232 to USB with the Raspberry.

When I run the basic Code:
```
import visa
rm = visa.ResourceManager('@py')
rm.list_resources()
```
I receive:

```
Traceback (most recent call last):
  File "pyvisa.py", line 1, in <module>
    import visa
  File "/usr/local/lib/python2.7/dist-packages/visa.py", line 16, in <module>
    from pyvisa import logger, __version__, log_to_screen, constants
  File "/home/pi/pyvisa.py", line 2, in <module>
    rm = visa.ResourceManager('@py')
AttributeError: 'module' object has no attribute 'ResourceManager'
```

as well when I try

`python -m visa info`

```
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 162, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/usr/local/lib/python2.7/dist-packages/visa.py", line 16, in <module>
    from pyvisa import logger, __version__, log_to_screen, constants
  File "pyvisa.py", line 1, in <module>
    import visa
  File "/usr/local/lib/python2.7/dist-packages/visa.py", line 16, in <module>
    from pyvisa import logger, __version__, log_to_screen, constants
ImportError: cannot import name logger
```
On the other hand i can't upgrade, because the requirements are already up-to-date.
```
pip install pyvisa-py --upgrade
Requirement already up-to-date: pyvisa-py in /usr/local/lib/python2.7/dist-packages
Requirement already up-to-date: pyvisa>=1.8 in /usr/local/lib/python2.7/dist-packages (from pyvisa-py)
Requirement already up-to-date: enum34 in /usr/local/lib/python2.7/dist-packages (from pyvisa>=1.8->pyvisa-py)
```
I would be very thankfull if somebody could help me with this issue.

MatthieuDartiailh, 2018-01-11:
Could you try running `python -m visa info` from a different folder or after renaming your module ? In python 2.7, the import system will always use files in the working directory over the one in site-packages and as your file is named pyvisa.py when importing visa.py it picks your own module instead of the 'real' pyvisa module.

Sakulas, 2018-01-11:
Thanks for the fast reply! Wouldn't had thought that this could be an issue. But you were absolutly right. I tried the same code with a different name in another folder and it worked.

Sadly it stopped after this as well in the query line, like in the topic https://github.com/pyvisa/pyvisa-py/issues/122

Probably it has to deal with the USB path?

The Multimeter Configuration is: RS 232:ON , BAUD: 9600, FLOW: NONE, TX TERM: LF

The same Issue occures with:
`import visa
rm = visa.ResourceManager('@py')
a=rm.list_resources()
print(a)
multi=rm.open_resource(u'ASRL/dev/ttyUSB0::INSTR')
multi.write('FETCh?\n')
print(multi.read())`

Output:

```
(u'ASRL/dev/ttyUSB0::INSTR',)
^CTraceback (most recent call last):
  File "pvtest.py", line 7, in <module>

  File "/usr/local/lib/python2.7/dist-packages/pyvisa/resources/messagebased.py"                      , line 332, in read
    message = self.read_raw().decode(enco)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa/resources/messagebased.py, line 306, in read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/highlevel.py", line 253, in read
    return self.sessions[session].read(count)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/serial.py", line 132, in read
    serial.SerialTimeoutException)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/sessions.py", line 313,in _read
    current = reader()
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/serial.py", line 114, in <lambda>
    reader = lambda: self.interface.read(1)
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 446, in read
    ready,_,_ = select.select([self.fd],[],[], self._timeout)
KeyboardInterrupt
```

The Path seems right to me, because i dont get a failure with:
```
import visa
rm = visa.ResourceManager('@py')
a=rm.list_resources()
print(a)
multi=rm.open_resource(u'ASRL/dev/ttyUSB0::INSTR')
print(multi)

```
Output:
`(u'ASRL/dev/ttyUSB0::INSTR',)
SerialInstrument at ASRL/dev/ttyUSB0::INSTR
`

MatthieuDartiailh, 2018-01-11:
I would suggest to try setting the read and write termination character according to the manual of your instrument (I noticed you were adding it manually). Also it can sometimes help to first check that the instrument does process the write, using for example a command changing the mode of the instrument (so that you can easily check that the instrument did what you expected). Let me know if you manage to get it to work.