pyvisa/pyvisa-py #124: USB input/output error when running query('*IDN?')

timhoeppner, 2018-02-01:
I'm getting the following error when trying to do a simple instrument query:

`2018-02-01 14:44:53,560 - pyvisa - DEBUG - SerialSession was correctly imported.
2018-02-01 14:44:53,570 - pyvisa - DEBUG - USBSession and USBRawSession were correctly imported.
2018-02-01 14:44:53,573 - pyvisa - DEBUG - TCPIPSession was not imported cannot import name TCPIPSession.
2018-02-01 14:44:53,573 - pyvisa - DEBUG - GPIBSession was not imported No module named gpib.
2018-02-01 14:44:53,574 - pyvisa - DEBUG - Created library wrapper for unset
2018-02-01 14:44:53,574 - pyvisa - DEBUG - Created ResourceManager with session 5348285
(u'ASRL/dev/ttyS0::INSTR', u'USB0::6833::1416::DM3R154200794::0::INSTR')
2018-02-01 14:44:53,585 - pyvisa - DEBUG - USB0::6833::1416::DM3R154200794::0::INSTR - opening ...
2018-02-01 14:44:53,895 - pyvisa - DEBUG - USB0::6833::1416::DM3R154200794::0::INSTR - is open with session 7344747
Traceback (most recent call last):
  File "visa_test.py", line 6, in <module>
    print(inst.query('*IDN?'))
  File "/usr/local/lib/python2.7/dist-packages/pyvisa/resources/messagebased.py", line 401, in query
    self.write(message)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa/resources/messagebased.py", line 207, in write
    count = self.write_raw(message.encode(enco))
  File "/usr/local/lib/python2.7/dist-packages/pyvisa/resources/messagebased.py", line 185, in write_raw
    return self.visalib.write(self.session, message)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/highlevel.py", line 271, in write
    return self.sessions[session].write(data)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/usb.py", line 117, in write
    count = self.interface.write(data)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/protocols/usbtmc.py", line 286, in write
    bytes_sent += raw_write(data)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/protocols/usbtmc.py", line 204, in write
    raise ValueError(str(e))
ValueError: [Errno 5] Input/Output Error
2018-02-01 14:44:53,901 - pyvisa - DEBUG - USB0::6833::1416::DM3R154200794::0::INSTR - closing
2018-02-01 14:44:53,901 - pyvisa - DEBUG - USB0::6833::1416::DM3R154200794::0::INSTR - is closed
`

My code:
`import visa
visa.log_to_screen()
rm = visa.ResourceManager('@py')
print(rm.list_resources())
inst = rm.open_resource(u'USB0::6833::1416::DM3R154200794::0::INSTR')
print(inst.query('*IDN?'))
`

Some more info:
`$ python -m visa info
Machine Details:
   Platform ID:    Linux-4.4.0-112-generic-x86_64-with-Ubuntu-16.04-xenial
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        2.7.12
   Compiler:       GCC 5.4.0 20160609
   Bits:           64bit
   Build:          Nov 20 2017 18:23:56 (#default)
   Unicode:        UCS4

PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.2
      ASRL INSTR: Available via PySerial (3.0.1)
      TCPIP INSTR: Available 
      USB RAW: Available via PyUSB (1.0.2). Backend: libusb1
      USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named gpib
      TCPIP SOCKET: Available 
`

I haven't had any luck with finding anyone with a similar error. Can anyone help? The weird thing to me is that I can query for instruments and it detects the USB instrument.

MatthieuDartiailh, 2018-02-01:
Could you let me know what kind of instrument you are trying to talk to ? Could you also check what termination character your instrument expects on write ? Sadly the usb backend is not bug free, you may try to update pyvisa-py to the master branch (`pip install https://github.com/pyvisa/pyvisa-py/tarball/master`) or to use the python-sbtmc package that in some cases do better. In any case please keep us posted about works and what fails.

timhoeppner, 2018-02-01:
@MatthieuDartiailh This was on a Rigol DM3058E

Also tried on a Rigol DG1022A. For this, I didn't get any errors but don't get any output from the equipment. The DG1022A does go into remote mode though, so that's promising.

I can't find anywhere what termination character the instruments expect. Doesn't say in the programming manual and the sample applications don't identify one either.

I'll give the master branch a try, how would I revert back if I don't have any luck?

```
2018-02-01 15:12:24,324 - pyvisa - DEBUG - SerialSession was correctly imported.
2018-02-01 15:12:24,335 - pyvisa - DEBUG - USBSession and USBRawSession were correctly imported.
2018-02-01 15:12:24,338 - pyvisa - DEBUG - TCPIPSession was not imported cannot import name TCPIPSession.
2018-02-01 15:12:24,338 - pyvisa - DEBUG - GPIBSession was not imported No module named gpib.
2018-02-01 15:12:24,338 - pyvisa - DEBUG - Created library wrapper for unset
2018-02-01 15:12:24,339 - pyvisa - DEBUG - Created ResourceManager with session 1097349
(u'ASRL/dev/ttyS0::INSTR', u'USB0::1024::2500::DG1F154600787\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00::0::INSTR')
2018-02-01 15:12:24,350 - pyvisa - DEBUG - USB0::1024::2500::DG1F154600787::0::INSTR - opening ...
2018-02-01 15:12:24,608 - pyvisa - DEBUG - USB0::1024::2500::DG1F154600787::0::INSTR - is open with session 9313457
2018-02-01 15:12:24,608 - pyvisa - DEBUG - USB0::1024::2500::DG1F154600787::0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)

2018-02-01 15:12:25,610 - pyvisa - DEBUG - USB0::1024::2500::DG1F154600787::0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)

2018-02-01 15:12:26,625 - pyvisa - DEBUG - USB0::1024::2500::DG1F154600787::0::INSTR - closing
2018-02-01 15:12:26,626 - pyvisa - DEBUG - USB0::1024::2500::DG1F154600787::0::INSTR - is closed
```

code:
```
import visa
visa.log_to_screen()
rm = visa.ResourceManager('@py')
print(rm.list_resources())
#inst = rm.open_resource(u'USB0::6833::1416::DM3R154200794::0::INSTR')
inst = rm.open_resource(u'USB0::1024::2500::DG1F154600787\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00::0::INSTR')
print(inst.query('*IDN?'))

inst.write('FREQ 1234.1\r\n')
print(inst.query('FREQ?'))
```

timhoeppner, 2018-02-01:
Trying on pyvisa-py master branch.

Same result with the DM3058E.

DG1022A is giving:
```
2018-02-01 15:42:48,469 - pyvisa - DEBUG - SerialSession was correctly imported.
2018-02-01 15:42:48,480 - pyvisa - DEBUG - USBSession and USBRawSession were correctly imported.
2018-02-01 15:42:48,483 - pyvisa - DEBUG - TCPIPSession was correctly imported.
2018-02-01 15:42:48,484 - pyvisa - DEBUG - GPIBSession was not imported No module named gpib.
2018-02-01 15:42:48,484 - pyvisa - DEBUG - Created library wrapper for unset
2018-02-01 15:42:48,484 - pyvisa - DEBUG - Created ResourceManager with session 9549665
(u'ASRL/dev/ttyS0::INSTR', u'USB0::1024::2500::DG1F154600787\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00::0::INSTR')
2018-02-01 15:42:48,495 - pyvisa - DEBUG - USB0::1024::2500::DG1F154600787::0::INSTR - opening ...
2018-02-01 15:42:48,808 - pyvisa - DEBUG - USB0::1024::2500::DG1F154600787::0::INSTR - is open with session 6124265
2018-02-01 15:42:48,808 - pyvisa - DEBUG - USB0::1024::2500::DG1F154600787::0::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2018-02-01 15:42:49,810 - pyvisa - DEBUG - USB0::1024::2500::DG1F154600787::0::INSTR - exception while reading: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
Traceback (most recent call last):
  File "visa_test.py", line 7, in <module>
    print(inst.query('*IDN?'))
  File "/usr/local/lib/python2.7/dist-packages/pyvisa/resources/messagebased.py", line 407, in query
    return self.read()
  File "/usr/local/lib/python2.7/dist-packages/pyvisa/resources/messagebased.py", line 332, in read
    message = self.read_raw().decode(enco)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa/resources/messagebased.py", line 306, in read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/highlevel.py", line 258, in read
    raise errors.VisaIOError(ret[1])
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
2018-02-01 15:42:49,817 - pyvisa - DEBUG - USB0::1024::2500::DG1F154600787::0::INSTR - closing
2018-02-01 15:42:49,817 - pyvisa - DEBUG - USB0::1024::2500::DG1F154600787::0::INSTR - is closed
```

MatthieuDartiailh, 2018-02-01:
This is typical of an improperly configured read_termination. You could try setting it to \n or \r or \n\r as those are common values.

timhoeppner, 2018-02-01:
Tried every combination with no success.

MatthieuDartiailh, 2018-02-04:
Try using python-usbtmc, if it works please let me know. I have on my to-do list to improve the usb back-end based on it for some time now, but never found enough time. If it works for you, it would be one more incentive for me to try to fix it once and for all, hopefully.

timhoeppner, 2018-02-12:
No luck on usbtmc either....

```
import usbtmc
instr =  usbtmc.Instrument(6833, 1416)
print(instr.ask("*IDN?"))
```

gives:

```
Traceback (most recent call last):
  File "usbtmc_test.py", line 3, in <module>
    print(instr.ask("*IDN?"))
  File "build/bdist.linux-x86_64/egg/usbtmc/usbtmc.py", line 771, in ask
  File "build/bdist.linux-x86_64/egg/usbtmc/usbtmc.py", line 754, in read
  File "build/bdist.linux-x86_64/egg/usbtmc/usbtmc.py", line 673, in read_raw
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 402, in read
    return self.device.read(self, size_or_buffer, timeout)
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 988, in read
    self.__get_timeout(timeout))
  File "/usr/local/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 833, in bulk_read
    timeout)
  File "/usr/local/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 936, in __read
    _check(retval)
  File "/usr/local/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 32] Pipe error
```

timhoeppner, 2018-02-12:
Tried on python 3.5.2, both pyvisa-py and usbtmc, still no luck.

Do you think this is an instrument issue or setup issue on my end?

MatthieuDartiailh, 2018-02-12:
One last thing you could try is checking the instrument configuration. It may need some kind of specific udev rules. You can try following the instruction in the README of python-usbtmc. https://github.com/python-ivi/python-usbtmc
After that I must say I am running out of ideas and furthermore I do not really have experience with running usb devices on linux. Any way should you find any fix please let me know and feel free to open a pull request to improve the docs.

timhoeppner, 2018-02-12:
Okay this is interesting....

Fresh reboot, tried usbtmc again and it worked!

Then, if I try pyvisa-py I get a "Pipe error" and switching back to usbtmc I also get the same error.

Restarting the instrument allow me to communicate with it over usbtmc again until I try anything over pyvisa-py.

terminal log for reference:
```
tim@tim-H97M-D3H:~/dev$ python usbtmc_test.py 
Rigol Technologies,DM3058E,DM3R154200794,01.01.00.02.01.00
-2.526636E-04
-2.496206E-04
-2.507633E-04
-2.548248E-04
tim@tim-H97M-D3H:~/dev$ python usbtmc_test.py 
Rigol Technologies,DM3058E,DM3R154200794,01.01.00.02.01.00
-2.573585E-04
-2.557190E-04
-2.521171E-04
-2.516079E-04
tim@tim-H97M-D3H:~/dev$ python visa_test.py 
2018-02-12 09:13:53,883 - pyvisa - DEBUG - SerialSession was correctly imported.
2018-02-12 09:13:53,894 - pyvisa - DEBUG - USBSession and USBRawSession were correctly imported.
2018-02-12 09:13:53,896 - pyvisa - DEBUG - TCPIPSession was correctly imported.
2018-02-12 09:13:53,897 - pyvisa - DEBUG - GPIBSession was not imported No module named gpib.
2018-02-12 09:13:53,897 - pyvisa - DEBUG - Created library wrapper for unset
2018-02-12 09:13:53,897 - pyvisa - DEBUG - Created ResourceManager with session 7248516
(u'ASRL/dev/ttyS0::INSTR', u'USB0::6833::1416::DM3R154200794::0::INSTR')
2018-02-12 09:13:53,909 - pyvisa - DEBUG - USB0::6833::1416::DM3R154200794::0::INSTR - opening ...
2018-02-12 09:13:54,086 - pyvisa - DEBUG - USB0::6833::1416::DM3R154200794::0::INSTR - is open with session 3038767
Traceback (most recent call last):
  File "visa_test.py", line 12, in <module>
    print(inst.query('*IDN?'))
  File "/usr/local/lib/python2.7/dist-packages/pyvisa/resources/messagebased.py", line 401, in query
    self.write(message)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa/resources/messagebased.py", line 207, in write
    count = self.write_raw(message.encode(enco))
  File "/usr/local/lib/python2.7/dist-packages/pyvisa/resources/messagebased.py", line 185, in write_raw
    return self.visalib.write(self.session, message)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/highlevel.py", line 276, in write
    ret = self.sessions[session].write(data)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/usb.py", line 120, in write
    count = self.interface.write(data)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/protocols/usbtmc.py", line 286, in write
    bytes_sent += raw_write(data)
  File "/usr/local/lib/python2.7/dist-packages/pyvisa-py/protocols/usbtmc.py", line 204, in write
    raise ValueError(str(e))
ValueError: [Errno 32] Pipe error
2018-02-12 09:13:54,093 - pyvisa - DEBUG - USB0::6833::1416::DM3R154200794::0::INSTR - closing
2018-02-12 09:13:54,093 - pyvisa - DEBUG - USB0::6833::1416::DM3R154200794::0::INSTR - is closed
tim@tim-H97M-D3H:~/dev$ python usbtmc_test.py 
Traceback (most recent call last):
  File "usbtmc_test.py", line 3, in <module>
    print(instr.ask("*IDN?"))
  File "build/bdist.linux-x86_64/egg/usbtmc/usbtmc.py", line 771, in ask
  File "build/bdist.linux-x86_64/egg/usbtmc/usbtmc.py", line 754, in read
  File "build/bdist.linux-x86_64/egg/usbtmc/usbtmc.py", line 673, in read_raw
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 402, in read
    return self.device.read(self, size_or_buffer, timeout)
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 988, in read
    self.__get_timeout(timeout))
  File "/usr/local/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 833, in bulk_read
    timeout)
  File "/usr/local/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 936, in __read
    _check(retval)
  File "/usr/local/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 32] Pipe error
tim@tim-H97M-D3H:~/dev$ lsusb
Bus 002 Device 002: ID 8087:8001 Intel Corp. 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:8009 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 045e:00cb Microsoft Corp. Basic Optical Mouse v2.0
Bus 003 Device 006: ID 05e3:0607 Genesys Logic, Inc. Logitech G110 Hub
Bus 003 Device 004: ID 05e3:0607 Genesys Logic, Inc. Logitech G110 Hub
Bus 003 Device 003: ID 046d:c31c Logitech, Inc. Keyboard K120
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
tim@tim-H97M-D3H:~/dev$ lsusb
Bus 002 Device 002: ID 8087:8001 Intel Corp. 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:8009 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 045e:00cb Microsoft Corp. Basic Optical Mouse v2.0
Bus 003 Device 006: ID 05e3:0607 Genesys Logic, Inc. Logitech G110 Hub
Bus 003 Device 007: ID 1ab1:0588 Rigol Technologies DS1000 SERIES
Bus 003 Device 004: ID 05e3:0607 Genesys Logic, Inc. Logitech G110 Hub
Bus 003 Device 003: ID 046d:c31c Logitech, Inc. Keyboard K120
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
tim@tim-H97M-D3H:~/dev$ python usbtmc_test.py 
Rigol Technologies,DM3058E,DM3R154200794,01.01.00.02.01.00
-2.488133E-04
-2.499684E-04
-2.529245E-04
-2.540920E-04
tim@tim-H97M-D3H:~/dev$ python usbtmc_test.py 
Rigol Technologies,DM3058E,DM3R154200794,01.01.00.02.01.00
-2.580292E-04
-2.554458E-04
-2.544273E-04
-2.515706E-04
```

MatthieuDartiailh, 2018-02-12:
Thanks for posting ! It looks like pyvisa-py is somehow messing with some usb parameters and cause the instrument to even disappear ! I won't have time to look more in details at this this week but I will try to come back to it next week-end. Sadly when I tried to debug the usb backend on windows, everything was working fine ...

timhoeppner, 2018-02-12:
No rush. My python knowledge is basic at best otherwise I would try and help dig into this. However, I would be happy to run tests for you if you think of something. In the meantime, I can work with usbtmc for what I'm doing at home.

Thanks a lot for your help so far.

CompThing, 2019-11-12:
This is possibly the same as https://github.com/pyvisa/pyvisa-py/issues/71 or at least matches my comment on that bug. Once an error occurs with pyvisa-py a restart of the instrument is required to get python-usbtmc working again.
The issue shows up as a timeout because the low-level USB transaction has an error and fails. This gets mangled into a Timeout as it works its way up. Sorry have not got the traceback, but Wireshark trace attached to other bug.