pyvisa/pyvisa-py #125: Win7 AttributeError: 'Device' object has no attribute 'serial_number'

marcDesm, 2018-02-06:
I'm trying to access a USBTMC device (B&K Precision power supply) with a non NI backend with pyvisa on a Win 7 PC.  It works with NI as the back end. But not with pyvisa-py. 

A problem happen with the serial library (which I don't need pyvisa-py to use) when I call  "visa.ResourceManager('@py')".  

I get an exception that ends with  AttributeError: 'Device' object has no attribute 'serial_number'

Here are some details:

C:\Users\ss745c\Documents\IT\Python\pyvisa\pyvisa-py-master>python -m visa info
Machine Details:
   Platform ID:    Windows-7-6.1.7601-SP1
   Processor:      Intel64 Family 6 Model 79 Stepping 1, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Python27\python.exe
   Version:        2.7.13
   Compiler:       MSC v.1500 64 bit (AMD64)
   Bits:           64bit
   Build:          Dec 17 2016 20:53:40 (#v2.7.13:a06454b1afa1)
   Unicode:        UCS2

 PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      #1: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 64
        Vendor: National Instruments
         Impl. Version: 17825792
         Spec. Version: 5244672
      #2: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 17825792
         Spec. Version: 5244672
   py:
      Version: 0.3.dev0
      ASRL INSTR: Available via PySerial (3.2.1)
      TCPIP INSTR: Available
      USB RAW: Available via PyUSB (N/A). Backend: libusb10
      USB INSTR: Available via PyUSB (N/A). Backend: libusb10
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named gpib
      TCPIP SOCKET: Available

 Result of trying to use visa.ResourceManager('@py')

  File "C:\Users\ss745c\Documents\El Segundo\PROJECTS\FMS\Python\fmsControllerV51c\power.py", line 311, in connect
    self.parent.getListOfVisaInstruments()
  File "C:\Users\ss745c\Documents\El Segundo\PROJECTS\FMS\Python\fmsControllerV51c\power.py", line 55, in getListOfVisaInstruments
    self.resourceList= self.rm.list_resources()
  File "C:\Python27\lib\site-packages\pyvisa-1.8-py2.7.egg\pyvisa\highlevel.py", line 1560, in list_resources
    return self.visalib.list_resources(self.session, query)
  File "C:\Python27\lib\site-packages\pyvisa_py-0.3.dev0-py2.7.egg\pyvisa-py\highlevel.py", line 233, in list_resources
    for key, st in sessions.Session.iter_valid_session_classes()], [])
  File "C:\Python27\lib\site-packages\pyvisa_py-0.3.dev0-py2.7.egg\pyvisa-py\usb.py", line 206, in list_resources
    serial_number=dev.serial_number,
AttributeError: 'Device' object has no attribute 'serial_number'

MatthieuDartiailh, 2018-02-06:
What version of PyUSB are you using ? Pyvisa report the version as not being available, but the usb module should have a __version__ string and serial number property was added 4 years ago, so save if you use a old version I don't get how you can end up with that issue.

marcDesm2, 2018-02-06:
"pip list" reports that I have pyusb (1.0.0a2) installed

marcDesm, 2018-02-09:
Complete list from pip list

C:\Users\marc>pip list
DEPRECATION: The default format will switch to
mns) in your pip.conf under the [list] section
altgraph (0.15)
dis3 (0.1.2)
enum34 (1.1.6)
future (0.16.0)
libusb1 (1.6.4)
macholib (1.9)
matplotlib (1.4.3)
numpy (1.9.2)
pefile (2017.11.5)
PIL (1.1.7)
pip (9.0.1)
py2exe (0.6.9)
PyAudio (0.2.8)
pydub (0.16.6)
pyftpdlib (1.5.3)
PyInstaller (3.3.1)
pyparsing (2.0.3)
pyserial (3.4)
python-dateutil (2.4.2)
pyusb (1.0.0a2)
pyvisa (1.8)
PyVISA-py (0.2)
pywin32 (221)
pywinusb (0.4.2)
scipy (0.16.0)
setuptools (16.0)
six (1.9.0)

MatthieuDartiailh, 2018-02-09:
The version of pyusb on PyPI is 1.0.2. Could you update ?

marcDesm, 2018-02-09:
I ran this (BTW, is say something has been deprecated, is it pyusb?)

**C:\Users\marc>pip install --upgrade pyusb
Collecting pyusb
  Downloading pyusb-1.0.2.tar.gz (54kB)
    100% |################################| 61kB 183kB/s
Installing collected packages: pyusb
  Found existing installation: pyusb 1.0.0a2
    DEPRECATION: Uninstalling a distutils installed project (pyusb) has been deprecated and will be removed in a future version. This is due to the fact that uninstalling a distutils project will only partially uninstall the project.
    Uninstalling pyusb-1.0.0a2:
      Successfully uninstalled pyusb-1.0.0a2
  Running setup.py install for pyusb ... done
Successfully installed pyusb-1.0.2**

A check shows that the new version is installed

**C:\Users\marc>pip list
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|colu
mns) in your pip.conf under the [list] section) to disable this warning.
altgraph (0.15)
dis3 (0.1.2)
enum34 (1.1.6)
future (0.16.0)
libusb1 (1.6.4)
macholib (1.9)
matplotlib (1.4.3)
numpy (1.9.2)
pefile (2017.11.5)
PIL (1.1.7)
pip (9.0.1)
py2exe (0.6.9)
PyAudio (0.2.8)
pydub (0.16.6)
pyftpdlib (1.5.3)
PyInstaller (3.3.1)
pyparsing (2.0.3)
pyserial (3.4)
python-dateutil (2.4.2)
pyusb (1.0.2)
pyvisa (1.8)
PyVISA-py (0.2)
pywin32 (221)
pywinusb (0.4.2)
scipy (0.16.0)**

Then if I run "python -m visa info" I get the same "USB INSTR: Available via PyUSB (N/A)" . I'm assuming from your previous email that the presence of "N/A" is a bad thing.  I'm on my home machine, and I don't have the USBTMC power supply to try out with PyVisa.  But I assume I still have the problem.

C:\Users\marc>python -m visa info
Machine Details:
   Platform ID:    Windows-7-6.1.7601-SP1
   Processor:      Intel64 Family 6 Model 23 Stepping 10, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Python27\python.exe
   Version:        2.7.10
   Compiler:       MSC v.1500 32 bit (Intel)
   Bits:           32bit
   Build:          May 23 2015 09:40:32 (#default)
   Unicode:        UCS2

PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      #1: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 32
         Vendor: National Instruments
         Impl. Version: 17825792
         Spec. Version: 5244672
      #2: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 32
         Vendor: National Instruments
         Impl. Version: 17825792
         Spec. Version: 5244672
   py:
      Version: 0.2
      ASRL INSTR: Available via PySerial (3.4)
      TCPIP INSTR: Available
      USB RAW: Available via PyUSB (N/A). Backend: libusb10
      USB INSTR: Available via PyUSB (N/A). Backend: libusb10
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named gpib
      TCPIP SOCKET: Available

MatthieuDartiailh, 2018-02-09:
The first deprecation warning you get may mean that even after uninstalling pyusb some old parts are left behind. Try to go to the Lib/site_packages folder of your Python installation and remove anything related to pyusb manually and then re-install using pip.

marcDesm, 2018-02-09:
I deleted a couple of folders in Lib/site_packages and reinstalled using pip. Seems like the "N/A" is gone.
I'll try it on my work computer tomorrow.  BTW, the work computer has got pip blocked and I need to installed from a setup.py l which I download. Is running "python setup.py install" the same you think?
[image]

C:\Users\marc>python -m visa info
Machine Details:
   Platform ID:    Windows-7-6.1.7601-SP1
   Processor:      Intel64 Family 6 Model 23 Stepping 10, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Python27\python.exe
   Version:        2.7.10
   Compiler:       MSC v.1500 32 bit (Intel)
   Bits:           32bit
   Build:          May 23 2015 09:40:32 (#default)
   Unicode:        UCS2

PyVISA Version: 1.8

Backends:
   ni:
      Version: 1.8 (bundled with PyVISA)
      #1: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 32
         Vendor: National Instruments
         Impl. Version: 17825792
         Spec. Version: 5244672
      #2: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 32
         Vendor: National Instruments
         Impl. Version: 17825792
         Spec. Version: 5244672
   py:
      Version: 0.2
      ASRL INSTR: Available via PySerial (3.4)
      TCPIP INSTR: Available
      USB RAW: Available via PyUSB (1.0.2). Backend: libusb1
      USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
      GPIB INSTR:
         Please install linux-gpib to use this resource type.
         No module named gpib
      TCPIP SOCKET: Available

MatthieuDartiailh, 2018-02-09:
Yes running setup.py would be equivalent but be sure to remove any left over from the previous install.

MatthieuDartiailh, 2018-02-09:
I actually played today a bit with USB on windows and found some interesting things. One reason you may be unable to read the serial number is because the right driver is not installed for your device (the USBTMC driver of NIVISA does not work with libusb). You can use http://zadig.akeo.ie/ to actually install the WinUSB driver in its place but note that you won't be able to use NI after that (you will have to first fully uninstall the driver for the instrument).

marcDesm, 2018-02-09:
Oh, so it's not sufficient to have  "C:\Windows\System32\libusb-1.0.dll" ... this library needs a the "WinUSB" driver? OK I'll look into it.  I'm not sure if I can run an full exe on the company PC however (such as zadig). I'll see if there is a more manual way of installing "WinUSB".

MatthieuDartiailh, 2018-02-09:
The point is that you need to somehow specify which usb driver to use for the instrument as the one set-up by NI-VISA cannot be used as backend by liusb. This is similar to the udev configuration on Linux it seems. Zadig is just a convenient way to do this, but it may be possible to do it manually.

marcDesm, 2018-02-09:
Is this the driver in question?

https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/winusb-installation

MatthieuDartiailh, 2018-02-09:
It may very well be so. If you open the windows device manager, your instrument should indeed appear as a generic usb device (at least it was the case for me).