pyvisa/pyvisa-py #129: Cannot write hex characters above 7F

themikemachine, 2018-03-18:
I tried using PyVISA to communicate via RS232 with a Neslab RTE-111 water bath. I can find the instrument, but I cannot communicate with it because the pyvisa 'write' command gives an error when sending any hex character that is larger than '7F'. Unfortunately the water bath requires sending commands such as "CA 00 01 F0 02 01 2C DF".

The code is below.

```
import pyvisa
rm = pyvisa.ResourceManager()
inst = rm.open_resource('ASRL::INSTR')
inst.write('\xca\x00\x01\x20\x00\xde')
```

This gives the error:
```

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\Owner\AppData\Local\Programs\Python\Python36-32\lib\site-packages\pyvisa\resources\messagebased.py", line 223, in write
    count = self.write_raw(message.encode(enco))
UnicodeEncodeError: 'ascii' codec can't encode character '\xca' in position 0: ordinal not in range(128)

```
The write command seems to work fine for hex values below \x7f. Is it possible to send, say, '\xca'?

MatthieuDartiailh, 2018-03-18:
The issue is that `write` expect a unicode string to encode. It looks like your device relies on some lower level interface and requires you to explicitly write bytes. Here this fails because the ascii encoding fail to encode your data. You could either try to set the encoding to `utf-8` using the encoding keyword argument, or directly passing a byte string to `write_raw`

```python
inst.write_raw(b'\xca\x00\x01\x20\x00\xde')
```

themikemachine, 2018-03-19:
Thanks that helps. I can now communicate with the instrument and send it temperatures. However, I cannot read the replies coming from the instrument, always getting a 'timed out' error. 

To read the reply, I am following the "inst.write_raw" command with
inst.read()

Any ideas on what to try?

MatthieuDartiailh, 2018-03-19:
Usually instruments use a special character to indicate the end of a message (\r or \n are the usual markers). This should be given in the documentation. Given that your instrument apparently relies on closer to the metal protocol, it may use fixed length message. If it is so you can use the `read_bytes` function to read a given number of bytes.

themikemachine, 2018-03-19:
Perfect. That works great. I believe that my instrument does not use any end of message characters, that must be why it was hanging. Now I have all I need.

MatthieuDartiailh, 2018-03-19:
Perfect I will close then feel free to re-open if necessary.

closed 2018-03-19 by MatthieuDartiailh